MarlinFirmware/Marlin #18258: [BUG] Compilation error on STM32F1 (Robin Nano) with VOLUMETRIC_EXTRUDER_LIMIT enabled

Miraculix200, 2020-06-11:
### Bug Description

When enabling VOLUMETRIC_EXTRUDER_LIMIT on BOARD_MKS_ROBIN_NANO (STM32F1) board, compilation fails, showing errors like this:

Marlin/src/module/configuration_store.cpp: In static member function 'static void MarlinSettings::report(bool)':
/home/soul/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple/avr/pgmspace.h:8:23: error: expected primary-expression before ')' token

Marlin/src/module/../inc/../core/serial.h:264:60: error: expected ')' before string constant

Marlin/src/module/../inc/../core/serial.h:166:64: error: expected ')' before ';' token

Marlin/src/module/../inc/../core/serial.h:166:64: error: expected ')' before ';' token

 #define _SELP_3(a,b,c)     do{ _SEP_2(a,b); SERIAL_ECHOLNPGM(c); }while(0)
                                                                ^
Marlin/src/module/../inc/../core/serial.h:168:45: note: in expansion of macro '_SELP_3'
 #define _SELP_5(a,b,V...)  do{ _SEP_2(a,b); _SELP_3(V); }while(0)
                                             ^~~~~~~
Marlin/src/module/../inc/../core/serial.h:170:45: note: in expansion of macro '_SELP_5'
 #define _SELP_7(a,b,V...)  do{ _SEP_2(a,b); _SELP_5(V); }while(0)
                                             ^~~~~~~
Marlin/src/module/../inc/../core/serial.h:162:28: note: in expansion of macro '_SELP_7'
 #define __SELP_N(N,V...)   _SELP_##N(V)
                            ^~~~~~
Marlin/src/module/../inc/../core/serial.h:163:28: note: in expansion of macro '__SELP_N'
 #define _SELP_N(N,V...)    __SELP_N(N,V)
                            ^~~~~~~~
Marlin/src/module/../inc/../core/serial.h:195:33: note: in expansion of macro '_SELP_N'
 #define SERIAL_ECHOLNPAIR(V...) _SELP_N(NUM_ARGS(V),V)
                                 ^~~~~~~
Marlin/src/module/configuration_store.cpp:2785:9: note: in expansion of macro 'SERIAL_ECHOLNPAIR'
         SERIAL_ECHOLNPAIR("  M200 S", int(parser.volumetric_enabled)

### My Configurations

Configuration.h and Configuration_adv.h are mostly vanilla, but pins_MKS_ROBIN_NANO.h has lots of modifications. No other files have been modified. This configuration uses FSMC_GRAPHICAL_TFT with FSMC_UPSCALE 3

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4763717/config.zip)

### Steps to Reproduce

1. Checkout current bugfix-2.0.5.3 (bbe8400c5acab77e0ecde71bf824820c2a821875)
2. Replace Configuration.h, Configuration_adv.h and pins_MKS_ROBIN_NANO.h with my files
3. Compile under Linux using "platformio run -e mks_robin_nano"

MoellerDi, 2020-06-11:
PR #18262 should fix it and your config-files builds just fine on my PC

```
Linking .pio/build/mks_robin_nano/firmware.elf
Building .pio/build/mks_robin_nano/firmware.bin
Checking size .pio/build/mks_robin_nano/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.9% (used 12360 bytes from 65536 bytes)
Flash: [====      ]  37.0% (used 193844 bytes from 524288 bytes)
encrypt([".pio/build/mks_robin_nano/firmware.bin"], [".pio/build/mks_robin_nano/firmware.elf"])
```

@thinkyhead, It seems there is another small typo due to last minute code refactoring in #17017 Once again I'm very sorry for the trouble caused

Miraculix200, 2020-06-11:
Closed because #18262  fixes the problem

closed 2020-06-11 by Miraculix200