MarlinFirmware/Marlin #18260: [BUG] Inconsistant walls in marlin 2.x bugfix

danis101, 2020-06-11:
### Bug Description
Hi, Ive been using klipper and marlin 1.x on Stock board from creality cr10-s5 and My walls looks like this: https://www.dropbox.com/s/h94lebp2aihi3bo/nice.jpg?dl=0
But  I wanted to make my printer more quiet so I moved on SKR1.3 with newest bugfix marlin and then walls looks like this -> https://www.dropbox.com/s/p2r5pm14rfsd82v/bad.png?dl=0

Actually ive been changing lot of settings to get off that inconsistant layers and nothing helps(s-curve, linear advance, jerks, brezier curves) I have no Idea whats going on (Printing with octoprint via USB and from SD card gives me same results) The point is I know that everything mechanically is OK with printer becouse of earlier good quality prints. I love to use marlin thats why I bought SKR instead of RAMPS 

### My Configurations
configuration_adv.h - https://pastebin.com/gchhurU2
configuration.h - https://pastebin.com/EgsSDsQB

### Steps to Reproduce

1. Run Klipper on 8bit AVR (stock CR10-s5 board)
2. Change Board to skr 1.3 with tmc 2208 on x and y and install marlin
3. wall difference

**Expected behavior:** I'd like to see walls from marlin to looks like in 8 bit avr klipper config

**Actual behavior:** Walls on every prints looks poor (photo in description)

#### Additional Information
once again photos of printed walls
AVR + klipper - https://www.dropbox.com/s/h94lebp2aihi3bo/nice.jpg?dl=0
SKR1.3 + Marlin(SD Card print) https://www.dropbox.com/s/p2r5pm14rfsd82v/bad.png?dl=0

swilkens, 2020-06-11:
Have you printed the same file (same gcode) between both? You seem to be comparing a thin walled object with a much smaller cube with infill.

ruggb, 2020-06-11:
The pic is not too good, but I would venture to say that it is a missing steps issue. Try playing with the step pulse width settings. But I have no visibility on a TMC driver. That issue was suppose to have been fixed months ago, but a lot of changes have happened since.

danis101, 2020-06-11:
To be honest, I can't go back to old setup now to print cube But those lines are not Z wobble (one square is slightly bigger than other but they are not shifted) 
https://www.dropbox.com/s/tdk7m32asqc8x3y/104106898_688174005368103_6018784242586751225_n.jpg?dl=0 This picture looks better I think
TMC that I use are BTT 2208 As I linked in pastebin Ive even changed those settings with pulse width from defaults (I haven't printed cubes but some older prints https://www.dropbox.com/s/y1lhx9e2rjlvrv9/104282440_267305394478669_4404844134327423229_n.jpg?dl=0)

qwewer0, 2020-06-11:
Print the same gcode with the same filament on the same printer.

danis101, 2020-06-11:
> Print the same gcode with the same filament on the same printer.

As I told I cant go back to AVR + klipper and octoprint so I have only oldprints where walls was perfect for me
Now every print I do is with those walls Can you just trust me with this? I could reproduce just same wall as it was in picture as this nice one to proove my words about walls

ruggb, 2020-06-11:
yes, I think there are only 2 options.
1. it is missing steps in the X or Y direction - not talking about Z wobble
2. the extrusion is changing, which might be due to missing steps
BUT since the line across the cube is consistent a missing step or 2 in the perpendicular direction would cause that line to be offset. Extrusion changes would probably make the line vary.

danis101, 2020-06-11:
> yes, I think there are only 2 options.
> 
> 1. it is missing steps in the X or Y direction - not talking about Z wobble
> 2. the extrusion is changing, which might be due to missing steps
>    BUT since the line across the cube is consistent a missing step or 2 in the perpendicular direction would cause that line to be offset. Extrusion changes would probably make the line vary.
> 
> Changing the microstepping to a lesser number would make it look even worse as the steps it misses would be bigger.
> I assume you are using a recent bugfix.
> Different drivers react to the step pulse width differently. Mine 8825 changed for the better when it was "fixed" but not all people saw the improvement. There is a couple of things you can play with RE step pulse
> One is square wave pulse and there may be a pulse width setting. but I haven't used them so you will have to search for them.

Im gonna try that:
#define MINIMUM_STEPPER_PULSE 1 to set as 2 
also 
uncomment this:
//#define SQUARE_WAVE_STEPPING

Im gonna post cube tomorrow as it is late night for me ATM 
Anyway I thought that missing step would look like layer shift instead of changed sizes of each layer
Currently using microstepping as 16 I'd try to set for TMC's interpolation for 32 then

Marlor, 2020-06-11:
2.x bugfix is very much in flux and sometimes buggy, maybe you should start with stable Marlin 2.0.5.3 instead.
Is the TMC communication ok? what is the M122 output?
800mA sounds a bit high for y axis, especially if you use 600 at x witch feels low
Remember TMC uses RMS so 800mA would be 1.131A peak 
I have the smaller cr10-s and 650mA on both, less than that and they are skipping steps

ruggb, 2020-06-12:
If it misses a step or 2 on the X axis, when it turns the corner it will be shifted inward on that line. If it doesn't make up that step, the it will continue the shift. If it does make it up it will not. The steps you take will help determine exactly what it is.
Since there are a number of things that can cause this, u need to do something to narrow it down.
Trying 2.0.5.3 may also be a good troubleshooting idea. A lot of water has passed since it was released and changes happen. Along with Marlor's ideas for current

danis101, 2020-06-12:
> 2.x bugfix is very much in flux and sometimes buggy, maybe you should start with stable Marlin 2.0.5.3 instead.
> Is the TMC communication ok? what is the M122 output?
> 800mA sounds a bit high for y [...]

Its okay (Y axis motor is 2x bigger than x and its current is for 1.5A Max as I have read about that motor)

Anyways Tried to increase, decrease stepper current, disabled even microstepping interpolation, changed those upper settings from my previous post and it is the same thing.
Trying to reconfigure older marlin to figure it out

danis101, 2020-06-12:
Ookay here is my results for this moment(just cancelled after some layers)
https://www.dropbox.com/s/vumtc6ue4l4qmnd/comp.jpg?dl=0
From the left:
Klipper + s3d slicer(this gcode went to second print)
middle print is from Marlin + s3d slicer (same gcode as first from left print)
Right one is a thing from slic3r++ and Marlin 
Marlin stable 2.0.5.3

What I've noticed - Simplify3D doesn't have any acceleration control and jerk, If I touch those settings they affect walls. 
Currents used in Klipperr are 600 for x and y
Marlin 950 (y) 700(x)
I really dont get it what is going on Im using 500 mm/s2 acceleration for x/y and 10 jerk, also tried with 250 and 5 jerk That sounds weird to me Feel like lost 2 or 3 days for those walls to know that using jerk and accelerations will destroy walls

swilkens, 2020-06-12:
Your marlin + s3d appears to be the same as the Klipper + s3d?

I believe s3d doesn't overwrite printer acceleration / jerk values and just uses firmware defaults, while slicer++ / prusaslicer does output M204 commands in the gcode.  s3d also uses a different extrusion width compared to the default value for slicer++.

So this appears to be a slicer difference, not a Marlin issue?

Marlor, 2020-06-12:
What Config did you use as base? Default is 300 mm/s2 acceleration on CR-10_5S
Try the CR-10_5S examples with as few changes as possible: (but activate CLASSIC_JERK)
https://github.com/MarlinFirmware/Configurations/blob/release-2.0.5/config/examples/Creality/CR-10_5S/Configuration.h
https://github.com/MarlinFirmware/Configurations/blob/release-2.0.5/config/examples/Creality/CR-10_5S/Configuration_adv.h

XDA-Bam, 2020-06-12:
You have linear advance enabled according to your config. LA has to be tuned for each filament and acceleration + jerk combination, because the advance rate is calculated based on these values. If you change the accel mid-print, you basically break LA (it has the same effect as changing K).

Also, your e-jerk is set to 5. That is, in my experience, far too low for LA to work correctly. Try 15.

danis101, 2020-06-12:
> What Config did you use as base? Default is 300 mm/s2 acceleration on CR-10_5S
> Try the CR-10_5S examples with as few changes as possible: (but activate CLASSIC_JERK) [...]

this is for stock motherboard with a4988 drivers 

> You have linear advance enabled according to your config. LA has to be tuned for each filament and acceleration + jerk combination, because the advance rate is calculated based on these values. If [...]

Yeah Ive disabled it for those newest prints new configs:
config.h: https://pastebin.com/cUC5ezT5
config_adv.h: https://pastebin.com/ERqN5KpL
(just all day of changing code so I would paste it over and over for 10 or 20 times today for every update) anyway That can be a good idea to test it out, Ill try that tomorrow (I tested firmwares and gcodes for a whole day Im little tired of this for today) Anyway Ive disabled most features for test just for walls, If I get what I want I will try to enable Linear advance 

> Your marlin + s3d appears to be the same as the Klipper + s3d?
>
> I believe s3d doesn't overwrite printer acceleration / jerk values and just uses firmware defaults, while slicer++ / prusaslicer does [...]

Pretty same (minimum differences felt under finger but not that big) Thats the point s3d do not touch accelerations and jerks, I have used in slic3r++ 100% of 0.4, 105% of 0.4 and so on up to 120%, with S3D I just used 0.4 and those are results.

Marlor, 2020-06-12:
> this is for stock motherboard with a4988 drivers
> 
This is for CR-10_5S.
Use them as Base and change motherboard and drivers. Then the rest of your options match the printer and frame.
Your acceleration is now 2000 insted of the default 300

danis101, 2020-06-12:
> > this is for stock motherboard with a4988 drivers
> 
> This is for CR-10_5S.
> Use them as Base and change motherboard and drivers. Then the rest of your options match the printer and frame.
> Your acceleration is now 2000 insted of the default 300

But is it a problem when I cap acceleration in slicer? To be honest I build some improvements to avoid shaking, anyways Walls looks better with those marlin 2000 setting than mine 250-600 thats make me crazy

danis101, 2020-06-16:
I made 300 acceleration in Marlin with jerk = 5 and it isn't looking good :/

boelle, 2020-06-21:
@danis101 still an issue?

danis101, 2020-06-21:
I need  to print less than 20mm/s tor  outer walls to get nice smooth surface. Prob printing too fast cause inconsistant layers, there is no skipped steps, current now is ideal X 600, Y 700
In klipper I can  go faster thats the point  probably, If someone have something that can help - go on if not lets close it.

closed 2020-06-21 by danis101