MarlinFirmware/Marlin #18261: Using Latest Marlin 2.0.5.3 with power loss recovery does not function at all.

astro-smart, 2020-06-11:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

yanfoo, 2020-06-11:
You may want to edit your title to "bugfix-2.0.x", Your config files are not 2.0.5.3
It is still a bug, but for the beta.

I am using 2.0.5.3 stable and power loss recovery works as intended tested on an Ender 3 with SKR 1.3 board (set to onboard SDcard). You may want to go with the stable version if you want.
I did not specify a loss_pin and a pin state on my setup, try commenting theses out and see if it works.

boelle, 2020-06-21:
@astro-smart still an issue?

astro-smart, 2020-06-21:
Ive determined the only way to get this to work without the sd card being written to every 2 seconds is to use the power pin method instead. You need to get a mini ups for this to work as without it, the board powers off far to quickly for the states to be written the the sd card only with loss of power and power pin.. I ordered a mini ups for about 3 bucks each. When I get it, I will try the option and let you know. Currently, the software SD method does work but its too many writes to the sd card to use for a viable solution. After about a month, Im sure the 100k writes will kill the card. If fact I noticed even when not printing the SD card is written too. That part is a bug I would expect.

sjasonsmith, 2020-06-30:
Duplicate of #17378

@astro-smart, I'm consolidating a bunch of nearly identical issues involving power-loss recovery into one. Any further conversation can occur on the linked issue.

I'm going to quote you in a comment in that other issue. I'm not personally experienced with Power-Loss recovery, and I hope someone else will be able to contribute with what proper expectations should be for power-loss recover.

closed 2020-06-30 by sjasonsmith

thinkyhead, 2020-07-03:
Duplicate of #17378