MarlinFirmware/Marlin #18269: [BUG] LCD Chinese character half display at edge

ccccmagicboy, 2020-06-12:
### Bug Description

When use the zh_CN as language, some place in the LCD menu only display half the character at the edge, Could someone help to fix it?

### My Configurations

[cr10mini_chs.zip](https://github.com/MarlinFirmware/Marlin/files/4769862/cr10mini_chs.zip)

### Steps to Reproduce

0. checkout bugfix-2.0.x branch
1. copy the config files
2. compile the marlin for env:STM32F103RC_meeb 
3. upload the marlin

**Expected behavior:** [What you expect to happen]

Display the "开" Chinese character, which is mean on in English.

**Actual behavior:** [What actually happens]

Only display half the "开"

#### Additional Information

[image]
[image]
[image]
[image]

ellensp, 2020-06-12:
There is an issue
problem comes from that 开 is one character, and the string length function returns 1. but on the lcd it takes two 6 bit wide characters to draw..    but only moves 1 character back from the edge. 
really needs to use function to return number of pixels used not number of characters...

need to dig into  void MenuEditItemBase::draw some more...

ellensp, 2020-06-12:
Give this a try, looks good to me... but it not my language.

replace   void MenuEditItemBase::draw in ultralcd_DOGM.cpp with the following
```
  // Draw a menu item with an editable value
  void MenuEditItemBase::draw(const bool sel, const uint8_t row, PGM_P const pstr, const char* const data, const bool pgm) {
    if (mark_as_selected(row, sel)) {
      const uint8_t vallen = (pgm ? uxg_GetUtf8StrPixelWidthP(u8g.getU8g(),data) : uxg_GetUtf8StrPixelWidth(u8g.getU8g(),data));
      u8g_uint_t n = lcd_put_u8str_ind_P(pstr, itemIndex, LCD_WIDTH - 2 - vallen) * (MENU_FONT_WIDTH);
      if (vallen) {
        lcd_put_wchar(':');
        while (n > MENU_FONT_WIDTH) n -= lcd_put_wchar(' ');
        lcd_moveto(LCD_PIXEL_WIDTH - vallen -2, row_y2);
        if (pgm) lcd_put_u8str_P(data); else lcd_put_u8str((char*)data);
      }
    }
  }
```

ccccmagicboy, 2020-06-12:
Thanks for good information from you, I will try it this weekend 👍

ellensp, 2020-06-12:
It has issues... moves display up a few pixels.. no idea why...   but definitely in right place...

ellensp, 2020-06-12:
Something in broken in uxg_GetUtf8StrPixelWidthP  (which already existed but was unused...)

ellensp, 2020-06-12:
After some going down the rabbit hole I found what was causing the screen shift. It is in u8g_fontutf8.cpp. It has the following function
```
static int fontgroup_cb_draw_u8gstrlen(void *userdata, const font_t *fnt_current, const char *msg) {
  struct _uxg_drawu8_data_t * pdata = (_uxg_drawu8_data_t*)userdata;

  if (pdata->fnt_prev != fnt_current) {
    u8g_SetFont(pdata->pu8g, (const u8g_fntpgm_uint8_t*)fnt_current);
    u8g_SetFontPosBottom(pdata->pu8g);
    pdata->fnt_prev = fnt_current;
  }
  pdata->adv += u8g_GetStrPixelWidth(pdata->pu8g, (char*)msg);
  return 0;
}
```
comment out or remove     `u8g_SetFontPosBottom(pdata->pu8g);`

ellensp, 2020-06-12:
It looks to me like the characters should be down one pixel on the menus?
But this was there before my changes...

ccccmagicboy, 2020-06-14:
I post a PR https://github.com/MarlinFirmware/Marlin/pull/18294 base on your code @ellensp, let's discuss there.

closed 2020-06-16 by ccccmagicboy