MarlinFirmware/Marlin #18270: [BUG] Bilinear leveling disables set grid size with G29 P_

sdaitzman, 2020-06-12:
### Bug Description

When bilinear leveling is enabled, `G29 P_` cannot be used to set the grid size.

### My Configurations

https://github.com/sdaitzman/marlin

### Steps to Reproduce

1. Send G-code `G29 P5`

**Expected behavior:**
The Z-probe is used at 25 points, in a 5x5 grid, to probe the bed level.

**Actual behavior:**
The Z-probe is used at 9 points, in a 3x3 grid, to probe the bed level (this is based on the firmware setting).

boelle, 2020-06-21:
@sdaitzman still an issue?

sdaitzman, 2020-06-21:
Just merged latest bugfix-2.0.0 (fun 5am merge) and will go downstairs and flash and see ☺️

https://github.com/sdaitzman/Marlin

sdaitzman, 2020-06-21:
Still only measuring and returning a 3x3 grid, sadly

[image]
[image]

boelle, 2020-06-21:
did you do M502 and M500 after the update of firmware?

sdaitzman, 2020-06-21:
When I powered on the printer it had a CRC EEPROM fail and so was reset anyway, but will try that and let you know if it changes things

boelle, 2020-06-21:
just to rule out that any stupid setting in EEprom is left to play tricks

sdaitzman, 2020-06-21:
Yeah, makes sense. I ran `M502` and `M500` and power-cycled for good measure, and `G29 P4` still probes a 3x3 grid unfortunately.

sjasonsmith, 2020-07-04:
The G29 documentation indicates that the `P` option is only available with `AUTO_BED_LEVELING_LINEAR`.
https://marlinfw.org/docs/gcode/G029-abl.html

Looking at the code I can verify that the option is only parsed when `AUTO_BED_LEVELING_LINEAR` is enabled.

LINEAR and BILINEAR operate in very different ways. With LINEAR leveling, you an probe any number of points you want, but the stored output is always a flat plane defined by three points. This means you can dynamically change the number of points without impacting the layout of the EEPROM.

With BILINEAR, even point probed has to be stored to the EEPROM. The space allocated for this is decided at compile time, and cannot be adjusted dynamically.

Since this is not a bug in Marlin, I am closing this issue report.  For additional help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-04 by sjasonsmith