MarlinFirmware/Marlin #18272: repo tag/1.1.9 is wrong?

belm0, 2020-06-12:
I tried to checkout tag/1.1.9 of this repository.

I expect the log to end at July 31, 2018 (the time of Marlin 1.1.9 release).

But the log shows changes up to Nov 3, 2019, including support for BLTouch 3, etc.

```
$ git checkout 1.1.9
$ git log
commit 014ec3e81c8efc44b5aa73940e25cc0669d90859 (HEAD, tag: 1.1.9)
Author: Chris Pepper <p3p@p3psoft.co.uk>
Date:   Sun Nov 3 21:04:33 2019 +0000

    Update bug_report.md

    Make the configs zip file requirement more prominent

commit 0110faa717e642574ade9eb7aed250ee2840e942
Author: Scott Lahteine <github@thinkyhead.com>
Date:   Sun Aug 25 04:49:49 2019 -0500

    Fix Creality bed thermistor

    Co-Authored-By: thisiskeithb <13375512+thisiskeithb@users.noreply.github.com>
...
```

belm0, 2020-06-12:
Perhaps the correct hash for tag 1.1.9 is 5cf6c44f47737bdf93df6ae7c835fe2d54d0675d

```
$ git checkout 5cf6c44f47737bdf93df6ae7c835fe2d54d0675d
$ git log
commit 5cf6c44f47737bdf93df6ae7c835fe2d54d0675d (HEAD)
Author: Scott Lahteine <github@thinkyhead.com>
Date:   Sat Sep 22 03:43:32 2018 -0400

    Version 1.1.9

commit 624fd8c9ff71f114329dc6da69a6909ab568afc2
Merge: 0640bd028 0ad8646a6
Author: Scott Lahteine <github@thinkyhead.com>
Date:   Sat Sep 29 19:29:48 2018 -0500

    Merge bugfix-1.1.x

commit 0ad8646a6ca0b979cfdce2f70c5ebb44fd1fa66a
Author: Scott Lahteine <github@thinkyhead.com>
Date:   Sat Sep 29 19:16:17 2018 -0500

    Fix block->mix_steps

    Fix #11484
...
```

belm0, 2020-06-12:
If I have that right, it could be corrected as follows:

```
$ git tag -f 1.1.9 5cf6c44f47737bdf93df6ae7c835fe2d54d0675d
$ git push --tags -f
```

belm0, 2020-06-12:
cc: @thinkyhead

thisiskeithb, 2020-06-12:
It’s been retagged.

belm0, 2020-06-13:
@thisiskeithb Thank you.  I see the corrected tag in git, however the github release still points to the old hash (and significantly, the source code asset files sill correspond to the old hash).

https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.9

It may be necessary to delete the release/tag in the github UI, then create it again.

closed 2020-06-21 by boelle

thinkyhead, 2020-06-21:
Seems I forgot to include `-f` when I pushed the tag. Fixed!