MarlinFirmware/Marlin #18274: [BUG] three extruders show in motion / move axis for a dual extruder flashforge creator pro

Omeric, 2020-06-12:
i have installed marlin for the first time on my flashforge creator pro because i was fed up with the lack of support with sailfish to test it and see how it went.  i am using Marlin-bugfix-2.0.x, i am also using the flackforge creator pro  example config files.  i have only changed a few settings in them like menu timeout, tool offset and esteps.

should this not only show two extruders on a dual extruder machine?

i have also gone back to the provided config files just to check, and this still happens.

my config files are attached  [config files.zip](https://github.com/MarlinFirmware/Marlin/files/4770727/config.files.zip)

picture:    
[image]

boelle, 2020-06-23:
@thinkyhead have the code changed any in regards to this one?

MoellerDi, 2020-06-24:
it seems it's intended to show the active extruder first and a list of all existing extruders. At least that's what I understood from the code - see below. However, the menu to change K for ```LIN_ADVANCE``` is either showing the active extruder OR is looping all existing extruder (showing only 1 extruder or 2 extruders but not 3 ;-) )

Anyway, I'm very new and not familiar with the code base at all; so I could be wrong. Just trying to help by collecting information for someone who knows how it should be.

 
```void menu_move()```in ```menu_motion.cpp```
is adding a ```SUBMENU```for the active extruder and then looping and adding ```SUBMENU_MOVE_E(n)```for each existing extruder (including the active extruder):
```cpp
  #if E_MANUAL

    // The current extruder
    SUBMENU(MSG_MOVE_E, []{ _menu_move_distance(E_AXIS, []{ lcd_move_e(); }, -1); });

    #define SUBMENU_MOVE_E(N) SUBMENU_N(N, MSG_MOVE_EN, []{ _menu_move_distance(E_AXIS, []{ lcd_move_e(MenuItemBase::itemIndex); }, MenuItemBase::itemIndex); });

    #if EITHER(SWITCHING_EXTRUDER, SWITCHING_NOZZLE)

      // ...and the non-switching
      #if E_MANUAL == 7 || E_MANUAL == 5 || E_MANUAL == 3
        SUBMENU_MOVE_E(E_MANUAL - 1);
      #endif

    #elif MULTI_MANUAL

      // Independent extruders with one E-stepper per hotend
      LOOP_L_N(n, E_MANUAL) SUBMENU_MOVE_E(n);

    #endif

  #endif // E_MANUAL
```

```void menu_advanced_steps_per_mm()```in ```menu_advanced.cpp```
is similar and showing the active extruder first and then looping all existing extruder (including the active one):
```cpp
  #if ENABLED(DISTINCT_E_FACTORS)
    EDIT_ITEM_FAST(float51, MSG_E_STEPS, &planner.settings.axis_steps_per_mm[E_AXIS_N(active_extruder)], 5, 9999, []{ planner.refresh_positioning(); });
    LOOP_L_N(n, E_STEPPERS)
      EDIT_ITEM_FAST_N(float51, n, MSG_EN_STEPS, &planner.settings.axis_steps_per_mm[E_AXIS_N(n)], 5, 9999, []{ _planner_refresh_e_positioning(MenuItemBase::itemIndex); });
  #elif E_STEPPERS
    EDIT_ITEM_FAST(float51, MSG_E_STEPS, &planner.settings.axis_steps_per_mm[E_AXIS], 5, 9999, []{ planner.refresh_positioning(); });
  #endif
```

same for ```void menu_advanced_acceleration()``` 
```cpp
    #if ENABLED(DISTINCT_E_FACTORS)
      EDIT_ITEM_FAST(long5_25, MSG_AMAX_E, &planner.settings.max_acceleration_mm_per_s2[E_AXIS_N(active_extruder)], 100, max_accel_edit_scaled.e, []{ planner.reset_acceleration_rates(); });
      LOOP_L_N(n, E_STEPPERS)
        EDIT_ITEM_FAST_N(long5_25, n, MSG_AMAX_EN, &planner.settings.max_acceleration_mm_per_s2[E_AXIS_N(n)], 100, max_accel_edit_scaled.e, []{ _reset_e_acceleration_rate(MenuItemBase::itemIndex); });
    #elif E_STEPPERS
      EDIT_ITEM_FAST(long5_25, MSG_AMAX_E, &planner.settings.max_acceleration_mm_per_s2[E_AXIS], 100, max_accel_edit_scaled.e, []{ planner.reset_acceleration_rates(); });
    #endif
```

```void menu_advanced_velocity()``` is somewhat implemented differently; I'm not sure why
```cpp
    #if E_STEPPERS
      EDIT_ITEM_FAST(float3, MSG_VMAX_E, &planner.settings.max_feedrate_mm_s[E_AXIS_N(active_extruder)], 1, max_fr_edit_scaled.e);
    #endif
    #if ENABLED(DISTINCT_E_FACTORS)
      LOOP_L_N(n, E_STEPPERS)
        EDIT_ITEM_FAST_N(float3, n, MSG_VMAX_EN, &planner.settings.max_feedrate_mm_s[E_AXIS_N(n)], 1, max_fr_edit_scaled.e);
    #endif
```

```void menu_advanced_filament()``` in ```menu_advanced.cpp``` is showing either the active extruder **OR** looping all existing extruders but not both:
```cpp
    #if ENABLED(LIN_ADVANCE)
      #if EXTRUDERS == 1
        EDIT_ITEM(float42_52, MSG_ADVANCE_K, &planner.extruder_advance_K[0], 0, 999);
      #elif EXTRUDERS > 1
        LOOP_L_N(n, EXTRUDERS)
          EDIT_ITEM_N(float42_52, n, MSG_ADVANCE_K_E, &planner.extruder_advance_K[n], 0, 999);
      #endif
    #endif
```
same in ```void menu_advanced_settings()``` in ```menu_advanced.cpp```
```cpp
  #if DISABLED(NO_VOLUMETRICS) || ENABLED(ADVANCED_PAUSE_FEATURE)
    SUBMENU(MSG_FILAMENT, menu_advanced_filament);
  #elif ENABLED(LIN_ADVANCE)
    #if EXTRUDERS == 1
      EDIT_ITEM(float42_52, MSG_ADVANCE_K, &planner.extruder_advance_K[0], 0, 999);
    #elif EXTRUDERS > 1
      LOOP_L_N(n, E_STEPPERS)
        EDIT_ITEM_N(float42_52, n, MSG_ADVANCE_K_E, &planner.extruder_advance_K[n], 0, 999);
    #endif
  #endif
```

thinkyhead, 2020-06-24:
Thanks! This is intentional, as described above.

closed 2020-06-24 by thinkyhead