MarlinFirmware/Marlin #18277: [BUG] TMC connection Error while M122 return all OK

ChipTechno, 2020-06-12:
### Bug Description
TMC connection error while M122 return all OK.

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4772092/config.zip)

### Steps to Reproduce

- Compile and flash firmware via sd card
- Printer startup and showed "TMC connection error"
- Try init eeprom and M502,M500 then restart the printer
- 

**Expected behavior:**
No error showed.

**Actual behavior:**
All axis can be move with the LCD menu. M122 return no error, but the info screen showed "TMC connection error"

#### Additional Information

- Firmware version : 2.0.5.3  
- Board : BTT SKR 1_3  
- Compile env : Linux , PlatformIO, version 4.3.1

M122 result
<coce>
Send: M122
Recv: 		X	Y	Z	Z2	E
Recv: Address		0	0	0	0	0
Recv: Enabled		true	false	false	false	false
Recv: Set current	500	600	450	450	600
Recv: RMS current	489	581	428	428	581
Recv: MAX current	689	819	603	603	819
Recv: Run current	15/31	18/31	13/31	13/31	18/31
Recv: Hold current	7/31	9/31	6/31	6/31	9/31
Recv: CS actual	7/31	9/31	6/31	6/31	9/31
Recv: PWM scale	17	20	11	10	11
Recv: vsense		1=.18	1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true
Recv: msteps		16	16	8	8	16
Recv: tstep		max	max	max	max	max
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false	false
Recv: off time	4	4	4	4	4
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv:  -end		2	2	2	2	2
Recv:  -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	Z2	E
Recv: sg_result	216	186	170	16	36
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:07:00:00
Recv: 		Y	0xC0:09:00:00
Recv: 		Z	0xC0:06:00:00
Recv: 		Z2	0xC0:06:00:00
Recv: 		E	0xC0:09:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: Testing E connection... OK
Recv: ok
</code>

ChipTechno, 2020-06-13:
Turnout that something wrong with one of the stepper driver, i swapped the driver and the error disappear. But i still don't know why the error showed up while it seem to works without problem and the m122 showed nothing wrong. Note that the error only show on some config condition, this is my old config that does not have the problem with the error. 

Compiled using the same 2.0.5.3 source code
[Configuration.h](https://raw.githubusercontent.com/ChipTechno/Marlin-BTT-SKR1_3/32f5d8a33429fa4f7daadcad301512e05bdc2148/Marlin/Configuration.h)
[Configuration_adv.h](https://raw.githubusercontent.com/ChipTechno/Marlin-BTT-SKR1_3/32f5d8a33429fa4f7daadcad301512e05bdc2148/Marlin/Configuration_adv.h)

psavva, 2020-06-16:
Check your writing, did it work on the past it is this a new change to your printer?

ChipTechno, 2020-06-16:
The wiring is fine, that the first thing i checked. It work with both my old and new config. But the new config show TMC connection error on startup, it still printing fine with the error though. The M122 return no error...

dandantsui, 2020-06-17:
I have a similar issue when running bugfix branch on my MKS_ROBIN board with TMC2209 single wire UART.

M122 returns with sporadic connections errors. Sometimes multiple drivers, sometimes only one, sometimes no problem at all.  The testing connection sometimes fails as well and driver registers may be also completely fine or erroring out too. 

No such issues running on 2.0.5.3 but on latest:bugfix it occurs.

ChipTechno, 2020-06-17:
@dandantsui Mine only show error on the LCD once after power on. The serial monitor disconnect when the board restarted so i cannot check if any error when startup, but during operation it does not return any error. Both problem appear in 2.0.5.2, 2.0.5.3 and bugfix.

boelle, 2020-06-23:
@ChipCE 

Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

sjasonsmith, 2020-07-04:
@ChipCE are you able to see the serial output that occurs when you reboot the controller? You need a terminal that reconnects immediately. I use TeraTerm for this, although I believe Pronterface works as well.

As part of the normal boot messages, a brief status of each driver is printed to the serial port. Can you view this output and report which driver(s) shows the errors? It would be best to reproduce the issue several times, to determine whether it is always the same driver.

ChipCE, 2020-07-04:
@sjasonsmith Thank you! I will try to flash the fw with the error and get the log after I'm done with the current print job.

ChipCE, 2020-07-09:
I built a new printer will all new hardware and can re-produce the error will the new hardware. 
I'm using BTT SKR1.3 with Z2 motor connected on E1.

This is the error at startup
<pre>
Recv: echo: Free Memory: 13199  PlannerBufferBytes: 3456
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N0 M110 N0*125
Recv: echo:V76 stored settings retrieved (697 bytes; crc 54129)
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... Error: All LOW
Recv: Testing Z2 connection... OK
Recv: Testing E connection... OK
</pre>

I tried many times and got the same error. But after startup the M122 return all OK.
<pre>
Send: M122
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: Testing E connection... OK
</pre>

ChipCE, 2020-08-10:
The problem is not solved yet. The error still randomly showed up at startup on both 2.0.6 and fixbug branch build.

el-quique, 2020-08-30:
In Rumba32 with 6 TMC2209 (X, Y, Z, E0, E1 and E2). Error in E2.

sjasonsmith, 2020-09-11:
@el-quique are your errors consistent or random? I tested a Rumba32 with 6 drivers the other day and it was fine.

closed 2020-10-20 by github-actions[bot]