MarlinFirmware/Marlin #18278: [BUG] Heater_ID: bed - Printer halted.

uli6, 2020-06-12:


XDA-Bam, 2020-06-12:
Looks like thermal runaway protection kicking in, because the bed is heating up too slowly. Try to raise your `WATCH_TEMP_PERIOD` to 25 or 30 s. For support, it's best to ask on discord.

ruggb, 2020-06-14:
the question to ask may be why it apparently heated faster with the Anet MB since it did not through the error. Is there a heater LED and is it full on while it is heating up?

boelle, 2020-06-21:
@uli6 still an issue?

sjasonsmith, 2020-06-21:
```
#define BANG_MAX 128     // Limits current to nozzle while in bang-bang mode; 255=full current
#define PID_MAX BANG_MAX // Limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full current
```

These lines are limited your bed output to half power.

The `B@:127` in the temperature output is the current power being given to the bed, where 255 is the maximum.

sjasonsmith, 2020-06-21:
Given that this clearly looks like a configuration issue, I'm going to close the bug report. If you continue to have problems I invite you to participate in one of the several Marlin forums where many users are willing to help answer questions and debug configuration issues.

There are several of these options presented to you on the screen you see when reporting a new issue:
https://github.com/MarlinFirmware/Marlin/issues/new/choose

closed 2020-06-21 by sjasonsmith