MarlinFirmware/Marlin #18284: [BUG] Y axis only moving 25% of what it should

cmer, 2020-06-13:
### Bug Description

I just installed an SKR Mini E3 v2.0 on my Ender 5 and flashed with the latest `bugfix`.

Oddly, my Y axis is moving only 25% of what it should. My e-steps are 80,80,400.

This has been discussed with @txt4nk from BigTreeTech at: 
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/266#issuecomment-643467988 -- he seems just as stomped as me.

The following line should be around the entire bed.

[image]

### My Configurations

See: https://github.com/cmer/creality_ender5_config_and_notes/tree/master/marlin-2.0.x

### Steps to Reproduce

The same happens wether I print a sliced model, or if I manually move the Y axis with the printer screen.

**Expected behavior:** Y axis able to navigate the entire 220mm

**Actual behavior:** Every Y axis movement is 25% of what it should be. Maximum Y distance is 55mm.

txt4nk, 2020-06-13:
Just as an additional note, the first firmware I supplied was from scratch using the latest bugfix and example configurations. I went through the firmware and couldn't see anything blatantly obvious. I've got an ender 5 on an E3 Mini V1.2, so I modified my personal firmware and updated the board to the V2.0, and the result is the same.

cmer, 2020-06-13:
@txt4nk just to clarify you were also able to reproduce the issue on an E3 Mini v2 board?

txt4nk, 2020-06-13:
Negative, my V2.0 board is on the way. But taking in to account that A. a known working configuration on a V1.2  was used and you are still getting this result. B. Your printer was working just find on your previous board, which tells me it is not a printer issue.. I dont see how this is not a bug.

cmer, 2020-06-13:
`Send: M122 S1
Recv: 		X	Y	Z	E
Recv: Address		0	0	0	0
Recv: Enabled		false	false	false	false
Recv: Set current	580	650	580	650
Recv: RMS current	550	642	550	642
Recv: MAX current	776	905	776	905
Recv: Run current	17/31	20/31	17/31	20/31
Recv: Hold current	8/31	10/31	8/31	10/31
Recv: CS actual	10/31	10/31	10/31	10/31
Recv: PWM scale	33	33	33	33
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		16	16	16	16
Recv: tstep		max	max	max	max
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false
Recv: off time	4	4	4	4
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv:  -end		2	2	2	2
Recv:  -start		1	1	1	1
Recv: Stallguard thrs	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E
Recv: sg_result	54	54	54	54
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:0A:00:00
Recv: 		Y	0xC0:0A:00:00
Recv: 		Z	0xC0:0A:00:00
Recv: 		E	0xC0:0A:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK`

cmer, 2020-06-13:
```
Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z400.00 E93.00
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z5.00 E25.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R500.00 T500.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.08
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H200 B60 F255
Recv: echo:  M145 S1 H240 B70 F255
Recv: echo:; PID settings:
Recv: echo:  M301 P21.73 I1.54 D76.55
Recv: echo:; Z-Probe Offset (mm):
Recv: echo:  M851 X-45.00 Y-14.00 Z-1.40
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X580 Y650 Z580
Recv: echo:  M906 T0 E650
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:  M569 S1 T0 E
Recv: ok
```

Rikthepixel, 2020-06-14:
Okey bud, do you mean by Y axis the heated bed? also if you have a newer vesion of the Ender 5 your Z axis Steps/mm should be 1600. because the sleadscrew moves 2mm per rotation.

I think you meant to say Z instead of Y

txt4nk, 2020-06-14:
No it is his Y.. its the front/back horizontal movement.

cmer, 2020-06-14:
I oddly got it to work. I don't understand why. I rebuilt my config file by hand from the vanilla Marlin config file. This commit works for me: https://github.com/cmer/creality_ender5_config_and_notes/commit/83aae28058b7ae6b8cf9e455c590fd0e5217a67a -- You can see the diff with my old config files.

I will update this thread if I find out why it didn't work before.

closed 2020-06-14 by cmer

coolio986, 2020-07-13:
The core of the issue is the improper setting of the slave addresses for the serial communications.
In "Configuration_adv.h"

This needs to be changed:
```
  #define  X_SLAVE_ADDRESS 0
  #define  Y_SLAVE_ADDRESS 0
  #define  Z_SLAVE_ADDRESS 0
  #define X2_SLAVE_ADDRESS 0
  #define Y2_SLAVE_ADDRESS 0
  #define Z2_SLAVE_ADDRESS 0
  #define Z3_SLAVE_ADDRESS 0
  #define Z4_SLAVE_ADDRESS 0
  #define E0_SLAVE_ADDRESS 0
  #define E1_SLAVE_ADDRESS 0
  #define E2_SLAVE_ADDRESS 0
  #define E3_SLAVE_ADDRESS 0
  #define E4_SLAVE_ADDRESS 0
  #define E5_SLAVE_ADDRESS 0
  #define E6_SLAVE_ADDRESS 0
  #define E7_SLAVE_ADDRESS 0
```

To this:

```
  #define  X_SLAVE_ADDRESS 0
  #define  Y_SLAVE_ADDRESS 2
  #define  Z_SLAVE_ADDRESS 1
  #define X2_SLAVE_ADDRESS 0
  #define Y2_SLAVE_ADDRESS 0
  #define Z2_SLAVE_ADDRESS 0
  #define Z3_SLAVE_ADDRESS 0
  #define Z4_SLAVE_ADDRESS 0
  #define E0_SLAVE_ADDRESS 3
  #define E1_SLAVE_ADDRESS 0
  #define E2_SLAVE_ADDRESS 0
  #define E3_SLAVE_ADDRESS 0
  #define E4_SLAVE_ADDRESS 0
  #define E5_SLAVE_ADDRESS 0
  #define E6_SLAVE_ADDRESS 0
  #define E7_SLAVE_ADDRESS 0
```

thinkyhead, 2020-07-13:
> The core of the issue is the improper setting of the slave addresses for the serial communications.

@coolio986 — That is an issue for the [MarlinFirmware/Configurations](https://github.com/MarlinFirmware/Configurations) repository, and any fixes you submit should target the specific configuration(s) which need these settings.

sjasonsmith, 2020-07-13:
The examples for the SKR E3 Mini 2.0 already set these.

This family is a little confusing, because 1.0/2.0 use hardware serial with slave addresses, but 1.2 uses software serial and needs them all set to 0.