MarlinFirmware/Marlin #18286: [BUG] TMC2209 with LDO 0.9 VERY noisy/resonant/vibratory

freighttt, 2020-06-13:
Ive got a strange issue with a new printer im settting up. i have a set of tmc2209's, on a 24v corexy.

the original 1.8* motors are louder than they should be, but i didnt think much of it. this week i installed a set of LDO 0.9* steppers, and they vibrate so hard that the x gantry sings. they arent missing steps, the printer operates fine, theres just insane resonance, even at low speeds. i spent the last 3 days working on the frame, replacing belts, went through marlin systematically and turned on and off every setting i could think of(LA, square wave, etc). i played with microstepping numbers, voltages, etc, to no avail. i had reached the end of my sanity when i decided to stick the steppers on another machine i have, running 2208s. suprise...smooth as butter.

another thing to note is that turning stealthchop on and off results in less vibration, but not lower noise. stealthchop is actually where most of the noise and vibration comes from, with it off there is less vibration(but still tons) and the 'r2d2' sounds are pretty pronounced. i do not use hybrid threshold, but have tested it on and off when i went through the various options systematically testing them(and combinations of them).

im positive now that this is something wrong with my build, not the motors themselves. in doing some online research, i see some people mentioning issues with PWM? Beyond config/config_adv im pretty lost with coding. i thought id ask here before i go randomly messing with stuff. the video this gent posted sounds pretty similar to what im experiencing.

https://github.com/teemuatlut/TMCStepper/issues/82

I believe this to be a bug because this is a build of marlin based off of the "stock" firmware from the manufacturer. there are a few people with similar issues, but most dont seem to be experiencing this. that leads me to believe this is marlin/tmc not getting along in certain situations. this resonance/vibration is experienced with both 1.8*, and 0.9*, however, 0.9* exacerbate the issue.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Marlin-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4773934/Marlin-2.0.x.zip)

### Steps to Reproduce

1. begin print. 
2. machine sounds like chewbacca
3. enable/disable stealth chop
4. similar but chewbacca has taken xanax and has calmed down a little with stealthchop off
5. systemically test and eliminate settings like linear advance, square wave, etc
6. still chewbacca
7. install rubber dampers anywhere that vibration could be amplified
8. han solo, please take your dog back, i dont want him anymore

**Expected behavior:** [What you expect to happen]

normal levels of stepper noise and vibration

**Actual behavior:** [What actually happens]
video:
imgur.com/a/ZEBlkVX

Logo2001, 2020-06-14:
@freighttt 
Did you activate SPREAD?
Set also the Stepper to 256 Microsteps and Current @800 in configurations_adv.h (// @section tmc_smart)

freighttt, 2020-06-14:
@logo2001 
what do you mean by spread? spreadcycle? i currently have hybrid threshold off so they run either in stealthchop or spreadcycle only.

i currently have it set to 16 microsteps, should i increase it to 256 so the driver doesnt interpolate itself?(is that how it works?)

i will try setting the current via the firmware vs lcd and saving to eeprom.

Logo2001, 2020-06-15:
@freighttt 
Yes, i mean Spreadcycle.
Only on my mainboard I have to set a jumper to activate the spread function of the TmC2209 !!!
Which Board you are using?
You can also try to deactivate Stealthchop & Hybrid to check is Spreadcycle possible/available.

Logo2001, 2020-06-15:
one more question: which Stepper Driver and Version with TMC2209 do you have?

freighttt, 2020-06-15:
@Logo2001 
i am running BTT 2209 v1.2 on a BTT skr1.3

yes i have them running in spreadcycle, it is the only way to run them smoothly. unfortunately i run a small printing business and i have to run the machine, hopefully this issue isnt causing damage to the motors. it almost feels like a poor signal, but it is worse on stealthchop to where its not usable. spreadcycle is at least smooth enough that it doesnt seem to cause damage to the print. i would assume it was the drivers themselves but when i plug them into another machine, or the motors into another machine running 2208s, everything is silent and operates as it should. i have tried new fresh firmware, with no luck. i also tried and older version to no avail. 

here i have found another thread with people having the same issue just last month in may.
https://reprap.org/forum/read.php?415,852017,873784#msg-873784

Marlor, 2020-06-15:
Did you remember to reset your eeprom after the adjustment of steps/mm and speed?
Your max speed is halved and steps are double

freighttt, 2020-06-15:
@marlor   steps have been doubled. i wasnt aware that speed needed to be halved, 0.9's should have enough torque to handle normal print speed, but i will try dropping the feedrate to 250. am i understanding you correctly though?

even in stealthchop with the horrendous noise, they dont miss steps at 65mm/s, and in spreadcycle they dont have any issues with my current feedrates or acceleration.

Logo2001, 2020-06-15:
I don't understand how you can switch it around when you need soldering on the back...
[image]

Logo2001, 2020-06-15:
only when the jumper is removed, the software can set stealth Chop, Spread Cycle and Hybrid
It is preset on your drivers and you cannot change it by software. 
It is written in the manual.
It might confuse the drivers.
But I have not tested the differences.
Fact is that in Stealth Chop Mode my steppers (1,8°) are also loud.

freighttt, 2020-06-15:
Y’all have lost me now. 

BTT 2209s on an skr1.3 do not require any soldering to operate as standard with full uart control through the lcd provided your jumpers are correct.

My jumpers are correct and I can alternate between spread cycle and stealthchop through the lcd with a push of the button. There is an audible change in motor noise to confirm this. When operating correctly, they would be significantly quieter in stealth chop. In my case, the grinding noise goes away when placing them in spreadcycle, but they don’t get any louder than they already are. Stealthchop is as loud as spread cycle, though the tone and noises are different. 

Most easily the difference can be demonstrated by switching between the modes at idle. With stealth chop on, there is no noise at idle with motors enabled, whereas with stealthchop there is some white noise that is audible if held close to your ear. It is under operation that stealth chop seems to send a ‘dirty’ signal causing the motors to be extremely loud and resonant, even when held loose in your hand, despite the motion being seemingly okay.

Logo2001, 2020-06-15:
my statement was not entirely accurate.
I should finish reading the datasheet :)
see picture.
My setup definitely runs pure Spread Cycle. (Jumper to VDD)
[image]

Logo2001, 2020-06-15:
i have tested with M569 and its says "Stealth Chop mode" on all axis.
I no longer understand the world.
Stealthchop is commented out in Marlin config and the Jumper "Spread" on the MB is switched to VDD (measured).
hmmm very suspicious

freighttt, 2020-06-16:
@Logo2001 there is no spoon man

Loreagle, 2020-06-16:
I have exactly the same problem as you!!! I switched to Marlin on my Sapphire Pro in order to enable UART mode on my new ERYONE TMC2209 drivers but I have some coil whine noise also when motors are not moving. The noise reduces when I disable StealthChop but starts again when motors are running... I tried everything (chopping time, lowering or increasing current) and it's impossible to remove, it only disappear in standalone mode so using the drivers not in UART 🙁
It's exactly as you described, motor works and moves correctly without missing steps but I don't want to have a singing printer (belts resonates a lot propagating the sound).
I hope someone can help us, I would like to use these drivers at the maximum potentialities.

Logo2001, 2020-06-16:
@Loreagle 
please upload your Config and adv.h files.
I want to see if I can find something for you and freighttt.

Loreagle, 2020-06-16:
@Logo2001 Thank you for your help :) 
I'm using Marlin version from @le3tspeak that is modified and pre-configured for my printer (sapphire pro) but the problem source is related to TMC drivers configuration in my opinion. 
As I said when I use it in standalone the problem totally disappear and drivers are extremely silent, in UART disabling StealthChop noise reduces when motors not running but when running this terrible white noise at high frequency come again...
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/4789326/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/4789327/Configuration.h.txt)

freighttt, 2020-06-17:
@Logo2001 there are one or two people on discord in the trinamic section, who are also experiencing the same issue. they just popped in this morning to say they too feel that their motors have gotten louder recently.

Portzal, 2020-06-17:
I have just fitted E3D 0.9 degree steppers, I don’t think They are running as quite as the 1.8’s I replaced. There doesn’t seem to be a big difference in noise when swapping between Stealthchop and SpreadCycle either. I was on 2.0.5.3, then yesterday updated to the latest bugfix 2.0.x
No change in the amount of noise in Stealthchop.

freighttt, 2020-06-17:
@Portzal do you mind sharing your motor specs/rough print speeds?

Portzal, 2020-06-17:
@freighttt I have tried different speeds from 100mm/s to 40mm/s, I would say noise increases with speed. I am on a Bigtreetech SKR V1.4 Turbo with Bigtreetech TMC2209’s, I tried increasing and decreasing current from 950 to 1100. No noticeable improvement. I have it set to 950 at present. 
Steps, 160. 
Micro steps, 16
Motor specs,
https://e3d-online.dozuki.com/Document/FY6IQwcZVcDccZ5Q/Compact_But_Powerful_Datasheet.pdf

Logo2001, 2020-06-17:
@Loreagle 
please try to change the microsteps in adv.h from 16 to 32 or 64 on the axis you want.
Also update your config.h for the correct steps/mm.
Please Report. thx
[image]
[image]

Loreagle, 2020-06-17:
> @Loreagle 
> please try to change the microsteps in adv.h from 16 to 32 or 64 on the axis you want. (X+Y?)
> Also update your config.h for the correct steps/mm. (X+Y?)
> Please Report. thx
> [image]
> [image]
> 
> 

Ok I will try to change microsteps! Can I try also 256? 
80 Steps/mm for X and Y are correct, is a coreXY printer...

Logo2001, 2020-06-17:
No, when you change to 32 Microsteps the steps per mm will double. 80 -> 160.
@ 64 microsteps 320 steps per mm and so on.
@ 256 microsteps 1280 Steps per mm
Take care, 256 microsteps basic setting needs a lot of CPU Resources. may be SD Card Read is no longer possible :)
when you have a 32bit system u can try the basic setting raise up in acc to your final print speed
the tmc220x interpolate from your basic to 256 automaticaly. normaly

Loreagle, 2020-06-17:
Uh okay thanks, so I will try with 32 first and 160steps/mm :)

Loreagle, 2020-06-17:
@Logo2001 Can this be related to our problem?
[image]

Logo2001, 2020-06-17:
maybe, i dont now.
find it out :)

Logo2001, 2020-06-17:
regarding harware-serial:
i have looked in your board MKS_Robin_Nano.h at my place (Marlin 2.0 bugfix) 
I don't have these #defines with me.
Adapted from let3speak ?

Loreagle, 2020-06-17:
> regarding hardware-serial:
> i have looked in your board MKS_Robin_Nano.h at my place (Marlin 2.0 bugfix)
> I don't have these #defines with me.
> Adapted from let3speak ?

Yes is adapted from let3speak but I tried and it doesn't work with those lines decommented.

Btb I made a video to show you the noise: https://youtu.be/TCpenqh1t78
The repetitive noise like a metronome is due to #define MONITOR_DRIVER_STATUS, when I disable it that sounds go away but the background noise remains (I don't know if it's possible to hear it).

Logo2001, 2020-06-17:
please deactivate #define Monitor_driver_status and make the video again.
It´s not possible to hear out the noise.
thx

Loreagle, 2020-06-17:
Ok I'm so sorry, I correct my message about this because I have just found something :
[image]

When those 4 lines are uncommented I have totally NO noise also with StealthChop enabled! (when I tried before it was not working because I had motor disconnected -.-). 
But there is another problem occurring: motors does not move correctly, If I move X about 10mm also Y is moving.

Logo2001, 2020-06-17:
yes, right.
but I am happy about the way forward

Loreagle, 2020-06-17:
Wait, there is no UART mode on original Marlin for Robin Nano ahah :(

freighttt, 2020-06-17:
@Loreagle it sounds like the issue you’re having is different from the one in question, though maybe they are connected?

Loreagle, 2020-06-17:
@freighttt I don't know, the noise is as described so maybe something is wrong with UART mode. I just tried software UART instead of hardware and it's the same :(

Logo2001, 2020-06-17:
Question to the round:
Is it possible that the DEactivated #Stealth Chop is active and vice versa? So the opposite?
My printer is loud (rumbling, hissing) with #Stealth Chop enabled and very, very quiet with #Stealth Chop disabled.

freighttt, 2020-06-17:
@Logo2001 

Yes, it is possible to switch between the two. Stealthchop enabled us very very quiet at idle. In motion extremely loud, gritty, resonant. Stealthchop off is hissy at idle, and loud but smooth in motion. 

Volume is the same between both modes in motion, but stealthchop introduces a very gritty and resonantly signal/motion/whatever. During printing you can cycle between the modes to toggle on and off the vibration and resonance.

Logo2001, 2020-06-17:
please make a test on your printer.
at idle make a M569 in TFT-Terminal.
which mode is activ?

freighttt, 2020-06-17:
@Logo2001 drivers report as they should in correspondence with turning stealth chop on and off on the lcd

Here you can see each the same motor plugged into two different printers. Same voltage and steps, both on stealth chop. I’m sure you can tell which is the issue, but the one held in my hand is the problem machine, the one with it sat on a glass plate is the working machine. I also removed the driver from the problematic machine, plugged it into the working machine, and achieved the same silent result.
https://imgur.com/a/z6D2kc7

thinkyhead, 2020-06-17:
For those experiencing unusual vibration, please revert to earlier commit dates and run a quick test on each in order to determine the exact day where the problem first appears. Remember to send `M502`, `M500` to Reset EEPROM and reboot again before testing each date.

Loreagle, 2020-06-17:
I'm starting to think that my problem is related to my drivers, other people with same printer and other brand drivers doesn't have any noise issue!

Logo2001, 2020-06-17:
@Loreagle 
Stupid question: You removed the MS jumpers on the TMC Slots from the mainboard?

Loreagle, 2020-06-17:
> @Loreagle 
> Stupid question: You removed the MS jumpers on the TMC Slots from the mainboard?

For software UART yes, for hardware UART I used jumpers to select the correct address

Logo2001, 2020-06-17:
ok, fine

freighttt, 2020-06-17:
@thinkyhead 

i tested 2.0.4.4 last night. i will continue rolling backwards. i must admit i dont know any faster way than to move my configs over manually using meld, so its a slow process.

Portzal, 2020-06-17:
I have a 2.0.3 build, if I get the time I will try it out today.

Portzal, 2020-06-18:
@thinkyhead 2.0.3 did not seem to produce any positive. There is still very little difference between StealthChop and SpreadCycle as far as stepper noise goes.

freighttt, 2020-06-18:
@Portzal I also rolled tmcstepper back to 0.5.2 without improvement.

Loreagle, 2020-06-18:
@freighttt I received this feedback from @let3tspeak but I didn't try yet. He said to try with this setting and software UART.

https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/commit/1e859d163589300ff12207cf677044eafda36e0b?fbclid=IwAR35RjZDeN_dFOlDX4QYxyhLl29jrD8cNgOVitcuKsFJPKJ12A-fWlNXDDE#diff-237285c047dad808f73698c139144e1aR352

freighttt, 2020-06-18:
@Loreagle i am not use an mks Robin

Loreagle, 2020-06-18:
But if you use the drivers in standalone mode you have noise?

freighttt, 2020-06-18:
@Loreagle no? My drivers never run in standalone. They’re in uart as I’ve said. I am not having the issue you are having.

Portzal, 2020-06-19:
@freighttt I have noticed it’s more noticeable on my X and Y steppers (0.9’s) while the Z and E steppers (1.8’s) are still relatively quiet.

Portzal, 2020-06-19:
@thinkyhead I am still new to Marlin and trying to Learn as much as I can, is it possible that the TMC2209’s are not Auto tuning for some reason? (If they even do) The change in stepper to 0.9’s May be further away from the saved values of the 1.8’s and because they are not being updated they are still running with old values. Just a thought. 
Also could it be anything to do with the firmware forcing on StealthChop for sensor less homing?

freighttt, 2020-06-19:
@Portzal yeah I’d agree that it only causes issues with XY but that’s no surprise as ZE don’t move very very fast. I have 1.8 for my extruder still and my bed doesn’t move fast enough to “grind”(though it is still very loud). I run them both in spreadcycle anyways, typically.

Portzal, 2020-06-19:
@freighttt I am way out of my depth here, but I am trying to research automatic Tuning on TMC2209’s
There is a description of it in the TMC2209 data sheet. But I have not been able to cross reference that with Marlin firmware. I assume it is happening. 
https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V103.pdf

Logo2001, 2020-06-19:
https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V105.pdf

please use the newer one...

Logo2001, 2020-06-19:
BTW, i have deactivated "Hybrid" with Stealthchop on. Since is my printer totaly Silent. Only by G0 commands i hear a quiet "griiip"
Hybrid was on 100mm threshold

Logo2001, 2020-06-19:
just found:
https://marlinfw.org/docs/hardware/tmc_drivers.html#troubleshooting

freighttt, 2020-06-19:
Yeah I run hybrid threshold off and just toggle stealthchop on and off via the lcd.

I did adjust my chopper timings using the tmc provided form, but they don’t affect stealthchop, only spreadcycle. If you make headway on how to tune stealthchop, that would be awesome. Unfortunately I can build the rocket that goes to mars, but unless I’m modifying the simple config file for it, I’m useless on the software side.

Logo2001, 2020-06-19:
@freighttt 
please have a look at both configurations regarding Rsense in Config_adv.h (smart section) and the real resistor on the steppers. (closest to the motor connector)
Imprint like R100 or R110 or so and report here please.
Thanks

[75 more replies]

closed 2021-05-16 by github-actions[bot]