MarlinFirmware/Marlin #18287: SKR 1.4 Turbo Dual Cooling Fan

toophast2catch, 2020-06-13:
I have an SKR 1.4 Turbo with a Cyclops hot end and I want to utilize dual cooling fans on my setup.
In the pins file, it defines the cooling fans as:

  #define FAN_PIN                          P2_03
  #define FAN1_PIN                       P2_04

Problem I'm running into is thtat only FAN_PIN works on P2_03
I reversed these pins and made FAN_PIN P2_04 and it works that way as well but only one fan will work, not both. 

Everytime I try to set FAN 2 speed to any number it always go back to 0. 

I did notice with my testing  of FAN_PIN with P2_04 (i left FAN_1PIN commented out all together) that the option was a Stored Fan 2 and it was set to 0. 

It looks like each pin will operate by itself indepently as a single cooling fan, but they wont work together. The way I see it, it looks like some where in Marlin its storing a permanent value of 0 for FAN_1PIN and wont allow me to change it. 

I will upload my config files and a picture of my screen with the Stored Fan 2. 

To note, the picture is when FAN_PIN was set to P2_04
[Marlin 2053-TT-SKR1.4T-V2-Cyclops Plus.zip](https://github.com/MarlinFirmware/Marlin/files/4774517/Marlin.2053-TT-SKR1.4T-V2-Cyclops.Plus.zip)
[image]

ellensp, 2020-06-13:
have you tried gcode vs lcd control? might just be a lcd issue.

zonalimitatore, 2020-06-20:
Sorry but, I've trying to work on a auto_hotend_fan(for power on at 50°c, but seems that only one fan are controllable. Other fans I can't control on ours boards

boelle, 2020-06-21:
@toophast2catch still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle