MarlinFirmware/Marlin #18288: Marlin 2.0.5.3 compilation error, SKR Pro board

hackebike, 2020-06-13:
Compiling Marlin 2.0.5.3 with my custom configuration files for dual extruder
This is for a SKR Pro v1.1 dual extruder, with TMC2209 and sensorless homing/probing enabled.

Here is the error during compilation with VS Code:

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4774557/configs.zip)

`Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/usb/usb_device_ctlreq.c.o
/Users/biskero/.platformio/packages/framework-arduinoststm32@3.10601.190716/cores/arduino/stm32/usb/cdc/usbd_cdc.c: In function 'USBD_CDC_DataIn':
/Users/biskero/.platformio/packages/framework-arduinoststm32@3.10601.190716/cores/arduino/stm32/usb/cdc/usbd_cdc.c:650:3: error: unknown type name 'PCD_HandleTypeDef'; did you mean 'USBD_HandleTypeDef'?
   PCD_HandleTypeDef *hpcd = pdev->pData;
   ^~~~~~~~~~~~~~~~~
   USBD_HandleTypeDef
/Users/biskero/.platformio/packages/framework-arduinoststm32@3.10601.190716/cores/arduino/stm32/usb/cdc/usbd_cdc.c:654:65: error: request for member 'IN_ep' in something not a structure or union
     if ((hcdc->TxLastLength > 0U) && ((hcdc->TxLastLength % hpcd->IN_ep[epnum].maxpacket) == 0U)) {`

ellensp, 2020-06-13:
Need top part of log that shows what versions of frameworks and libraries your using.

hackebike, 2020-06-13:
sorry, here is the complete log:
`> Executing task in folder Marlin-2.0.5.3: platformio run <

Processing BIGTREE_SKR_PRO (platform: ststm32; board: BigTree_SKR_Pro; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/BigTree_SKR_Pro.html
PLATFORM: ST STM32 5.6.0 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 3.10601.190716 (1.6.1) 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 18 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.2
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.4.0
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit MAX31865 library> 1.2.1
|   |-- <Adafruit BusIO> 1.3.2
|   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduinoVariant/PeripheralPins.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduinoVariant/variant.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/Print.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/RingBuffer.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/USBSerial.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/WInterrupts.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/WSerial.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/WString.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/abi.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/avr/dtostrf.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/board.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/hooks.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/itoa.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/main.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/new.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/pins_arduino.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/HAL/stm32yyxx_hal.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/HAL/stm32yyxx_hal_adc.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/HAL/stm32yyxx_hal_adc_ex.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/HAL/stm32yyxx_hal_can.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/HAL/stm32yyxx_hal_cec.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/HAL/stm32yyxx_hal_comp.c.o
[... 298 lines cut ...]
mks_robin_nano            IGNORED
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           FAILED    00:00:14.577
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
===================================================================================== 1 failed, 0 succeeded in 00:00:14.577 =====================================================================================
The terminal process terminated with exit code: 1`

ellensp, 2020-06-13:
your PLATFORM: ST STM32 5.6.0  is  old  ST STM32 6.1.0 works.
Update everything.
Hit the platformio Home button along the bottom. click on Platforms and update everything red

hackebike, 2020-06-13:
I am on VS Code how I do that?

hackebike, 2020-06-13:
ok, I did the update on VS Code, getting a new error:

`> Executing task: platformio run <

Processing BIGTREE_SKR_PRO (platform: ststm32; board: BigTree_SKR_Pro; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/BigTree_SKR_Pro.html
PLATFORM: ST STM32 6.1.0 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 3.10500.190327 (1.5.0) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 18 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.2
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.4.0
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit MAX31865 library> 1.2.1
|   |-- <Adafruit BusIO> 1.3.2
|   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduinoVariant/PeripheralPins.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduinoVariant/variant.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/HardwareSerial.cpp.o
In file included from /Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/variants/BIGTREE_SKR_PRO_1v1/variant.cpp:31:
/Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/pins_arduino.h:24:15: error: expected constructor, destructor, or type conversion before '(' token
   24 | _Static_assert(LastPort <= 0x0F, "PortName must be less than 16");
      |               ^
/Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/pins_arduino.h:53:15: error: expected constructor, destructor, or type conversion before '(' token
   53 | _Static_assert(NUM_ANALOG_INPUTS <= MAX_ANALOG_INPUTS,
      |               ^
/Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/pins_arduino.h:55:15: error: expected constructor, destructor, or type conversion before '(' token
   55 | _Static_assert(NUM_ANALOG_FIRST >= NUM_ANALOG_INPUTS,
      |               ^
/Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/pins_arduino.h:63:14: error: 'uint8_t' does not name a type
   63 | static const uint8_t A0 = PIN_A0;
      |              ^~~~~~~
/Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/pins_arduino.h:67:14: error: 'uint8_t' does not name a type
   67 | static const uint8_t A1 = PIN_A1;
      |              ^~~~~~~
compilation terminated due to -fmax-errors=5.
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/IPAddress.cpp.o
In file included from /Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h:250,
                 from /Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/stm32/stm32_def.h:16,
                 from /Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/stm32/clock.h:43,
                 from /Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/wiring_time.h:23,
                 from /Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/wiring.h:38,
[... 681 lines cut ...]
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           FAILED    00:00:06.809
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
===================================================================================== 1 failed, 0 succeeded in 00:00:06.809 =====================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
`

shitcreek, 2020-06-14:
PIO gets all screwed up sometimes. Try running this script:
pio run -t program

hackebike, 2020-06-14:
no luck! The strange thing is that a Marlin build from May 19th works fine both for SKR Pro and GTR.

**same error:**
`Mac-Pro:Marlin-2.0.5.3 biskero$ 
Mac-Pro:Marlin-2.0.5.3 biskero$ 
Mac-Pro:Marlin-2.0.5.3 biskero$ 
Mac-Pro:Marlin-2.0.5.3 biskero$ pio run -t program
Processing BIGTREE_SKR_PRO (platform: ststm32; board: BigTree_SKR_Pro; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/BigTree_SKR_Pro.html
PLATFORM: ST STM32 6.1.0 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 3.10500.190327 (1.5.0) 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.190707 (10.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 18 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.2
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.4.0
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit MAX31865 library> 1.2.1
|   |-- <Adafruit BusIO> 1.3.2
|   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduinoVariant/PeripheralPins.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduinoVariant/variant.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/IPAddress.cpp.o
In file included from /Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/variants/BIGTREE_SKR_PRO_1v1/variant.cpp:31:
/Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/pins_arduino.h:24:15: error: expected constructor, destructor, or type conversion before '(' token
   24 | _Static_assert(LastPort <= 0x0F, "PortName must be less than 16");
      |               ^
/Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/pins_arduino.h:53:15: error: expected constructor, destructor, or type conversion before '(' token
   53 | _Static_assert(NUM_ANALOG_INPUTS <= MAX_ANALOG_INPUTS,
      |               ^
/Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/pins_arduino.h:55:15: error: expected constructor, destructor, or type conversion before '(' token
   55 | _Static_assert(NUM_ANALOG_FIRST >= NUM_ANALOG_INPUTS,
      |               ^
/Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/pins_arduino.h:63:14: error: 'uint8_t' does not name a type
   63 | static const uint8_t A0 = PIN_A0;
      |              ^~~~~~~
/Users/biskero/.platformio/packages/framework-arduinoststm32@3.10500.190327/cores/arduino/pins_arduino.h:67:14: error: 'uint8_t' does not name a type
   67 | static const uint8_t A1 = PIN_A1;
      |              ^~~~~~~
[... 684 lines cut ...]
mks_robin_nano            IGNORED
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           FAILED    00:00:06.176
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
===================================================================================== 1 failed, 0 succeeded in 00:00:06.176 =====================================================================================
Mac-Pro:Marlin-2.0.5.3 biskero$ `

peeff, 2020-06-14:
I have the same problem.

hackebike, 2020-06-15:
There is a strange thing in the Platformio in VS Code, if I update all I get the following, why there are 2 entries for **Platform ST STM32** ?

`Mac-Pro:Marlin-2.0.5.3 biskero$ platformio update
Updating contrib-piohome                 @ 3.2.3          [Up-to-date]
Updating contrib-pysite                  @ 2.37.191020    [Up-to-date]
Updating tool-unity                      @ 1.20500.200322 [Up-to-date]
Updating tool-scons                      @ 3.30102.0      [Up-to-date]

Platform Manager
================
Platform Atmel AVR
--------
Updating atmelavr                        @ 2.2.0          [Up-to-date]
Updating toolchain-atmelavr              @ 1.50400.190710 [Up-to-date]
Updating framework-arduino-avr           @ 5.0.0          [Up-to-date]

Platform ST STM32
--------
Updating ststm32                         @ 6.1.0          [Up-to-date]
Updating toolchain-gccarmnoneeabi        @ 1.70201.0      [Up-to-date]
Updating framework-arduinoststm32        @ 4.10800.200207 [Up-to-date]
Updating tool-stm32duino                 @ 1.0.1          [Up-to-date]
Updating tool-openocd                    @ 2.1000.190707  [Up-to-date]
Updating tool-dfuutil                    @ 1.9.200310     [Up-to-date]
Uninstalling framework-arduinoststm32 @ 4.10700.200103:         [OK]
Uninstalling toolchain-gccarmnoneeabi @ 1.90201.191206:         [OK]

Platform ST STM32
--------
Updating ststm32                         @ 5.4.3          [Detached]
Updating toolchain-gccarmnoneeabi        @ 1.70201.0      [Up-to-date]
Updating framework-arduinoststm32        @ 3.10500.190327 [Up-to-date]
Updating tool-stm32duino                 @ 1.0.1          [Up-to-date]
Updating tool-openocd                    @ 2.1000.190707  [Up-to-date]

Library Manager
===============
Library Storage: /Users/biskero/.platformio/lib
Mac-Pro:Marlin-2.0.5.3 biskero$ `

ellensp, 2020-06-16:
I used your platformio, Configuration.h and Configuration_adv.h on Marlin-2.0.5.3 
It compiled without any issues.  (a few expected warnings only)
Here is the full log https://pastebin.com/BPGjfFCv
See if you can see any differences in your build, library versions etc.

hackebike, 2020-06-16:
ok,
first think I noticed:
mine: framework-arduinoststm32 3.10500.190327 (1.5.0)
your: framework-arduinoststm32 3.10700.191028 (1.7.0)

how do I upgrade to the 1.7.0 in vs code?

ellensp, 2020-06-16:
In platforio.ini is
```
[env:BIGTREE_SKR_PRO]
platform          = ststm32
board             = BigTree_SKR_Pro
platform_packages = framework-arduinoststm32@>=3.107,<4
```

the last line should force it to use 3.107  

Try this click on platformio Home icon then click on platforms
find ST STM32 and uninstall it.

Then try a compile. It will re install it, and hopefully the correct version.

hackebike, 2020-06-16:
So I uninstall 5.4.3, than upon compilation of Marlin 2.0.5.3 it installed the framework-arduinoststm32 3.10700.191028 (1.7.0) and everything compiled ok !

Thanks

closed 2020-06-16 by hackebike