MarlinFirmware/Marlin #18291: [BUG] Preheat doesn't work after printing from Octoprint

NameOfTheDragon, 2020-06-13:
Note: I have raised this on the forum as suggested but got no response.

### Bug Description

After printing a job via Octoprint, the Preheat and temperature controls on the LCD panel have no effect. The printer "knows" that the print job has finished because it no longer displays "Stop Print". However any attempt to set the extruder or bed temperature is simply ignored.

Octoprint also sees the job as "finished" and the temperatures can successfully be controlled from the Octoprint dashboard.

So it looks like the GCode commands are still working, but the LCD is somehow "frozen out".

This can't be correct, a local interactive user should always be in control.

### My Configurations

Zip file attached as requested. See also: https://github.com/Tigra-Astronomy/Marlin/tree/TPL-SKR-E3-Mini
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4775316/Configuration.zip)

I'm running a Creality Ender-3 with BTT SKR Mini E3 mainboard. I use Octoprint running on a Raspberry Pi 3.

### Steps to Reproduce

1. Print a job using Octoprint.
2. Let the job complete (or cancel it to save time)
3. On the LCD display, try to use the "Preheat PLA" menu, or simply try to set the hot end temperature.
4. Although the menu action is apparently accepted, the temperature setpoint does not change. The setpoint cannot be changed locally until the power is cycled.

**Expected behavior:** 

Changing the temperature setpoint from the local LCD menu should always succeed.
The local interactive user must be in control.

**Actual behavior:**

After printing a job via Octoprint, temperature setpoints cannot be adjusted on the LCD until the power is cycled.

#### Additional Information

This problem began happening a few days ago when I updated my build from the `Bugfix-2.0.x` branch. My branch diverged on 6 June from the `Bugfix-2.0.x` branch and contains only changes to `Configuration.h` and `Configuration_adv.h` and I haven't modified any of the code.
[image]

boelle, 2020-06-21:
@NameOfTheDragon still an issue?

NameOfTheDragon, 2020-06-21:
It was still an issue as of 4 days ago. I'm currently trying out Klipper so I haven't been paying attention since then. I don't have the same issue with Klipper so I suppose that's a data point that suggests that OctoPrint is probably not the source of the issue.

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle