MarlinFirmware/Marlin #18295: [FR] Hotend Cooling Fan Curve?

Power3DPrinting, 2020-06-14:
### Description

Hotend cooling fan does not need to spin as fast when heated to 350 degrees celsius as it does at 150 degrees celsius for example. I thought it might be more convenient to have an automatic fan curve rather than using a written table to manually set the fan speed with M106 at the start of every print.

### Feature Workflow

1. Set hotend fan cooling pin
2. Set PWM/fan speed curve with 5 to 10 speeds relative to the nozzle temperature or seperate thermistor
3. Firmware interpolates linearly between the points the relative fan speeds (honestly, if this is too difficult, I don't mind manually setting 100 or more points for a fan curve)
4. Upload firmware

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

I have a lot of trouble understanding programming and I really appreciate the time all of you put into adding new features. Sorry if this already exists, I could not find it in config_adv.h.

AnHardt, 2020-06-14:
> Hotend cooling fan does not need to spin as fast when heated to 350 degrees celsius as it does at 150 degrees celsius for example.

Really? That direction?

ellensp, 2020-06-14:
users wont be happy till the printer can actually lower the ambient noise in a room...
This would be yet another way users would stuff up their setting and cause clogs...

Power3DPrinting, 2020-06-14:
Just a feature request... somebody put frivolous games in so I didn't think this was okay. What has you guys so stressed out these days?

Power3DPrinting, 2020-06-14:
> > Hotend cooling fan does not need to spin as fast when heated to 350 degrees celsius as it does at 150 degrees celsius for example.
> 
> Really? That direction?

> users wont be happy till the printer can actually lower the ambient noise in a room...
> This would be yet another way users would stuff up their setting and cause clogs...

Just a feature request... somebody put frivolous games in so I thought this was okay just to ask. What has you guys so stressed on a Sunday?

Marlor, 2020-06-14:
How would you even calibrate the fan curve?
Turn the Speed down until it clogs and 10% up again?
I cant imagen the small difference in power consumption and lifespan would be worth such a calibration (which cost Power and lifespan)

You are talking about M106, do you mean the hotend cooling fan or the part cooling fan?

Power3DPrinting, 2020-06-15:
> How would you even calibrate the fan curve?
> Turn the Speed down until it clogs and 10% up again?

A crude solution with the right idea of trial and error and a margin for error. Calibration should be done with a thermistor attatched to the hotend heat sink. Either the bed or second extruder thermistor or external device will work (accuracy improved with thermal paste). Then adjust fan speed and hotend temperature keeping the hotend heatsink below 40,60,80 for hotend temps 180-215 PLA, 220-245 PETG, 260-300 Other respectively. Record fan speed values. Test with actual filament. Adjust from there. Tedious, but I have 5 printers with the same hotend and another 5 to be built in the next few months so it spreads the time cost.

> I cant imagen the small difference in power consumption and lifespan would be worth such a calibration (which cost Power and lifespan)

You're not wrong, for one printer running a one or two hour print once or twice a month, it is not that much. A small print farm is perhaps the most practical savings where the cost in power and life span are multiplied significantly with non stop prints. Even a few hundred fan revolutions per minute saved add up really fast over a week or month and even more so with more printers.

> You are talking about M106, do you mean the hotend cooling fan or the part cooling fan?

I have to apologize, I really don't understand the coding that well. Both the hotend cooling fan and part cooling fan have their own controlling pin and if M106 does not work, calling out the individual pin like for on ramps and setting the frequency between 0-255 will work. I do this for fans and LED's and seems reliable enough. Just varies from hardware and setup of different machines.

Thanks for the really good questions. Really made me think about the idea and request a lot more.

AnHardt, 2020-06-15:
Energy transfer over the heatbreak is about linear to the temperature difference temperature between heater-block and heatsink. Energy transfer from the heatsink to the surrounding air depends not only upon the airspeed but also on the temperature of the surrounding air (heated chamber).
For that a system trying to hold the heatsinks temperature constant, only based on the heaterblocks temperature, and setting the fan-speed will fail when the environments temperature is not constant.

Making the cold-end-fan adjustable makes perfect sense but not when based only on the heaterblocks temperature - it needs the heatsinks temperature and a pid, or the heatsinks temperature, the environments temperature and a guessing table.

> Hotend cooling fan does not need to spin as fast when heated to 350 degrees celsius as it does at 150 degrees celsius for example.

To get at least the direction right it needs a faster spinning fan at higher temperatures and a slower spinning fan at low temperatures. What you told us is counterproductive. 

-----------------

Up to now Marlins heater subsystem is not really object oriented. For every kind of heater (nozzle, bed, chamber, ...) there is(are) its own function(s) for controling heat/cool, protection, ... . This is unmaintainable since a long time. Any extension of that system reduces the portability this system will be structurally improved sometime (if its not already to late).

Power3DPrinting, 2020-06-15:
> Energy transfer over the heatbreak is about linear to the temperature difference temperature between heater-block and heatsink. Energy transfer from the heatsink to the surrounding air depends not only upon the airspeed but also on the temperature of the surrounding air (heated chamber).
> For that a system trying to hold the heatsinks temperature constant, only based on the heaterblocks temperature, and setting the fan-speed will fail when the environments temperature is not constant.
> The temperature of the air surrounding the fins of the heatsink is the more important part of the equation. The air-stream just helps to move away the heated air near the surface of the fins.

You're absolutely right. Ambient air temperature plays a big role in cooling. I would need to conduct some more testing. Ideally the margin of error of 20 degrees celsius on the heatsink would be sufficient. I am aware fan speed does not cool linearly with higher fan speeds requiring increa

> Making the cold-end-fan adjustable makes perfect sense but not when based only on the heaterblocks temperature - it needs the heatsinks temperature and a pid, or the heatsinks temperature, the environments temperature and a guessing table.
> 
> > Hotend cooling fan does not need to spin as fast when heated to 350 degrees celsius as it does at 150 degrees celsius for example.
> 
> To get at least the direction right it needs a faster spinning fan at higher temperatures and a slower spinning fan at low temperatures. What you told us is counterproductive.
> 
> Up to now Marlins heater subsystem is not really object oriented. For every kind of heater (nozzle, bed, chamber, ...) there is(are) its own function(s) for controling heat/cool, protection, ... . This is unmaintainable since a long time. Any extension of that system reduces the probability this system will be structurally improved sometime (if its not already to late).

AnHardt, 2020-06-15:
20°C span (+-10°C) could be achievable without further measurement only in my living room (f.E. 10°C - 30°C) but not in the workshop (5°C - 35°C) and even less in a heated chamber (20°C - 80°C or more).

Power3DPrinting, 2020-06-15:
> 20°C span (+-10°C) could be achievable without further measurement only in my living room (f.E. 10°C - 30°C) but not in the workshop (5°C - 35°C) and even less in a heated chamber (20°C - 80°C or more).

You're right, a blind system would need to be calibrated within its heated chamber the same as the hotend and heated bed.

Power3DPrinting, 2020-06-16:
Heated chambers really make a good argument for remote cooling either by water and an external radiator or air compressor. Particularly with compressors which are most practical for both hotend cooling and part cooling, noise, life span, and power consumption are a fairly larger concern to optimize them than a traditional fan.