MarlinFirmware/Marlin #18298: No Output on Fan0 for Part Cooling

quad121, 2020-06-14:
[image]

I have recently fitted a BTT SKR 1.4 Turbo motherboard to my printer.

I’ve got everything working except for the Part Cooling fan which doesn’t turn when needed.

I’m using Marlin 2.0. Bugfix version. 
I have my hot end fan connected to the HE1 connection (pin 2_4) . Which is defined in Config.h as; E0_Auto_Fan_PIN P2_04 , which switches on at 40 degrees.

The fan for the Part Cooling is connected to Fan0 (pin 2_3). I only found reference to this in the ‘pins_BTT_SKR_common.h which is under the heading Heaters/Fans.

I checked the cabling etc and tested the fan0 skt but it was only putting out around 3v even when setting the screen settings to full or off.

I have attached the pin file, if someone could see if there is an error which would be stopping the fan from working I would be very gracious.

quad121, 2020-06-16:
I have included a copy of the Config.h and Config_adv.h files as well as the Pin file.
The fan won't turn on with the m106 s255 command, or won't turn on during printing when called for by the slicer program
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/4785099/Archive.zip)

boelle, 2020-06-21:
@quad121 still an issue?

quad121, 2020-06-21:
Under further investigation, the fault lied with the MOSFET which supplied Fan0 not the Marlin firmware. 
So apologies, I will close this ‘request’ error.

closed 2020-06-21 by quad121