MarlinFirmware/Marlin #18300: #error "ADVANCED_PAUSE_FEATURE requires NOZZLE_PARK_FEATURE."

Vern-K, 2020-06-14:
Help, I went to turn on the advanced pause features for my SKR mini E3 and received  #error "ADVANCED_PAUSE_FEATURE requires NOZZLE_PARK_FEATURE." I have the nozzle park turned on, so not sure what I have wrong and have spent a bit of time searching the issue and haven't seen any answers. 
Thanks 
Vern
[image]
[image]
[image]

shitcreek, 2020-06-14:
It looks like you are using an older version and this issue has been addressed:
The issue is that the condition is asking: `#elif **ENABLED**(NOZZLE_PARK_FEATURE)`
when it should be asking `#elif **DISABLED**(NOZZLE_PARK_FEATURE)`.

Please try the a newer version or simply correct that bit of code in `SanityCheck.h`

closed 2020-06-14 by shitcreek

Vern-K, 2020-06-14:
> It looks like you are using an older version and this issue has been addressed:
> The issue is that the condition is asking: #elif **ENABLED**(NOZZLE_PARK_FEATURE)
> when it should be asking #elif [...]

I updated to the 2.0.5.x last night, when I started working on it. Well changed the ENABLED to DISABLED and now have 3 issues
[image]

[image]
[image]

shitcreek, 2020-06-14:
Those lines should be comments. How did they get uncommented? Anyway, just go to check of those lines (451,455 and 656)and comment them out with `//`

Vern-K, 2020-06-14:
> Those lines should be comments. How did they get uncommented? Anyway, just go to each of those lines (451, 455 and 656) and comment them out with //

Not sure on that, I never went into there, now it's back down to 1 error [image]

Should I just be starting over?

shitcreek, 2020-06-14:
Not sure about the error, the code looks fine to me from a glance. I'd first save copies of your config and config_adv files, grab the latest Marlin, then overwrite the config files with your saved copies. The sanity check should tell you what's deprecated or has been changed.

"or does that have to do with the heaters and I need to change the 255 to what I have set as my max temp" not sure what you are referring to.

If you need further assistance, please check out:
- Our Discord channel - Join link: https://discord.gg/n5NJ59y
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration vids](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube

Vern-K, 2020-06-15:
> Not sure about the error, the code looks fine to me from a glance. I'd first save copies of your config and config_adv files, grab the latest Marlin, then overwrite the config files with your saved [...]

Yea tried to put my config files in a new download of the 2.0.5.3 and ended up with "update config files" and a bunch of other issues, so that's a no go, going to check out some of the resources you listed and will drop my full file here. 
[Marlin-bugfix-2.0.xmodified.zip](https://github.com/MarlinFirmware/Marlin/files/4777479/Marlin-bugfix-2.0.xmodified.zip)