MarlinFirmware/Marlin #18301: Skr 1.3 BLTouch 3.1 printer shutdown on second probe (change probe pin)

dertester, 2020-06-14:
Hey, 

i dont know where my config is wrong or if it is a bug.

Hardware: skr1.3, genuine bltouch3.1, tmc2209 sensorless homing for y,x ,2208 with optical
endstops for dual z and e.
Software: latest bugfix 2.0.x (same on stable 0.5)

Problem: 
I cant use the zmin prin as i want to keep z alignment with dual stepper/endstop.
When probing (g29 or m48 with multiprobing) the tip will deploy, on contact it will retract and z will move up. Then printer shuts down and i get as last message "probing failed".

My best guess that there is a problem with re-deploying the tip or some wrong readings from pin definitions.

What i tried:
M119 shows triggered when tip is retracted and open when deployed.
M280 p0 s120 selftest seems to give no error.
I use zmin and zmax for my z endstops. 

I will post my config below.
Things i changed basically:
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_PIN                  P1_26
in configuration.h and pins_btt_skr_v1_3.h

https://github.com/dertester/Marlin/blob/Chiron_BLTouch/Marlin/Configuration.h
https://github.com/dertester/Marlin/blob/Chiron_BLTouch/Marlin/Configuration_adv.h
https://github.com/dertester/Marlin/blob/Chiron_BLTouch/Marlin/src/pins/lpc1768/pins_BTT_SKR_V1_3.h

boelle, 2020-06-21:
@dertester still an issue?

dertester, 2020-06-21:
Its working now. The problem was on my side.

My probe was mounted a little to high.

I had a few problems that confused me:
1)I did not expect it to fail in that way, especially because i had a high value for "Z_PROBE_LOW_POINT".

2)I found different sources stating that it should show open when retracted. I assume 3.1 has
diffrent behaviour.

3) Different behaviours: 
 With probe much to high it would probe fine the first time but throw error after raising z the first time.
With probe a little to high it will stop after probe and deploy pin a few times crashing into the bed.

=> It would be nice if standart error messages could give more info. g30 verbose or debug output didnt help. I saw the option for probing debug after i fixed it manually. So again i assume my lack of knowledge :)

closed 2020-06-21 by dertester