MarlinFirmware/Marlin #18304: BLtouch missing bed on Y axis on Creality CR-10 S

Winovator, 2020-06-15:
-->

### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [Install ANY version of Merlin]
2. [Run Bed Leveling]
3. [BLtouch Pin will move to Y-axis at approx -2 mm beyond the bed and halt as it misses the bed and gets no feedback]

**Expected behavior:** [The BLtouch Pin should be in contact with the bed and complete the bed leveling process]

**Actual behavior:** [BLtouch Pin will move to Y-axis at approx -2 mm beyond the bed and halt as it misses the bed and gets no feedback. I have changed every Y-axis variable, I have cycled through multiple builds from Merlin, Creality, th3d,  from April 2019 to present. I have watched every video available from our friends that publish on  Youtube. I have switched between versions of Arduino.  I have switched between slicers Cura Prussia, Simplify 3D.I have compiled and uploaded no less than 100 variations. ]

#### Additional Information
I am a very patient, methodical and persistent. It appears this one is beyond me
*I have spent no less than 70 hours over the last 4 days trying to debug this issue and I have simply run out of time and options. I have stepped through and compiled countless Y-Axis

/probe related issues, debugged, and searched the web.

At this point, I must ask? Has anyone successfully added touch to a CR-10s successfully at any version of any OS? 
[Marlin 3D Printer Firmware.txt](https://github.com/MarlinFirmware/Marlin/files/4777449/Marlin.3D.Printer.Firmware.txt)
[Configuration H.txt](https://github.com/MarlinFirmware/Marlin/files/4777450/Configuration.H.txt)

or are they incompatible?

This is driving me insane! please let me know if you have a solution

Board is version 2.1
BLtouch version 3.1
Marlin is 2.x bugfix

The attached code may have a is now a mix of various releases.
th3d
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

glennerichall, 2020-06-16:
Just wondering, here for function `min_y` defining front edge
https://github.com/MarlinFirmware/Marlin/blob/c352aeb32ea80f628755f2d28ad6fde6f1cb7be9/Marlin/src/module/probe.h#L162

It does not seem to take into account home offsets from M206 nor M92 commands, have you changed those offsets ? I have the same problem with my Ender 3 with Micro Swiss direct drive extruder with X-6 and Y-14 offsets that I have set with M206 command.

`MIN_PROBE_EDGE_FRONT` has the value of:

https://github.com/MarlinFirmware/Marlin/blob/c352aeb32ea80f628755f2d28ad6fde6f1cb7be9/Marlin/Configuration.h#L963

If this is the case, it seems that it would be more intuitive to consider XY offsets set by M206 or M92 commands in edge calculation...

Marlor, 2020-06-16:
Measure if your NOZZLE_TO_PROBE_OFFSET is correct, Y is seldom 0
And see if 0 and 300 of X and Y are on the Buildplate

Winovator, 2020-06-16:
Problem Solved!

Than you for your comments and support.

The problem was that:
1. I did not have Z_SAFE_HOMING enabled.
2 I was replacing variables with numbers -hard coding values wherever possible. This causes a break in reference points.
I reverted back to 1.1.19 and only enabled a few parameters (BLtouch & Safe Homing).
Works great!

closed 2020-06-16 by Winovator