MarlinFirmware/Marlin #18305: [FR] Automatic Nozzle/Tool Offset Sensing Function

Protean-Man, 2020-06-15:
**Description**
Having to manually adjust and set the offsets for each tool works but is very slow and prone to errors as you use the printer. There will always be small shifts and movements of every piece - especially separate nozzles throughout the life a printer. What I am proposing is a function that allows the use of the Z Probe or any other pin with added sensors to have the printer test each independent nozzle the same in the same way that it homes each axis before printing. 

As an example I would add a STRAIN GAUGE to my main carriage which would be used to sense the bed no matter the active tool - when using the nozzle as the actual sensor tip. This strain gauge could also be used to sense the horizontal position of the nozzle by sensing when each tool's nozzle touches the bed. It could also be used to sense when the nozzle touches something on the horizontal axis's. 
https://www.haascnc.com/service/codes-settings.type=gcode.machine=mill.value=G37.html

**Function**
G37 in CNC is for Automatic Tool Offset Measurements - this is typically a probe on the Z but we should include X and Y automatic calibration options.
    **Automatic_Tool_Offset_Measurement**: this would enable Marlin to run G37 with X, Y, and Z parameters. In the function with #define Automatic_Tool_Offset_Measurement - there would be a DEFINE X, Y, and Z option. 

**Z** - would probe each tool at the default Z height probing location. This may or may not depend on which tools are being used within each GCode file (if the print uses tools: 0, 2, 3 - then it would only probe those tools - although I don't know if common slicer practice is good enough to list used/unused tools in each file).

**X & Y** - would probe at a specific location where the nozzle could touch something on two sides. This could be based on each printer design and would most likely be two sets of coordinates for each of the two probings. The same logic applies to all tools or all tools being used in the current Gcode print file. 
My thoughts are to create a small metal ring at the side of the bed - in line with the bed on the Z axis. Coordinates would be needed for the X/Y location of the ring, the inner diameter of the ring, outer diameter of the ring, and the Z height limit of the ring. This would ensure that the Z height would lift to avoid the ring and then drop into it and do the probing, then lift out - for each tool. 

**Operation Example**
After G28, and G29 at the start of a print, I could add G37 which would call this function and test each nozzle for their offsets. Each offset in X/Y/Z would then be saved into the firmware and used during that print. There could be further options to save to EEPROM for a one time use, or not and running it each time so that an accurate measurement is used each print. Options to probe with heaters off or on can be kept and an option to call a servo for an OFFSET RING to be lifted when needed.

I hope this makes sense. Let me know if you need further clarification.

InsanityAutomation, 2020-06-15:
Are you talking about G425 that lulzbot implemented?

```
/**
 * Automatic backlash, position and hotend offset calibration
 *
 * Enable G425 to run automatic calibration using an electrically-
 * conductive cube, bolt, or washer mounted on the bed.
 *
 * G425 uses the probe to touch the top and sides of the calibration object
 * on the bed and measures and/or correct positional offsets, axis backlash
 * and hotend offsets.
 *
 * Note: HOTEND_OFFSET and CALIBRATION_OBJECT_CENTER must be set to within
 *       ±5mm of true values for G425 to succeed.
 */
#define CALIBRATION_GCODE
```

Protean-Man, 2020-06-15:
@InsanityAutomation - thank you.
I have been looking with different terminology. 

Now my question is whether this applied to all tools? Looking at it within Marlin I don't see any comments that say it applied to each tool in the printer?

e.g. I have 4 separate tools using servo lock design - how do I apply CALIBRATION_GCODE to each seperate tool?

InsanityAutomation, 2020-06-17:
This will run through all tools sequentially

ekozan, 2020-09-30:
the only problem is ... you need to set up the tool offset and diameter before that's anoying  and if you got many tool width that's not even possible.
and backlash is only needed once or after many hours for check is maintenance is neeeded 

maybe we can and an another behavior for this command 
- we give to him -U100 -P10 (Uncertain 100mm,  Max tool width 5mm add 5mm for be sure)
- Probe 3 faces ( top,front,left )  for know where the tool is with 100mm of move  
- Probe 2 last faces for know tool width ( right, back )  with 10mm of move ( we aready know where the tool is -+10mm )