MarlinFirmware/Marlin #18306: [BUG]U8GLIB_SSD1306 error

ccccmagicboy, 2020-06-15:
### Bug Description
When I enable the U8GLIB_SSD1306 in Configuration.h, got the compile error below

```
<SCons.Script.SConscript.SConsEnvironment object at 0x0000000003DDCBE0>
<SCons.Script.SConscript.SConsEnvironment object at 0x000000000592DF28>
('Current build targets', ['buildprog', 'size'])
Use the 0x8002000 address as the marlin app entry point.
Use the 512KB flash version of stm32f103rct6 chip.
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduinoVariant\board.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduinoVariant\wirish\boards.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduinoVariant\wirish\start.S.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduinoVariant\wirish\start_c.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduinoVariant\wirish\syscalls.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\WString.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\cxxabi-compat.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\ext_interrupts.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\hooks.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\itoa.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\adc.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\adc_f1.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\bkp_f1.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\dac.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\dma.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\dma_f1.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\exc.S.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\exti.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\exti_f1.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\flash.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\fsmc_f1.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\gpio.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\gpio_f1.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\i2c.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\i2c_f1.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\iwdg.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\nvic.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\pwr.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\rcc.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\rcc_f1.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\spi.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\spi_f1.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\stm32f1\performance\isrs.S.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\stm32f1\performance\vector_table.S.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\systick.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\timer.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\timer_f1.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\usart.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\usart_f1.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\usart_private.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\usb\stm32f1\usb.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\usb\stm32f1\usb_cdcacm.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\usb\stm32f1\usb_reg_map.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\usb\usb_lib\usb_core.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\usb\usb_lib\usb_init.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\usb\usb_lib\usb_mem.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\usb\usb_lib\usb_regs.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\libmaple\util.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\main.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\new.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\pwm.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\sdio.cpp.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\stm32f1\util_hooks.c.o
Compiling .pio\build\STM32F103RC_meeb\FrameworkArduino\stm32f1\wiring_pulse_f1.cpp.o
[... 67 lines cut ...]
*** [.pio\build\STM32F103RC_meeb\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
*** [.pio\build\STM32F103RC_meeb\src\src\HAL\STM32F1\Servo.cpp.o] Error 1
*** [.pio\build\STM32F103RC_meeb\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:41:0,
                 from Marlin\src\HAL\STM32F1\eeprom_sdcard.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:798:6: error: #error "ADVANCED_PAUSE_FEATURE currently requires an LCD controller or EMERGENCY_PARSER."
     #error "ADVANCED_PAUSE_FEATURE currently requires an LCD controller or EMERGENCY_PARSER."
      ^~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1431:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      ^~~~~
*** [.pio\build\STM32F103RC_meeb\src\src\HAL\STM32F1\eeprom_flash.cpp.o] Error 1
*** [.pio\build\STM32F103RC_meeb\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp.o] Error 1
*** [.pio\build\STM32F103RC_meeb\src\src\HAL\STM32F1\eeprom_sdcard.cpp.o] Error 1
```

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4778067/Marlin.zip)

### Steps to Reproduce

0. checkout the bugfix-2.0.x branch
1. in configuration.h uncomment all LCD but #define U8GLIB_SSD1306
2. compile the marlin

ccccmagicboy, 2020-06-15:
But #define U8GLIB_SH1106 is OK, VERY ODD

ellensp, 2020-06-15:
Conditionals_LCD.h is in a absolute nightmarish state, looks like someone half converter the syntax its needs.

In Conditionals_LCD.h Find these lines.
```
// Basic Ultipanel-like displays
#if ANY(ULTIMAKERCONTROLLER, REPRAP_DISCOUNT_SMART_CONTROLLER, G3D_PANEL, RIGIDBOT_PANEL, PANEL_ONE, U8GLIB_SH1106)
  #define IS_ULTIPANEL
#endif
```
change to
```
// Basic Ultipanel-like displays
#if ANY(ULTIMAKERCONTROLLER, REPRAP_DISCOUNT_SMART_CONTROLLER, G3D_PANEL, RIGIDBOT_PANEL, PANEL_ONE, U8GLIB_SH1106, U8GLIB_SSD1306)
  #define IS_ULTIPANEL
#endif
```

now update Configuration.h from 
```
#if ENABLED(SAV_3DGLCD)
  #define U8GLIB_SSD1306
```
to
```
#if ENABLED(SAV_3DGLCD)
  #define IS_U8GLIB_SSD1306
```

Some parts of the code are expecting "IS_" prefix, others are not..  its a mess.

ccccmagicboy, 2020-06-15:
> Conditionals_LCD.h is in a absolute nightmarish state, looks like someone half converter the syntax its needs.
>
> In Conditionals_LCD.h Find these lines.
>
> ```
> // Basic Ultipanel-like displays
> #if [...]

Still can't compile successfully 😞

ellensp, 2020-06-15:
check for errors it works for me... check in Conditionals_LCD.h

ccccmagicboy, 2020-06-15:
After add IS_U8GLIB_SSD1306 here also, it works. 😀 

// Basic Ultipanel-like displays
#if ANY(ULTIMAKERCONTROLLER, REPRAP_DISCOUNT_SMART_CONTROLLER, G3D_PANEL, RIGIDBOT_PANEL, PANEL_ONE, U8GLIB_SH1106, **IS_U8GLIB_SSD1306**)
  #define IS_ULTIPANEL
#endif

boelle, 2020-06-21:
@ccccmagicboy still an issue?

ccccmagicboy, 2020-06-21:
yes, need to locate **Conditionals_LCD.h**, than find and do the change as @ellensp said. 
If enable #define IS_U8GLIB_SSD1306, use code below:
```
/**
 * SPI Ultipanels
 */

// Basic Ultipanel-like displays
#if ANY(ULTIMAKERCONTROLLER, REPRAP_DISCOUNT_SMART_CONTROLLER, G3D_PANEL, RIGIDBOT_PANEL, PANEL_ONE, U8GLIB_SH1106, IS_U8GLIB_SSD1306)
  #define IS_ULTIPANEL
#endif
```
else if enable #define U8GLIB_SSD1306, use code below:
```
/**
 * SPI Ultipanels
 */

// Basic Ultipanel-like displays
#if ANY(ULTIMAKERCONTROLLER, REPRAP_DISCOUNT_SMART_CONTROLLER, G3D_PANEL, RIGIDBOT_PANEL, PANEL_ONE, U8GLIB_SH1106, U8GLIB_SSD1306)
  #define IS_ULTIPANEL
#endif
```

closed 2020-07-29 by github-actions[bot]