MarlinFirmware/Marlin #18307: power loss recovery no option to resume

B-rad888, 2020-06-15:
### Bug Description
Power loss recovery no option to resume print when power back on. I have a SKR 1,4 turbo with UPS installed. I cant get recovery to work with the unit (BTT UPS 24V V1.0 Resume Printing While Power Off Module) I also have a BTT TFT35-E3 V3.0 Display Touch Screen Two Working Modes installed if that makes a difference. Both modes do not allow me to recover after a power failure. I included the code changes below and installed according to Bigtree Tech instructions ( https://github.com/bigtreetech/BIGTREETECH-MINI-UPS-V2.0/blob/master/BTT%20UPS%2024V%20V1.0/BTT%20UPS%2024V%20V1.0%20Manual-2019-11-29.pdf

### My Configurations
[Config H and ADV 1.zip](https://github.com/MarlinFirmware/Marlin/files/4778548/Config.H.and.ADV.1.zip)

  
**Settings in configuration.adv:**

#define POWER_LOSS_RECOVERY
  #if ENABLED(POWER_LOSS_RECOVERY)
    #define PLR_ENABLED_DEFAULT   true // Power Loss Recovery enabled by default. (Set with 'M413 Sn' & M500)
    #define BACKUP_POWER_SUPPLY       // Backup power / UPS to move the steppers on power loss
    #define POWER_LOSS_ZRAISE       3 // (mm) Z axis raise on resume (on power loss with UPS)
    #define POWER_LOSS_PIN         P1_00 // Pin to detect power loss. Set to -1 to disable default pin on boards without module.
    #define POWER_LOSS_STATE     HIGH // State of pin indicating power loss
    #define POWER_LOSS_PULL           // Set pullup / pulldown as appropriate
    #define POWER_LOSS_PURGE_LEN   20 // (mm) Length of filament to purge on resume
    #define POWER_LOSS_RETRACT_LEN 10 // (mm) Length of filament to retract on fail. Requires backup power.

    // Without a POWER_LOSS_PIN the following option helps reduce wear on the SD card,
    // especially with "vase mode" printing. Set too high and vases cannot be continued.
    #define POWER_LOSS_MIN_Z_CHANGE 0.05 // (mm) Minimum Z change before saving power-loss data
  #endif

**Settings in configuration.h:**

#define SDSUPPORT

### Steps to Reproduce
Well start a print and turn it off. However its hard to reproduce without the hardware.

**Expected behavior:** recovery option

**Actual behavior:** no recovery option

#### Additional Information
Hardware
SKR 1,4 turbo
BTT UPS 24V V1.0 Resume Printing While Power Off Module
BTT TFT35-E3 V3.0 Display Touch Screen Two Working Modes
BTT SFS V1.0 Smart Filament Sensor Detection Stuck Blocking Filament Module

Firmware
Marlin bugfix-2.0.x (newest version)
BIGTREE_TFT35_V3.0_E3.26.x (screen firmware) (newest version)

B-rad888, 2020-06-15:
Solved! :) No Bug just bad instructions BTT  Good Work Marlin ;)

closed 2020-06-15 by B-rad888

Joniss77, 2020-07-02:
I have the same problem, how did you solve it?

hackra76, 2020-08-13:
> Solved! :) No Bug just bad instructions BTT Good Work Marlin ;)

Hi, would you be so kind and tell us how did you solve the Mini UPS issue ??? I am fighting with the same problem for weeks... On power off it just slowly fades, no Z raise, nothing... Please share your solution. Thank you !!!

nesergen, 2020-08-17:
@B-rad888 
Same problem with UPS. Please share your solution.

hackra76, 2020-08-17:
To everyone running into the same problem.... You just need to add M75 to start gcode and M77 to end gocde in the config.ini file of the touch screen. Just like the picture below. It should work.
[image]

alert83, 2020-08-28:
> Solved! :) No Bug just bad instructions BTT Good Work Marlin ;)

i have the same issue. could you please provide solution

Selcuk9847, 2020-09-08:
Solved Thanks

RavenWarrior, 2020-10-25:
Wait, so you can’t use this on their btt touch screen?
Is it only in the marlin screen menu?
BTT would be on top of their game if they just explained a little more about their products.

berrenger, 2020-10-30:
> Solved Thanks

Selamlar, tam olarak çözümü anlatabilir misiniz? Benim donanım: skr 1.4 turbo, btt relay 1.2, btt smart fil.sensor, btt ups 24v, Tft35v3.  Btt ups P1_25 pinine bağlı. Tft modunda iken btt ups hariç hepsi çalışıyor.

labotecno, 2020-11-19:
i have a same problem seem when is power off fallow to print little and stop ... i put pin from pwdet P1_00 and nothing at all , skr 1.4 turbo with tmc2209 thank you to help me

rlysmart83, 2020-12-09:
> Solved! :) No Bug just bad instructions BTT Good Work Marlin ;)

please...Share How did you fix ?

sadbass, 2021-01-01:
With no mini ups is possible to have the function working?

sjasonsmith, 2021-01-01:
Posting on closed issues is a poor way to actually receive a response, especially when other people have tried with no replies.

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y