MarlinFirmware/Marlin #18308: LCD shown temperature fluctuates wildly for bed and nozzle with second print job

Alphaelectric, 2020-06-15:
I have recently completed the build of my first CoreXY printer (SecKit SK-GO) with SKE 1.3. During calibration I stumbled upon a problem with the displayed temperatures for bed and nozzle. The heating phase for bed and nozzle runs perfectly stable for the first print job after a fresh power cycle. When leaving the printer powered on and starting a second print, these values start to change up and down almost with every screen refresh. The changes are so rapid (e.g. bed goes up 5C and then down 3C and then down another 3C) that there must be a disconnect between thermistor readings and display of values as temperatures cannot fluctuate that fast on the bed side. Bed and nozzle are affected. The board "thinks" these values are proper and therefore shuts down and starts the fans multiple times when the shown values pass thresh holds. 

Restarting the board via Reset button does not cure the issue.

The issue goes away when power cycling the machine. The measurements are stable and plausible again for the first print. 

### My Configurations

SKR 1.3
Marlin 2.0.5.3
Printing from SDCard
BLTouch 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4779086/Configuration.zip)

### Steps to Reproduce

1. Turn on printer
2. Issue first print job and let it finish
3. Issue second print job and observe fluctuations in display

**Expected behavior:** [What you expect to happen]
Issue second print job and see no unplausible fluctuations

**Actual behavior:** [What actually happens]
Issue second print job and observe unplausible fluctutations

#### Additional Information

I have a video of the effect which I can link on youtube.

ellensp, 2020-06-15:
If a reset doesn't cure it, it is very unlikely it is firmware related. 

Yes please link to the you tube video.

Alphaelectric, 2020-06-15:
> If a reset doesn't cure it, it is very unlikely it is firmware related.
> 
> Yes please link to the you tube video.

Thanks for your reply. I updated the description with my latest findings and uploaded a short demonstration to youtube. It is directly related to inserting or removing an SD card into the LCD panel slot.

boelle, 2020-06-21:
@Alphaelectric still an issue?

Alphaelectric, 2020-06-21:
> @Alphaelectric still an issue?

SOLVED: 
Issue was solved by replacing the SD card

SOLUTION:
The SD card (Intenso 4GB SDHC/SDXC Class10  3411450) seems to have caused the issue. I replaced it with a SanDisk (SDSDB-8192-BULK 8GB Secure Digital Class 4) and the issue did not return after 4 times inserting and removing the card.

boelle, 2020-06-21:
we can then safely close

is this comes back we can always reopen

closed 2020-06-21 by boelle