MarlinFirmware/Marlin #18309: Disabling the nozzle heating and extruder

vatecq, 2020-06-15:
Hello, 

I'm working on a custom designed printer project using the Tevo Tarantula equipments with Easyconfig (1.1.x) firmware. I attached an induction heater on a custom designed reservoir with externel temperature control. I add a composite material to melt inside, manually by hand. 

At this point, I'd like to disable the default **nozzle heating** (and the thermistor) system as well as the **extruder** since I manage these features externally.  How can I disable these from the Configuration file without an error in printing operation. Btw, I'm using Cura slicer. 

Thanks for your kind helps.

AnHardt, 2020-06-15:
Simplest is - just not using this systems.
Use a fake thermistor 999 or 998.
Edit out all M104/M109 commands out of your g-code-files.

vatecq, 2020-06-15:
> Simplest is - just not using this systems.
> Use a fake thermistor 999 or 998.
> Edit out all M104/M109 commands out of your g-code-files.
Thanks for your help!
However, how can I apply fake termistor? and should I directly remove these M commands out? Sorry for bothering but I'm a mechanical engineer and cannot manipulate the codes that I need.

AnHardt, 2020-06-15:
Sorry i will not help any further. The new questions are far below the Niveau i expect here.
Fake thermistors -> Configuration.h. Where else? Configuring is not programming.
G-code-files are human readable and editable with a normal text-editor.

vatecq, 2020-06-15:
No need to be rude. Me and many people are here for this "simple" questions thanks to the github concept. Besides, thank you so much for helping. I'll try my best.

yet-another-average-joe, 2020-06-19:
Off topic (not Marlin), but it could help : if you use OctoPrint, the DryRun plugin does exactly what you're asking for, without the need of editing gcode file and Marlin code or config files : https://plugins.octoprint.org/plugins/DryRun/

boelle, 2020-06-21:
@vatecq still an issue?

vatecq, 2020-06-21:
@boelle Actually I made a research about applying fake thermistors into config.h and found some good examples. However this is my PhD thesis (turning DIY polymer based printer into full-metal printer) and couldn't risk about editing the firmware because of not having an experience on it. So, I kept the advise of @AnHardt and my findings as a backup plan. 

I solved the issue for now by changing the print temperature to the room temperature and disabled the extruder by opening the gears manually. 

Btw, I was searching for a way totally similar to the comment of @yet-another-average-joe, but for a marlin

boelle, 2020-06-23:
i will close this one as it seems you have figured it out

closed 2020-06-23 by boelle