MarlinFirmware/Marlin #18313: [BUG] (SKR14T bugfix 2.0.x USB_FLASH_DRIVE_SUPPORT compilation error)

Evg33, 2020-06-15:
SKR14T bugfix 2.0.x USB_FLASH_DRIVE_SUPPORT compilation error

```
Compiling .pio\build\LPC1769\src\src\sd\usb_flashdrive\lib-uhs2\masstorage.cpp.o
In file included from Marlin\src\sd\usb_flashdrive\lib-uhs2\Usb.h:47,
                 from Marlin\src\sd\usb_flashdrive\lib-uhs2\Usb.cpp:31:
Marlin\src\sd\usb_flashdrive\lib-uhs2\usb_ch9.h:119:27: error: conflicting declaration 'typedef struct USB_DEVICE_DESCRIPTOR USB_DEVICE_DESCRIPTOR'
  119 | } __attribute__((packed)) USB_DEVICE_DESCRIPTOR;
      |                           ^~~~~~~~~~~~~~~~~~~~~
In file included from .platformio\packages\framework-arduino-lpc176x\cores\arduino/CDCSerial.h:29,
                 from d:\cod\marlin\marlin\src\hal\lpc1768\hal.h:48,
                 from d:\cod\marlin\marlin\src\hal\hal.h:26,
                 from d:\cod\marlin\marlin\src\inc\marlinconfig.h:30,
                 from Marlin\src\sd\usb_flashdrive\lib-uhs2\settings.h:28,
                 from Marlin\src\sd\usb_flashdrive\lib-uhs2\Usb.h:38,
                 from Marlin\src\sd\usb_flashdrive\lib-uhs2\Usb.cpp:31:
.platformio\packages\framework-arduino-lpc176x\system\CMSIS\lib/usb/usb.h:226:3: note: previous declaration as 'typedef struct _USB_DEVICE_DESCRIPTOR USB_DEVICE_DESCRIPTOR'
  226 | } USB_DEVICE_DESCRIPTOR;
      |   ^~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\sd\usb_flashdrive\lib-uhs2\Usb.h:47,
                 from Marlin\src\sd\usb_flashdrive\lib-uhs2\Usb.cpp:31:
```

ellensp, 2020-06-16:
Confirmed. conflicting variables USB_DEVICE_DESCRIPTOR in usb_flashdrive and in framework-arduino-lpc176x\system\CMSIS\lib/usb/usb.h Since CMSIS is official from ARM probably better to adjust code in Marlin.

ellensp, 2020-06-16:
sadly no provided config files...
In Configuration.h you also need 
    #define SDSUPPORT

and in Configuration_adv.h set some pins, I choose these...
   #define USB_CS_PIN    P1_31
   #define USB_INTR_PIN  P3_25

ellensp, 2020-06-16:
With this PR It now compiles, but I have no way to test it beyond compiling.

Evg33, 2020-06-16:
@ellensp  SUCCESS ! Thnk you !

Evg33, 2020-06-16:
Testing ok !
Thank you !

Evg33, 2020-07-09:
Broken again in https://github.com/MarlinFirmware/Marlin/pull/18302
```
In file included from Marlin\src\sd\usb_flashdrive\lib-uhs2/Usb.h:47,
                 from Marlin\src\sd\usb_flashdrive\Sd2Card_FlashDrive.cpp:85:
Marlin\src\sd\usb_flashdrive\lib-uhs2/usb_ch9.h:119:27: error: conflicting declaration 'typedef struct USB_DEVICE_DESCRIPTOR USB_DEVICE_DESCRIPTOR'
  119 | } __attribute__((packed)) USB_DEVICE_DESCRIPTOR;
      |                           ^~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\User\.platformio\packages\framework-arduino-lpc176x\cores\arduino/CDCSerial.h:29,
                 from c:\evg\cod\nail\marlin\src\hal\lpc1768\hal.h:48,
                 from c:\evg\cod\nail\marlin\src\hal\hal.h:26,
                 from c:\evg\cod\nail\marlin\src\inc/MarlinConfig.h:30,
                 from c:\evg\cod\nail\marlin\src\marlincore.h:24,
                 from Marlin\src\sd\usb_flashdrive\Sd2Card_FlashDrive.cpp:43:
C:\Users\User\.platformio\packages\framework-arduino-lpc176x\system\CMSIS\lib/usb/usb.h:226:3: note: previous declaration as 'typedef struct _USB_DEVICE_DESCRIPTOR USB_DEVICE_DESCRIPTOR'
  226 | } USB_DEVICE_DESCRIPTOR;
      |   ^~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\sd\usb_flashdrive\lib-uhs2/Usb.h:47,
                 from Marlin\src\sd\usb_flashdrive\Sd2Card_FlashDrive.cpp:85:
Marlin\src\sd\usb_flashdrive\lib-uhs2/usb_ch9.h:131:27: error: conflicting declaration 'typedef struct USB_CONFIGURATION_DESCRIPTOR USB_CONFIGURATION_DESCRIPTOR'
  131 | } __attribute__((packed)) USB_CONFIGURATION_DESCRIPTOR;
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\User\.platformio\packages\framework-arduino-lpc176x\cores\arduino/CDCSerial.h:29,
                 from c:\evg\cod\nail\marlin\src\hal\lpc1768\hal.h:48,
                 from c:\evg\cod\nail\marlin\src\hal\hal.h:26,
                 from c:\evg\cod\nail\marlin\src\inc/MarlinConfig.h:30,
                 from c:\evg\cod\nail\marlin\src\marlincore.h:24,
                 from Marlin\src\sd\usb_flashdrive\Sd2Card_FlashDrive.cpp:43:
C:\Users\User\.platformio\packages\framework-arduino-lpc176x\system\CMSIS\lib/usb/usb.h:270:3: note: previous declaration as 'typedef struct _USB_CONFIGURATION_DESCRIPTOR USB_CONFIGURATION_DESCRIPTOR'
  270 | } USB_CONFIGURATION_DESCRIPTOR;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\sd\usb_flashdrive\lib-uhs2/Usb.h:47,
                 from Marlin\src\sd\usb_flashdrive\Sd2Card_FlashDrive.cpp:85:
Marlin\src\sd\usb_flashdrive\lib-uhs2/usb_ch9.h:144:27: error: conflicting declaration 'typedef struct USB_INTERFACE_DESCRIPTOR USB_INTERFACE_DESCRIPTOR'
  144 | } __attribute__((packed)) USB_INTERFACE_DESCRIPTOR;
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\User\.platformio\packages\framework-arduino-lpc176x\cores\arduino/CDCSerial.h:29,
                 from c:\evg\cod\nail\marlin\src\hal\lpc1768\hal.h:48,
                 from c:\evg\cod\nail\marlin\src\hal\hal.h:26,
                 from c:\evg\cod\nail\marlin\src\inc/MarlinConfig.h:30,
                 from c:\evg\cod\nail\marlin\src\marlincore.h:24,
                 from Marlin\src\sd\usb_flashdrive\Sd2Card_FlashDrive.cpp:43:
C:\Users\User\.platformio\packages\framework-arduino-lpc176x\system\CMSIS\lib/usb/usb.h:293:3: note: previous declaration as 'typedef struct _USB_INTERFACE_DESCRIPTOR USB_INTERFACE_DESCRIPTOR'
  293 | } USB_INTERFACE_DESCRIPTOR;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\sd\usb_flashdrive\lib-uhs2/Usb.h:47,
                 from Marlin\src\sd\usb_flashdrive\Sd2Card_FlashDrive.cpp:85:
Marlin\src\sd\usb_flashdrive\lib-uhs2/usb_ch9.h:154:27: error: conflicting declaration 'typedef struct USB_ENDPOINT_DESCRIPTOR USB_ENDPOINT_DESCRIPTOR'
  154 | } __attribute__((packed)) USB_ENDPOINT_DESCRIPTOR;
      |                           ^~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\User\.platformio\packages\framework-arduino-lpc176x\cores\arduino/CDCSerial.h:29,
                 from c:\evg\cod\nail\marlin\src\hal\lpc1768\hal.h:48,
                 from c:\evg\cod\nail\marlin\src\hal\hal.h:26,
                 from c:\evg\cod\nail\marlin\src\inc/MarlinConfig.h:30,
                 from c:\evg\cod\nail\marlin\src\marlincore.h:24,
                 from Marlin\src\sd\usb_flashdrive\Sd2Card_FlashDrive.cpp:43:
C:\Users\User\.platformio\packages\framework-arduino-lpc176x\system\CMSIS\lib/usb/usb.h:313:3: note: previous declaration as 'typedef struct _USB_ENDPOINT_DESCRIPTOR USB_ENDPOINT_DESCRIPTOR'
  313 | } USB_ENDPOINT_DESCRIPTOR;
      |   ^~~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\sd\usb_flashdrive\lib-uhs2\Usb.h:47,
                 from Marlin\src\sd\usb_flashdrive\lib-uhs2\Usb.cpp:31:
Marlin\src\sd\usb_flashdrive\lib-uhs2\usb_ch9.h:119:27: error: conflicting declaration 'typedef struct USB_DEVICE_DESCRIPTOR USB_DEVICE_DESCRIPTOR'
  119 | } __attribute__((packed)) USB_DEVICE_DESCRIPTOR;
      |                           ^~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\User\.platformio\packages\framework-arduino-lpc176x\cores\arduino/CDCSerial.h:29,
                 from c:\evg\cod\nail\marlin\src\hal\lpc1768\hal.h:48,
                 from c:\evg\cod\nail\marlin\src\hal\hal.h:26,
                 from c:\evg\cod\nail\marlin\src\inc\marlinconfig.h:30,
                 from Marlin\src\sd\usb_flashdrive\lib-uhs2\settings.h:28,
[... cut ...]
                 from Marlin\src\sd\usb_flashdrive\lib-uhs2\masstorage.cpp:30:
C:\Users\User\.platformio\packages\framework-arduino-lpc176x\system\CMSIS\lib/usb/usb.h:313:3: note: previous declaration as 'typedef struct _USB_ENDPOINT_DESCRIPTOR USB_ENDPOINT_DESCRIPTOR'
  313 | } USB_ENDPOINT_DESCRIPTOR;
      |   ^~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\LPC1768\src\src\sd\usb_flashdrive\lib-uhs2\Usb.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\sd\usb_flashdrive\lib-uhs2\masstorage.cpp.o] Error 1

```

ellensp, 2020-07-09:
yes this commit undid my fix.. sigh.

ellensp, 2020-07-10:
This should be fixed again now.

closed 2020-07-10 by ellensp