MarlinFirmware/Marlin #18318: Will not compile with out errors like a lot of them

Vern-K, 2020-06-15:
[image]
[image]

So started with a fresh download and touched nothing, then closed folder put the config files in that I want, changed nothing and get the same thing.

randellhodges, 2020-06-16:
Is that the output of the compile? Do an actual compile and post the results of the compile.

Just opening it in VS Code before a successful compile will show a lot of things as "errors" that are not actual errors.

Vern-K, 2020-06-16:
> Is that the output of the compile? Do an actual compile and post the results of the compile.
> Just opening it in VS Code before a successful compile will show a lot of things as "errors" that are not actual errors.

That is what it shows when I try to build it, just tried again and got errors like this, I still haven't change anything 
[image]
[image]

randellhodges, 2020-06-16:
One problem is that board in the configuration.h file does not match whatever the default is set in the platformio.ini file.

You are going to need to post your config files and platformio.ini files. 

I think you were posting on Discord? That is probably where this should go unless it is found to be a code problem vs a configuration problem.

Vern-K, 2020-06-16:
> One problem is that board in the configuration.h file does not match whatever the default is set in the platformio.ini file.
> You are going to need to post your config files and platformio.ini files. [...]

The files are just as they came I have changed nothing other then to put the SKR mini E3 V1.2 config files in place of the original ones, shouldn't it build with no issues if no changes were made?

randellhodges, 2020-06-16:
It depends on how you do the build. I think auto build marlin will modify
the platformio.ini file based on what you select in the configuration.h
file, I'm not sure, I never use it.

You need to have the correct entry there, so it can download dependencies
and do a proper build of the source code.

If you are trying to use the auto build marlin, someone else will need to
pipe in on that

Vern-K, 2020-06-16:
> It depends on how you do the build. I think auto build marlin will modify
> the platformio.ini file based on what you select in the configuration.h
> file, I'm not sure, I never use it.
>
> You need to [...]

Shouldn't all the dependencies be there like I said I have changed nothing other then the config file that was downloaded from the link provided in the config folder, I haven't seen any instructions of anything else that would be needed, unlike when I first set my board up and had to go in and name the board and all that but the proper name is in there along with all the settings I had to do the first time now that there is a config file for the board.  I guess I will just leave my printer as is since it works and I seem to have a lot of issues updating that it looks like no one else has and no one has answers for. 
Thank you for your time.

randellhodges, 2020-06-16:
Post your platformio.ini and your config files please and your full build log (from the output tab I believe)

Vern-K, 2020-06-16:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4784177/Configuration_adv.zip)
[platformio.zip](https://github.com/MarlinFirmware/Marlin/files/4784176/platformio.zip)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4784178/Configuration.zip)

they are just as they download, I haven't made any changes

ellensp, 2020-06-16:
In platformio you have 
default_envs = mega2560

you need one of these 

#   STM32F103RC_btt ............. RCT6 with 256K
#   STM32F103RC_btt_USB ......... RCT6 with 256K (USB mass storage)
#   STM32F103RC_btt_512K ........ RCT6 with 512K
#   STM32F103RC_btt_512K_USB .... RCT6 with 512K (USB mass storage)

eg default_envs = STM32F103RC_btt

Vern-K, 2020-06-16:
[image]
Where would I find autobuild I have been just using the check on the bottom?

ellensp, 2020-06-16:
please read https://marlinfw.org/docs/basics/install_platformio.html

Vern-K, 2020-06-16:
> please read https://marlinfw.org/docs/basics/install_platformio.html
> In particular https://marlinfw.org/docs/basics/auto_build_marlin.html

Thank you installed that, but since my last post I did find the area I saw in the one video where you pick the build tried the large memory first and gave me errors on memory so did the lower memory that matches my board and still had errors just down loaded the autobuild and will restart and try again (BAC allowing) this is the last list of errors I got with trying to build the STM32F103RC_btt... Thank you for helping me printer is running on a basic pause that may or may not work with what I have running right now. If you want a 2mm hole drilled though 60mm to hard mount your direct drive to your hot end I can do that super easy but the software side I'm a little lacking 
[image]

ellensp, 2020-06-16:
in your platfomio.ini replace your [common] section with the following

```
[common]
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
extra_scripts = pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
build_flags = -fmax-errors=5 -g -D__MARLIN_FIRMWARE__ -fmerge-all-constants
lib_deps =
  LiquidCrystal
  TMCStepper@>=0.6.2,<1.0.0
  Adafruit MAX31865 library@==1.1
  Adafruit NeoPixel
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
  Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/0.8.0.zip
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
  SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip
  SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip
```

Vern-K, 2020-06-16:
> in your platfomio.ini replace your [common] section with the following
> [common]
> default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
> extra_scripts = pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
> build_flags = -fmax-errors=5 -g -D__MARLIN_FIRMWARE__ -fmerge-all-constants
> lib_deps =
>   LiquidCrystal
>   TMCStepper@>=0.6.2,<1.0.0
>   Adafruit MAX31865 library@==1.1
>   Adafruit NeoPixel
>   U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
>   LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
>   Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/0.8.0.zip
>   SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
>   SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip
>   SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip
> 
> This issue is that Adafruit released a new versionn of their MAX31865 library. Which doesn't work in
> Marlin any more. So you have to tell it to use the older version. This happened after this version of Marlin was released. Is already patched in bugfix.

Ok, thanks, I will down load the bugfix. So in theory I should be able to put the config files I want in that and be good other then the changes I need to make for the hardware I'm running?

ellensp, 2020-06-16:
NO Bugfix is not for you! 
Your configs will not work on bugfix. 
You would be starting again.

Vern-K, 2020-06-16:
> NO Bugfix is not for you!
> Your configs will not work on bugfix.
> You would be starting again.

Ok good to know.

ellensp, 2020-06-18:
Since we have worked through all your problems, can we close this issue?

Vern-K, 2020-06-18:
> Since we have worked through all your problems, can we close this issue?

Didn't get it working but sure since I won't have the time work on it for a while now...

sjasonsmith, 2020-06-20:
@Vern-K I'll go ahead and close this, then.
When you get back to it I suggest you drop by the Marlin Discord or one of the other support forums for help. If you end up confirming there is a bug somewhere you can open a new issue then.

closed 2020-06-20 by sjasonsmith