MarlinFirmware/Marlin #18319: A8 Clone running Marlin 2.0.5.3 Auto bed leveling bad corrections.

SagittariusC, 2020-06-16:
I have bi-linear bed leveling enabled on a P802ma (Anet A8 clone), and no matter what I do, it seems to fail to correct properly. I can level my bed mostly by hand, and it works fine. But if I try to use auto leveling, it gives me results showing a slightly warped bed, and appears to fail to correct it. It becomes worse than my hand leveling.

Any suggestions as to whats going on?? Thanks

My Configurations files, built off supplied A8 config files
https://drive.google.com/open?id=1jyv3vw48wWF2PphlNWfajTjJTXPgru84

Steps to Reproduce: Just normal bed leveling procedure: G28, then G29 then print.

Expected behavior: Even first layer

Actual behavior: Usually too close on the right side, and too far on the left side

Additional Information
Bi linear example data when it did not work correctly

```
      0      1      2      3      4
 0 -0.053 +0.053 -0.018 -0.085 -0.327
 1 +0.038 +0.090 -0.015 -0.108 -0.382
 2 +0.012 +0.072 +0.002 -0.067 -0.310
 3 +0.055 +0.110 +0.005 -0.100 -0.377
 4 +0.160 +0.092 +0.022 -0.048 -0.290
```

bill-orange, 2020-06-16:
Bilinear or UBL?

psavva, 2020-06-16:
I can't access your config files.
Please paste them in here.

Please follow these instructions and provide the output:

Download Marlin bugfix-2.0.x to test with the latest code.
Enable DEBUG_LEVELING_FEATURE and M114_DETAIL and re-flash the firmware.
Issue M111 S247 to enable maximum logging.
Issue G28
Issue G29 V4
Attach the log output into a .TXT file

SagittariusC, 2020-06-16:
> I can't access your config files.
> Please paste them in here.
>
> Please follow these instructions and provide the output:
>
> Download Marlin bugfix-2.0.x to test with the latest code.
> Enable [...]

I will find another way to get you the config files shortly. I'm guessing google drive is being silly? 

As for the debug feature, it takes a lot of memory to turn on, I'm already at 99% capacity. I will try to disable other features in order to make room, but in not 100% sure what I can turn off. 

I will get the latest bug fix 2.0.x version and try to get it all set up.

bill-orange, 2020-06-16:
Zip up the files. Edit your last post and drag and drop them into it.

AnHardt, 2020-06-16:
https://github.com/MarlinFirmware/Marlin/issues/17399#issuecomment-644094485

SagittariusC, 2020-06-16:
> 
> 
> Zip up the files. Edit your last post and drag and drop them into it.

Didn't realize that was possible, thanks. Its edited.
This is using Bilinear, would UBL have better results? will it even work on this printer?

SagittariusC, 2020-06-16:
> 
> 
> [#17399 (comment)](https://github.com/MarlinFirmware/Marlin/issues/17399#issuecomment-644094485)

I see what you mean there. I have the frame a straight as I can get it, and I can print fine without using bed leveling.  But it seems that when I try to use auto bed leveling, it makes bad corrections despite the fact that it prints reasonably even without it.

So is that because like you stated, Bi linear bed leveling can't fix bumps and such? only tilted beds?
Is UBL capable of doing this? Should I be using it instead?

floridaservices, 2020-06-17:
Just personal experiences here; I have found bilinear to be pretty good - but it has its limits.  I think you can change parameters to make it be more agressive but by default it only has so much correction at its disposal.  UBL has a lot more functionality but requires a lot more patience to get going as well.  Have you tried simply upping the probe points?  Bed leveling will take a LOT longer but more data may help.  Also, z probe offset plays a HUGE part in ABL results.  That could be all that is going on since you say manual leveling is giving you better results.  Play with your z offset and see what happens.

SagittariusC, 2020-06-17:
>

` `text' overflowed by 47584 bytes`

Ya as I suspected, I overflow quite a bit... not sure what I can turn off that wont cripple the printer.
But I'm going to try UBL on bugfix 2.0.x

SagittariusC, 2020-06-17:
> Just personal experiences here; I have found bilinear to be pretty good - but it has its limits. I think you can change parameters to make it be more agressive but by default it only has so much [...]

The issue is, I have the bed mostly level, as in I can print without autoleveling, and its "ok". but if I run bed leveling, it throws it right off.

SagittariusC, 2020-06-17:
> ` `text' overflowed by 47584 bytes`
>
> Ya as I suspected, I overflow quite a bit... not sure what I can turn off that wont cripple the printer.
> But I'm going to try UBL on bugfix 2.0.x

Ya, there is no way I'm fitting UBL on this, its way too big.
even after thinning out some stuff, I'm still overflowed by 20714 bytes...
But I could run bi-linear with debugging on, I've attached the print log.
Note: It still is not correcting properly with bugfix 2.0.x, I tested a print after this leveling, and its still far too low on the right side if the bed.
[print_log.txt](https://github.com/MarlinFirmware/Marlin/files/4790320/print_log.txt)

AnHardt, 2020-06-17:
> I see what you mean there.

No. You don't. You are still asking into the wrong direction.

Assuming your bed is level enough to print on your printers bed and getting
```
      0      1      2      3      4
 0 -0.053 +0.053 -0.018 -0.085 -0.327
 1 +0.038 +0.090 -0.015 -0.108 -0.382
 2 +0.012 +0.072 +0.002 -0.067 -0.310
 3 +0.055 +0.110 +0.005 -0.100 -0.377
 4 +0.160 +0.092 +0.022 -0.048 -0.290
```
as the result of probing the bed, it must be the probing what gives wrong results. With a difference of ~0.4mm front/left to rear/right you cant print on the whole surface.

Make a simple test. Switch off any ABL. At the assumed lowest and highest point deploy the probe and manually go down until the **probe** triggers. Measure the high of the **nozzle** (in sheets of paper or coins or coins and papers). If the measured hight differs - you have a geometry problem.
That can be a unsquare frame, but may also be a kable or bowden pulling on the x-sled and tilling it, omitted or added washers, inperfect printed parts, ... . There are literally dozens of ways to mess it up. One day you'll find the reason.

JoePrints, 2020-06-17:
You aren't by chance using an inductive probe on a magnetic bed are you?

floridaservices, 2020-06-17:
So when I was putting together my last printer it was late one night and I was putting the bed on using spacers.  Somehow I managed to put taller spacers on the back , so the whole bed was tilted forward but not enough to be noticeable right away.  No amount of z-offset correction or leveling correction would make it work.  Everything printed too high no matter what I did.  

If you can print manually then your z-offset is probably off.  Z-offset sets the probe's spacing between it and the bed- nothing else does this.  If your z-offset is .4 bump it up to 1.2, it should make a difference in some way.   If it doesn't then something is mechanically out of spec.

Based on your own leveling data, no matter what you say about manual leveling, your bed is not level

SagittariusC, 2020-06-18:
> > I see what you mean there.
> 
> No. You don't. You are still asking into the wrong direction.
> 
> Assuming your bed is level enough to print on your printers bed and getting
> 
> ```
>       0      1      2      3      4
>  0 -0.053 +0.053 -0.018 -0.085 -0.327
>  1 +0.038 +0.090 -0.015 -0.108 -0.382
>  2 +0.012 +0.072 +0.002 -0.067 -0.310
>  3 +0.055 +0.110 +0.005 -0.100 -0.377
>  4 +0.160 +0.092 +0.022 -0.048 -0.290
> ```
> 
> as the result of probing the bed, it must be the probing what gives wrong results. With a difference of ~0.4mm front/left to rear/right you can't print on the whole surface.
> 
> Make a simple test. Switch off any ABL. At the assumed lowest and highest point deploy the probe and manually go down until the **probe** triggers. Measure the high of the **nozzle** to the bed (in sheets of paper or coins or coins and papers). If the measured hight differs - you have a geometry problem.
> That can be a unsquare frame, but may also be a kable or bowden pulling on the x-sled and tilling it, omitted or added washers, inperfect printed parts, ... . There are literally dozens of ways to mess it up. One day you'll find the reason.
> 
> There is absolutely **NO** way to get **such** systematic errors by messing up a configuration.

I've already mentioned that I can level this thing by hand and get resonably even first layers. Even with a test print I designed that just prints a large square outline. That's why is suspected ABL is doing something weird, and why I'm asking here about it. I feel as though ABL is trying to overcorrect or something.
Could it he that the Z probe is just trash and giving bad results? It's a inductive probe, maybe I should replace it with something?

SagittariusC, 2020-06-18:
> You aren't by chance using an inductive probe on a magnetic bed are you?

No, inductive probe on aluminum bed, no magnetic layer.

SagittariusC, 2020-06-18:
Ok so, from what I think I'm getting out of all the resonces so far is that Bi-linear leveling cannot correct for the warping of the bed, only tilt? 
So it measures a mesh but only creates a simple plane that it uses for correction, and not a mesh where in corrects to follow the warping of the plate as well? 
I assume UBL does that? Which won't fit on my printer unless I remove a bunch of stuff, so that's not an option.

Seems as though the only options I have are to get a new, less warped bed. Or find a way to fit UBL on it. Or just write off ABL on my printer all together as not being useful and just doing it by hand.

floridaservices, 2020-06-18:
Everyone has tried to offer you things to try, have you done anything that has been suggested?  Bilinear bed leveling works just fine on a warped wonky bed; if I can make a mk2b bed useful your aluminum bed can work as well.  If abl was not useful there would be a lot less ender 3s in use.   Fix your z offset, please, then tell us that's not it  or wow! ABL is amazing!

SagittariusC, 2020-06-18:
> Everyone has tried to offer you things to try, have you done anything that has been suggested? Bilinear bed leveling works just fine on a warped wonky bed; if I can make a mk2b bed useful your [...]

Yes, I've tried most of the suggestions. 
I've adjusted my z offset to various levels and tested, I've upgraded to the latest bugfix 2.0.x build and retested everything, I've tried to straighten my frame by measuring distances and tightening everything, I've tried using more probe points dying bed leveling. I've enabled debugging and posted a log of the ABL procedure. 
So far it's still not working correctly, so I'm starting to wonder if my frame is just badly warped somewhere. 

I appreciate everyone giving suggestions.

floridaservices, 2020-06-19:
So lets go back to basics, it has to be something geometrically off.  Since you say it can print with manual leveling - have you printed successfully at each corner and center of the bed?  I can print cubes all day long on a poorly set up printer but move that cube from the center to a corner and it might be a different story.    

What is your frame material?  If it's acrylic that's going to be one of your issues - get rid of it asap.  

Y Axis -------Most important part of the printer in my opinion
  -What is your Y axis?  If it's threaded rod it MUST be as square as possible , and all four feet must touch the surface the printer sits on.  Pads and such on the feet can make this hard to determine so take any pads off to check this.  
  -The heatbed support - how is it mounted?  Depending how the bearings are mounted this can affect bed height, since you are seeing a different from right to left

When I was trying to  square up my prusa clone I used a machinist square to line up the Y axis with the X axis.  You might want to do the same.  If your bed has gridlines you can use these to check that your X axis is perpendicular with Y axis also.  I used the square like this: short side on the bed, long side touching both x rods.  If it won't touch all three points fully, everywhere  you place it on the bed, you have a geometry issue.  

You say you are using an inductive probe - what kind?  Have you verified that it triggers the same way every time?  They tend to be pretty reliable in my experience, but it is quite possible to fry them if you connect them the wrong way to your board.  I have fried a pinda v1 this way- the trigger LED would still behave normally when the probe was supposed to trigger, but the sensor would never register this.  So if you have ever at any time had the connector backwards while the printer was powered up ---- that could be your problem as well.

bill-orange, 2020-06-19:
Getting the X-axis square to the Y-Axis is indeed crucial and not easily accomplished.  The idea is to get the  two axis square before leveling the bed.

Regarding the acrylic frame, mine came with the acrylic frame and I have not replaced it.  I have not found it to be a problem if reinforced.  Here is the stl for a measuring rod you can try.  The design is not ideal.  It is a but too flexible when printed in PLA, but it does work.

https://1drv.ms/u/s!AmXKqAwyCrbxiOEbJ_0TirBm0zWVHg?e=h8g04N

SagittariusC, 2020-06-20:
> So lets go back to basics, it has to be something geometrically off. Since you say it can print with manual leveling - have you printed successfully at each corner and center of the bed? I can print [...]

For the leveling check, its a tool I made that is a big square frame that prints closer to the perimeter of the bed, I can get it to print pretty even on that.

Yes the frame is acrylic, but its not just a simple task to "get rid of it", that would be a huge undertaking as the whole frame is acrylic, and its really not worth replacing on this printer.

There are threaded rods along the Y axis, I have tried my best to square them using measurements. When I get time, I will find a square and check everything.

The bed carriage is mounted on 2 smooth rods, with a 3 bearing setup (one on left, 2 on right). The bed is mounted to that carriage using bolts and springs.

The Z probe is a Zhongde PL-08N. I have ran a repeatability test before on it and it did not give me alarming results, just a small margin of error.

Its sounding like getting everything square is the best thing to try. I will attempt that soon and get back to you.

Thanks

AnHardt, 2020-06-20:
> I have ran a repeatability test before on it and it did not give me alarming results, just a small margin of error.

That does not tell you anything about the geometry - just about repeatability of the probing process.
Did you try the test i suggested above?

The probe is the eye of the leveling system. If it looks thru Varifocals it will determine different (bed-nozzle) distances depending on if it is looking thru the high or low diopter part and does adjust the nozzle to what it sees. The repeatability test is always looking true the same part of the lense (testing at the same spot) - so will see always the same (but wrong) result.

SagittariusC, 2020-06-20:
> That does not tell you anything about the geometry - just about repeatability of the probing process.
> Did you try the test i suggested above?
>
> The probe is the eye of the leveling system. If it [...]

Mmmm the repeatability test is as it's called, it checking if the measured Z point is repeatable in the same spot, that's the whole point. 
It's just to check the reliability of the probe. Best I can do is find a way to have it do that repeatability test in multiple locations on the bed, but i don't think that's nessisary.

boelle, 2020-06-21:
@SagittariusC still an issue?

SagittariusC, 2020-06-21:
> @SagittariusC still an issue?

I've been busy, but soon I'll be trying to square everything up and will re-test ABL. Because it's sounding like a warped frame is the likely cause.

boelle, 2020-06-29:
what did the re-test tell?

SagittariusC, 2020-07-10:
> what did the re-test tell?

At this point, bed leveling still does not perform as expected, possibly because of a medeocre frame. 
Will be testing further at a later date.

sjasonsmith, 2020-07-18:
@SagittariusC have you tried using `M48` to determine how consistent your probe is when repeatedly probing in the same place?

closed 2020-08-24 by github-actions[bot]