MarlinFirmware/Marlin #18323: [BUG] (MKS_Robin_E3 does not work with MKS_MINI_12864)

cr20-123, 2020-06-16:
### Bug Description

When using the MKS_Robin_E3 with the MKS_MINI_12864 it does not work properly. The display goes blank after a while and random black bars pop up when going through the menus. There are also black bars that cover the menu to select the file you want to print which ends up covering the file names.

### My Configurations

[Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/4785319/Config.Files.zip)

### Steps to Reproduce

Select MKS_Robin_E3 and MKS_MINI_12864 and flash the firmware onto the printer.

**Expected behavior:** [What you expect to happen]
 For the display output to be clear and the display and SD card to function properly

**Actual behavior:** [What actually happens]

The display randomly times out and goes blank and the menu to select the file to print has horizontal black bars. Also, the SD card needs to be in the printer for it to boot.

#### Additional Information

When I put #define FORCE_SOFT_SPI in the pins file for the board, the display starts working properly, but the SD card completely stops working. With that option enabled and I insert the SD card the display freezes and it becomes unusable. The same thing happens if the SD card is in the printer when it powers on, the display freezes and becomes unusable. If I power it on without the SD in it, the menus look fine and there are no more black bars randomly popping up. I have tried pretty much every SD related setting in configuration.h and configuration_adv.h. I have also tried different baud rates, changing the display update delay described [here](https://marlinfw.org/docs/basics/troubleshooting.html). Sorry in advance if I am missing anything, this is my first time reporting an issue.

boelle, 2020-06-21:
@cr20-123 still an issue?

cr20-123, 2020-06-21:
Hi, yes it is still a issue

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

cr20-123, 2020-06-23:
Hi the issue still persits in the latest bugfix.

thinkyhead, 2020-06-23:
Have you tried messing with the display timing? For example…
```cpp
  #define ST7920_DELAY_1 DELAY_NS(125)
  #define ST7920_DELAY_2 DELAY_NS(125)
  #define ST7920_DELAY_3 DELAY_NS(125)
```

cr20-123, 2020-06-23:
I have tried messing with them, I started off here from your Troubleshooting Tips guide and went up in intervals of 100. Should I try going up in intervals of 10 instead and starting from the values in the pins file for the board?
#define ST7920_DELAY_1 DELAY_NS(200) // After CLK LOW
#define ST7920_DELAY_2 DELAY_NS(400) // After DAT
#define ST7920_DELAY_3 DELAY_NS(200) // After CLK HIGH

cr20-123, 2020-06-24:
I’m not sure if this helps at all, but the screen goes blank the most when accessing the SD card and scrolling through files. I have tried various SD cards and for each one I tried it has the lines across the screen and when scrolling the display goes blank. The screen goes blank randomly as well, but it is the most repeatable while accessing the SD card.

thinkyhead, 2020-06-24:
Has it helped at all to use shorter / different cables or isolate the display from the electronics?

cr20-123, 2020-06-24:
I have tried using jumpers connecting the pins in use from the display to the board. I have also tried it out of the printer with just power connected to the board over usb and the 24v input on the board. The display cables are only about 200mm long and the only other wires they are close to when everything is connected in the printer are the thermistor wires and endstop wires.

cr20-123, 2020-06-25:
Is it normal for a 32bit board to require a SD card to be in it for it to boot fully? With the SD in the board while it is booting it goes to the status screen, with the SD out when it’s booting it stays on the boot screen until a SD is inserted. After the SD is inserted it goes to the status screen. Another thought, since the SD slot and the screen share some of the same pins could it be possible that the display is getting confused from the data that is being sent to and from the SD card?

cr20-123, 2020-07-04:
The board is no longer in working condition so I can not do any further testing.

closed 2020-07-04 by cr20-123