MarlinFirmware/Marlin #18325: Enabling the probe port on the SKR V1.4 Turbo

Portzal, 2020-06-16:
There is a change needed in conditionals_LCD.h  to enable the probe port on the SKR V1.4 Turbo board. Smarter people than me have solved this problem, I am just bringing it up. I read It in a thread somewhere and have been doing it since. 
I did a build in the current version Bugfix 2.0.x and still had to make a change in conditionals_LCD.h 
The change is as follows. 

Marlin 
           src 
                  inc 
                        conditionals_LCD.h
Change  from 
2.0.5.3
#if Z_HOME_DIR < 0 && !HAS_CUSTOM _PROBE_PIN

Bugfix 2.0.x
#if Z_HOME_DIR < 0 && (!HAS_CUSTOM _PROBE_PIN || ENABLED(USE_PROBE_FOR_Z_HOMING))

Change To
2.0.5.3
#if Z_HOME_DIR < 0 //&& !HAS_CUSTOM _PROBE_PIN

Bugfix 2.0.x
#if Z_HOME_DIR < 0 //&& (!HAS_CUSTOM _PROBE_PIN || ENABLED(USE_PROBE_FOR_Z_HOMING))

Configuration.h
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

#define Z_MIN_PROBE _PIN P0_10

jiridoko, 2020-06-17:
I am having the same issue on skr 1.4 turbo. I can do auto home and bed levelling from the LCD and by sending the gcode over serial but I can't do that in the sd card gcode when printing. It is really inconvenient. I must say Marlin could use a bit of refactoring. It is hard to find anything in the code due to all the ifdefs.

zonalimitatore, 2020-06-20:
Change To
2.0.5.3
#if Z_HOME_DIR < 0 //&& !HAS_CUSTOM** _PROBE_PIN**

or
#if Z_HOME_DIR < 0 //&& !HAS_CUSTOM**_PROBE_PIN**

There is a space

Barconero, 2020-06-20:
> Change To
`#if Z_HOME_DIR < 0 //&& (!HAS_CUSTOM_PROBE_PIN || ENABLED(USE_PROBE_FOR_Z_HOMING))`

it is misstake.
you just have to use USE_PROBE_FOR_Z_HOMING

Portzal, 2020-06-20:
@zonalimitatore no space. I have fixed it up.

boelle, 2020-06-21:
@Portzal still an issue?

Portzal, 2020-06-21:
@Barconero @boelle Sorry gentlemen, I stand corrected. I did not notice the USE_PROBE_FOR_Z_HOMING option when moving my settings from 2.0.5.3 to Bugfix 2.0.x. I have made the changes to my firmware and tested the new settings and it works perfectly.

closed 2020-06-21 by Portzal