MarlinFirmware/Marlin #18328: [FR] add TMC 2226 to Marlin drivers

hackebike, 2020-06-16:
There is a new driver from BTT TMC 2226 v1.0
It would be great to add it into Marlin drivers supported list.

Evg33, 2020-06-22:
I use TMC2225 as TMC2208 with #define *_RSENSE  0.11.
I think it will be similar with them.

teemuatlut, 2020-06-22:
Yes. In time.
For now I think you can define them as 2209. I'm getting a few myself some time during the next few weeks to test and verify.

fred099, 2020-07-06:
Adding data sheet and calc sheet for TMC 2226.

[TMC2226_Datasheet_V106.pdf](https://github.com/MarlinFirmware/Marlin/files/4877377/TMC2226_Datasheet_V106.pdf)
[TMC2226_Calculations.xlsx](https://github.com/MarlinFirmware/Marlin/files/4877390/TMC2226_Calculations.xlsx)

hackebike, 2020-08-13:
I am testing TMC 2226, getting X to work with sensitivity value of 65, but cannot find one for Y and Z.

@Evg33 where to you change the value #define *_RSENSE 0.15 ?

btw, I use #define X_DRIVER_TYPE  TMC2209 and it works for TMC2226.

siddacious, 2020-09-08:
Has anyone else started on this?

siddacious, 2020-09-09:
> Has anyone else started on this?

If not, I'm interested in taking a shot at it.

ellensp, 2020-09-09:
Fairly sure the tmc2226 is a tmc2209 in a HTSSOP28 package. ie firmware wise they are identical.

sjasonsmith, 2020-09-09:
Maybe a comment should just be added to the Configuration.h file.
I wish TMC would name their packages using suffixes like everything else, instead of unique device numbers.

ellensp, 2020-09-09:
or a simple #ifdef tmc2226 #define tmc2209 so the user feels good about setting tmc2226 in their config.

thisiskeithb, 2020-09-09:
I use TMC226s configured as 2209s and they seem to work fine.

Same deal with TMC2225s configured as TMC2208s 🤷‍♂️

siddacious, 2020-09-09:
Huh. It seems strange that they would go to all that trouble of adding another "product" for a simple package change, but I can't argue with results!

@ellensp do you think it's worth the effort to dig deep in the datasheets or otherwise verify that there aren't any other notable differences?

ellensp, 2020-09-09:
same datasheet https://www.mev-elektronik.com/files/MEV%20Elektronik%20Service/produkt-downloads/motion-control/TMC2209_datasheet.pdf

thisiskeithb, 2020-09-09:
I believe the TMC2226 and TMC2225 are cheaper to buy/produce.

sjasonsmith, 2020-09-09:
BGAs tend to have more manufacturing complexity. Switching to non-BGA drivers may make their assembly process cheaper with a lower failure rate.

siddacious, 2020-09-09:
@ellensp huh! I guess I've been looking at the "new" datasheet:
https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2226_Datasheet_V106.pdf

Looking at prices, they haven't listed a T&R price for the 2226 yet but currently at DigiKey it's $1.56@5K vs $1.59@6k for a reel of the 2209; I suspect they're trying to suss out the popularity before finalizing their pricing, but yes it does seem like it will be a bit cheaper.

@sjasonsmith what chips were you referring to re:BGA? I know they have some of their higher end chips like the EtherCAT controllers in BGA, but most of the discrete stepper controllers are SOPs or QFNs, and QFNs are not going to have the issues associated with BGAs.

Anywhoo, it looks like ya'll have this handled. Feel free to @ me for testing or for pulling the trigger on a commit. Once I've got all axes of my printer performing nominally I can also test and submit a PR if no one has done so by then

hackebike, 2020-09-09:
@siddacious did you had a chance to test sensorless?
I have BTT TMC 2226 working with sensitivity 65 on X axis, but I cannot find a value for Y and Z.
I went through all 256 values!
Testing on a BTT GTR and I has sensorless working on all axis with BTT TMC 2229.

sjasonsmith, 2020-09-09:
@siddacious yeah you are right, they aren't BGAs. That was a naive comment based on not seeing pins when I glance at the chips.

siddacious, 2020-09-09:
@hackebike I made the unwise choice of replacing my  extruder and main board at the same time so I only just now have 2208s working in legacy mode (I assume, haven't verified). I have the 2226's on hand, just want to get everything else working before throwing in another variable.

I'll let you know when I switch over

siddacious, 2020-09-11:
@ellensp I was curious about the 2226 vs 2209, so I asked Trinamic and this was their response:

> Thanks for your question.  The TMC2226 and the TMC2209 are the same device in different packages.  The TMC2226 was initially released to target the Chinese market, but we realized later that we could service the broad market with this design as well and had received requests for a different package option, so the TMC2209 was created.  But these parts are the same silicon design, and the only difference is in the package options.

From this it appears the chicken came before the egg!

cobaltduff, 2020-09-26:
Do we have to make more changes in Marlin for the TMC2226? change rsens to 0.15 or leave it at 0.11?

thisiskeithb, 2020-11-06:
Closing since https://github.com/MarlinFirmware/Marlin/pull/20002 added a comment to the TMC section explaining TMC2225 = TMC2208 and TMC2226 = TMC2209:

https://github.com/MarlinFirmware/Marlin/blob/c574bcce880e769ae3851304da2a1b1ee19466ed/Marlin/Configuration.h#L677

closed 2020-11-06 by thisiskeithb