MarlinFirmware/Marlin #18333: Issue compiling when enabling Neopixel #define NEOPIXEL_LED

SilentZen, 2020-06-17:
Marlin\src\gcode\sd\M1001.cpp: In static member function 'static void GcodeSuite::M1001()':
Marlin\src\gcode\sd\M1001.cpp:93:85: error: 'wait_for_user_response' was not declared in this scope
       wait_for_user_response(1000UL * TERN(HAS_LCD_MENU, PE_LEDS_COMPLETED_TIME, 30));
                                                                                     ^
*** [.pio\build\mega2560\src\src\gcode\sd\M1001.cpp.o] Error 1

ellensp, 2020-06-17:
You NEED to attach your configuration files, as requested.

ellensp, 2020-06-17:
wasting time recreating test conditions.
Needs SDSUPPORT, and a display I used REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and NEOPIXEL_LED. before relevant part of M1001 is compiled in.

ellensp, 2020-06-17:
Compiles fine on Marlin 2.0.5.3 and bugbix 2.0.x

SilentZen, 2020-06-17:
Please find attached, sorry about not attaching.
Thanks.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4794377/Configuration.zip)

SilentZen, 2020-06-17:
[Config Files..zip](https://github.com/MarlinFirmware/Marlin/files/4794392/Config.Files.zip)
Sorry about wasting your time. I have attached config files that cause error.

thinkyhead, 2020-06-17:
See if this helps:
```diff
+ #if HAS_RESUME_CONTINUE
    wait_for_user_response(1000UL * TERN(HAS_LCD_MENU, PE_LEDS_COMPLETED_TIME, 30));
+ #endif
```

thinkyhead, 2020-06-17:
```
#error "You are using an old Configuration.h file, update it before building Marlin."
```

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

closed 2020-06-17 by thinkyhead

ellensp, 2020-06-18:
Seems to be non standard marlin, has Temperature sensors 995. After updating config incompatibilities, provided Configuration.h compiles without issues.