MarlinFirmware/Marlin #18336: (SerialException): Write timeout when M104, M140, M303

longlh, 2020-06-17:
Everytime I attempt to set temperature of hotend/bed (M104, M140) or PID tuning (M303), I get an error message: `[ERROR] Can't write to printer (disconnected?) (SerialException): Write timeout`

[image]

My configuration: https://github.com/longlh/Marlin/commit/88b702612c6abc42709f80c1fee4b67c3edfcf50
Board: FYSETC AIO II

boelle, 2020-06-21:
@longlh still an issue?

boelle, 2020-06-22:
do you use latest version? https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

updated 10 hours ago

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle