MarlinFirmware/Marlin #18337: Controller hangs when prinitng

LeererMann, 2020-06-17:
Marlin 2.0.5.3
Creality CR-10S
Stock PCB Atmega 2560
Swiss Made All Metal, BLTouch
SD printing

After completing a print the hotend would park and the controller would hang. This was repeatable. Also, when I Released the SD to add some files and then Attached the SD the controller would hang. When this occurred it looked like it was attempting to repaint the word Attach in the display to Release. I would see pixelation occurring on that line only then it would stop and be hung. This also was repeatable. Power cycle was the fix.

Expectation is for it not to hang...

I had been configuring, adding and removing features, compiling and installing 2.0.5.3 (from 1.1.8) for a couple days and was done when this hanging started occurring. Got to be a real inconvenience.

I got to thinking what I might have done, configured in or out that could cause these hangs. I started to suspect SDCARD_SORT_ALPHA. All my filenames are long, some over 40 characters so to see the full filename I have scrolling enabled. When these long filenames are scrolling to see the full filename, response to scrolling up and down was lagging and painfully slow. That's what made this suspect to me.

So I disabled SDCARD_SORT_ALPHA and haven't had a hang since
[Marlin2.0.5.3ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/4791719/Marlin2.0.5.3ConfigFiles.zip)

boelle, 2020-06-21:
@LeererMann still an issue?

LeererMann, 2020-06-21:
No because I disabled alpha sorting on the SD but yeah if I enable alpha sorting on the SD card so I can't use sorting on the SD card. It didn't fix itself.

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

Joe-Breznai, 2020-06-25:
Mac form TH3D suggested I contact you re this.  I have have been having the same issues.  Following is what I have sent to them. I am using their on-line complier for U2-R1-B3 on the EZBoard they sell.
 	Joe Breznai 

Jun 24, 2020, 2:09 PM CDT 
Mac,
 
OK, I have been driving myself nuts trying to sort out why-sometimes, my prints freeze before lifting the print head and presenting.  As I stated in prior emails, sometimes the machine freezes at 99% and just sits there with all heaters running and steppers active. I would have to re boot the printer to do anything else when this happened.  I also followed you initial suggestion about changes the printer settings in the setup code re the shut down sequence with no positive effect.
 
After six pages of notes, tracking which SD card is used, all of varying capacity, different brands, formatting under windows or using a separate format tool; I came to the incorrect conclusion that if the files on the card exceed 140 Meg, that this was causing the problem.  But when I did this, I was using short filenames with multiple copies of the same –known good- gcode file to increase the total file size on the SD card.  I just changed the file name (file1, file2, file3, etc.).
 
I was a day away from emailing you with this result, when I had a failure on a print using an SD card that only had 14 files on it and well under a total file size of 140MB.
 
This led me to wonder if the total file NAME length of all files had anything to do with this problem. I am now somewhat reasonably sure that this is the limiting factor. I seem to remember that the 8.3 file structure is limited to something like 255 characters, which includes the path name.  But; I thought that this limit was a per-file limit.   A as these files were/are being printed directly from the SD card inserted in the printer, which has your EZboard and software, I initially dismissed this as a cause.
 
Then it happed again on an SD card that had 11 files comprising 41.2Mb on it. The prior print job on this card had 10 files and printed a file correctly.  Then I added an additional file to print and it froze at the end of that print.  Looking at the files on the SD card I have come to the tentative conclusion that if the TOTAL count of charters comprising ALL the file names on said card is GREATER than 230 charters,( including spaces, and the “.gcode” suffix)  that the chance that the machine will freeze is increased.  Renaming one other file on that SD card with a shorter file name ( NOT the one that had previously failed), it printed correctly.
 
Note that Windows and DOS (CMD-DIR) do not reflect the size of the file names on a card.  When adjusting the file namesonly to test, the File Size reported using Windows or the CMD line for the SD card stay the same.  I tried incrementally increasing the Total Character count on the SD card one by one by adding a character to see if I could repeatedly invoke this problem, but, not always does it happen.  I made sure I was Releasing the card before reinserting just to make sure, but to no avail.  Sometime all was OK, other times it would freeze, but always with a longer total-card file name charter count than the previous test.  This doesn’t seem to happen if total file name characters’ total fewer than 230, but I have had total card character count as high as 295 that printed and presented OK. Frustrating. But when it does freeze, it always due to a larger number of the total character’s comprising all the file names present on that card.  Re naming any one file on the card to a shorter name will let the print finish and present.
 
In short: for a long print that I am going to run overnight, or unattended, I limit the files on the SD card and have not experienced this freezing when doing so.  Seems like anything over 230 TOTAL characters of all file names present on the card MAY trigger the problem regardless of the total size of the files comprising that contents on the SD card, no matter which file is printed.   It sure looks like the total File NAME size is the contributing factor.
 
Sorry for not being able to further sort this out, but I am now officially brain-dead. 
 
I hope that maybe it is a clue you can use to understand this problem when it happens to others.
 
Joe
 
 
Sent from Mail for Windows 10
 
From: Mac McMillen
Sent: Thursday, May 7, 2020 8:57 AM
To: lenape@msn.com
Subject: Re: Tech Support - Joe Breznai - Order #118633 EZABL™ Pro
 
Hey there Joe,
 
Let me know how things go with the script change and the smaller memory card when you get a chance.

Ticket: https://th3dstudio.freshdesk.com/helpdesk/tickets/27998
 
~Mac McMillen
 
3D Printer Support Tech
 
 
 
TH3D Studio, LLC
__________________________________________________________
 
Check out our communities here:http://community.th3dstudio.com
On Wed, May 6 at 3:27 PM, Joe Breznai <lenape@msn.com> wrote: 
Thanks  again Mac,
Checked the “Machine Settings” and actual GCode generated under both flavors Of Cure that had has this has happen .  4.5  and 4.6.1.
In both versions of Cura the ending Machine Settings are :
 
“G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positionning
G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M140 S0 ;Turn-off bed
M104 S0 ;Turn-off nozel
M117 Print Completed”
 
[... cut ...]
The way in the manual is not wrong, it is another way of setting the probes Z offset. It is just a little more complicated than the way I have above. 

For the second issue, you would need to go into your slicers end script and reverse the order that the bed and the hotend are turned off. 

The default for Cura is 
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed
 
Try the one I have below for those two lines. 
M140 S0 ;Turn-off bed
M104 S0 ;Turn-off hotend

Ticket: https://th3dstudio.freshdesk.com/helpdesk/tickets/27998
 
~Mac McMillen
 
3D Printer Support Tech
 
 
 
TH3D Studio, LLC
__________________________________________________________

thinkyhead, 2020-07-26:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

closed 2020-09-01 by github-actions[bot]