MarlinFirmware/Marlin #18338: [FR] Chamber Temp on LCD2004 missed!

karabas2011, 2020-06-17:
There no any displaying of chamber temp on status screen on displays lcd2004  (f.e. rrd smart controller)
It had implemented only for graphic displays.

ellensp, 2020-06-17:
Just where on the 20x4 char display is there room for a chamber temp?
Each temperature displayed uses 10 characters. A icon + 3 char current temp + / + 3 char set temp + degrees symbol + space separator.  
So with a hotend and a heated bed, you have no space to put any further temp displays.

karabas2011, 2020-06-17:
But two extruders + bed you can see it works. So at least 1 extruder + bed + chamber = no problem
3d temp displays insteed of X coordinate as I remember

karabas2011, 2020-06-25:
Dirty minimal hack for H_E0/H_BED/H_CHAMBER 

```
FORCE_INLINE void _draw_heater_status(const heater_ind_t heater, const char prefix, const bool blink) {
  
  #if HAS_HEATED_BED  
     const bool isBed = heater < 0;
   #endif 

  float t1=0,t2=0;
  if (heater == -2) // H_CHAMBER
        t1 = thermalManager.degChamber(), t2 = thermalManager.degTargetChamber();

  if (heater == -1)  //H_BED
        t1 = thermalManager.degBed(), t2 = thermalManager.degTargetBed();

  if (heater >= 0)   //H_E0, 1 2 3...
         t1 = thermalManager.degHotend(heater), t2 = thermalManager.degTargetHotend(heater);
//----3dq^^^
  
  if (prefix >= 0) lcd_put_wchar(prefix);
```
...
```
      #else // LCD_WIDTH >= 20

        lcd_moveto(0, 1);

        // If the first line has two extruder temps,
        // show more temperatures on the next line
        
        #if HOTENDS > 2 || (HOTENDS > 1 && HAS_HEATED_BED) || (HOTENDS < 2 && HAS_HEATED_BED && HAS_HEATED_CHAMBER) 
          #if  (HOTENDS < 2 && HAS_HEATED_BED && HAS_HEATED_CHAMBER )   //3dq
            _draw_heater_status(H_CHAMBER, LCD_STR_THERMOMETER[0], blink);
            lcd_moveto(10, 1);          
          #endif
          #if HOTENDS > 2
            _draw_heater_status(H_E2, LCD_STR_THERMOMETER[0], blink);
            lcd_moveto(10, 1);
          #endif

          _draw_bed_status(blink);

```