MarlinFirmware/Marlin #18341: [BUG] SKR 1.4 Filament Runout on Pin 1.26 & 1.25 non working

elmolinger, 2020-06-17:
### Bug Description

Filament Ronout Pin 1.26 on SKR1.4 (LPC1768) isnt working. After switching to 1.25 Pin in pins_BTT_SKR_V1_4.h i got m119 Signal (open/ triggerd) on runout sensor. But still non working M600 command during Print via LCD SD_Card (TFT24 v1.1)

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4795182/Marlin.zip)

### Steps to Reproduce

Connect a Prusa Filament sensor to Pin 1.26 on SKR 1.4. Make changes in configuration.h  and configuration_adv.h and start a print from LCD SD_Card. Cut the Filament during Print to simulate a runout.

[What you expect to happen]
Working Filament Sensor Signal on Pin 1.26
Within Printing and filament runout -> Advanced Pause ->manually Refill Filament -> restart printing

[What actually happens]
On Pin 1.26
After connecting the Out Pin to 1.26 on the SKR the Voltage stays LOW. M119 keeps Open (equals to settings in configuration.h #define FIL_RUNOUT_INVERTING false/true)
On 1.25
Normal Signal behaviour with m119 if I triggering the Sensor manually.
But during Print -> Filament runs out -> Print continues

#### Additional Information
Sensor is tested and working
If I only disconnect the Sensor Out Pin wire and meassure the voltage (gnd -> Out Pin on the Sensor) i get low if the sensor is triggerd and high if not. 

Best regards

thisiskeithb, 2020-06-17:
Did you cut/remove the diag pin for your extruder driver?

elmolinger, 2020-06-18:
> Did you cut/remove the diag pin for your extruder driver?

No, I did not. Can you explain this to me?
I have tmc2209 driver and I think the diag pin can stay connected.

ellensp, 2020-06-18:
diag pin on e0 is electrically connected to 1.26 and diag pin of e1 is electrically connected to 1.25.

You cannot use these pins if you have a stepper driver with a diag pin plugged in. (unless you cut off the pins)
 
You can test this by using 1.25 with the E1 stepper driver removed.

thisiskeithb, 2020-06-18:
[BigTreeTech’s TMC2209 documentation](https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/TMC2209%20problems%20solving.pdf) calls out the need to remove the diag pin if you need to use real endstops, but the reasoning is the same for a filament runout sensor as mentioned above.

elmolinger, 2020-06-18:
Ok, that's why I got I0 on 1.25.
I searched for 4days to find a solution by myself. Nevermind.

Thank you for your help!

In case of the non working m600:
If I connect the runout sensor to 1.25 and change the configuration of E0DET to 1.25 in the SKR1.4 pins.h -> M119 is working fine. So if I didnt use E1 I can leave the diag pin E0 connected?

But during print it doesn't trigger m600 anyway.

I need some more time to test changing m600 to m600 t0 in the filament runout configuration. Maybe that's the solution.

Best regards

elmolinger, 2020-06-18:
After changing #define FILAMENT_RUNOUT_SCRIPT "M600" to "...M600 T0" the runout script seems to work. Extruder goes into parking position and unload the filament but the continue prompt on the LCD doesnt show up. Acording to https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/758 this appears only in TFT Mode. The Marlin Mode should bring up this Action Promts and Filament runout runs fine for now.

Sorry for the Support Thread I really thought its a bug in Marlin!

Best regards.

closed 2020-06-18 by elmolinger