MarlinFirmware/Marlin #18343: bugfix-2.0.x.031f308 won't compile with display selected

ruggb, 2020-06-18:
Bug Description
When I compile bugfix-2.0.x.031f308 I get this error;

avr-gcc: error: CreateProcess: No such file or directory

previous build bugfix-2.0.x.18289 compiles fine

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4796205/Marlin.zip)

### Steps to Reproduce
compile with Arduino IDE using my configs

**Expected behavior:** [What you expect to happen]
Compile complete

**Actual behavior:** [What actually happens]
avr-gcc: error: CreateProcess: No such file or directory

#### Additional Information

If I comment out my  LCD selection it compiles without the error.
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

ellensp, 2020-06-18:
mate please stop cross positing!
I have already answered this on https://reprap.org/forum/read.php?415,873891

ruggb, 2020-06-18:
so you are saying it is not a bug if it doesn't compile with a defined display but does without a display defined. OK how do I make it compile in Windows. This is the first time this problem has occured.

ellensp, 2020-06-18:
Re read the reply! the majority of it is how to get around the problem. No way to fix it. It is a windows limitation. Complain to Microsoft.  As Marlin gets larger and more things are added to the link command this issue will happen more and more often. Until it cannot be compiled on windows at all.

ruggb, 2020-06-18:
OK, it compiles now.

closed 2020-06-18 by ruggb