MarlinFirmware/Marlin #18345: [BUG] (SKR E3 mini 1.2  compilation error - HOME BUMMP)

spooky79, 2020-06-18:
Marlin\src\module\motion.cpp:77:71: error: 'X_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin\src\module\motion.cpp:77:71: note: suggested alternative: 'HOMING_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin\src\module\motion.cpp:77:80: error: 'Y_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin\src\module\motion.cpp:77:80: note: suggested alternative: 'HOMING_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
compilation error - HOME BUMMP , files without modification outside of platformio (board type)

Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin\src\module\motion.cpp:77:89: error: 'Z_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin\src\module\motion.cpp:77:89: note: suggested alternative: 'HOMING_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\module\motion.cpp.o] Error 1
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4797317/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4797318/Configuration_adv.txt)

ellensp, 2020-06-18:
Your configuration files and source are out of sync
your Configuration_adv.h has 
`#define HOMING_BUMP_MM      { 5, 5, 2 }       // (mm) Backoff from endstops after first bump
`
when this version of code needs
```
#define X_HOME_BUMP_MM 5
#define Y_HOME_BUMP_MM 5
#define Z_HOME_BUMP_MM 2

```
This has been asked and answered hundreds of times already..

spooky79, 2020-06-18:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp: In member function 'bool Adafruit_I2CDevice::read(uint8_t*, size_t, bool)':
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:160:79: error: no matching function for call to 'TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)'
   size_t recv = _wire->requestFrom((uint8_t)_addr, (uint8_t)len, (uint8_t)stop);
                                                                               ^
In file included from C:\Users\Fg356xBfn\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/Wire.h:42:0,
                 from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_I2CDevice.h:1,
                 from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:1:
C:\Users\Fg356xBfn\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note: candidate: uint8 WireBase::requestFrom(uint8, int)
     uint8 requestFrom(uint8, int);
           ^~~~~~~~~~~
C:\Users\Fg356xBfn\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note:   candidate expects 2 arguments, 3 provided
C:\Users\Fg356xBfn\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note: candidate: uint8 WireBase::requestFrom(int, int)
     uint8 requestFrom(int, int);
           ^~~~~~~~~~~
C:\Users\Fg356xBfn\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note:   candidate expects 2 arguments, 3 provided
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp: In member function 'void Adafruit_SPIDevice::transfer(uint8_t*, size_t)':
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:99:31: error: no matching function for call to 'SPIClass::transfer(uint8_t*&, size_t&)'
     _spi->transfer(buffer, len);
                               ^
In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:1:0,
                 from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:1:
Marlin\src\HAL\STM32F1/SPI.h:268:11: note: candidate: uint8_t SPIClass::transfer(uint8_t) const
   uint8_t transfer(uint8_t data) const;
           ^~~~~~~~
Marlin\src\HAL\STM32F1/SPI.h:268:11: note:   candidate expects 1 argument, 2 provided
*** [.pio\build\STM32F103RC_btt_512K\lib40b\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp.o] Error 1
In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:17:0:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.h:84:32: warning: passing NULL to non-pointer argument 1 of 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, uint32_t, BitOrder, uint8_t, SPIClass*)' [-Wconversion-null]
corrected, now there have been errors that were not there before
 

  Adafruit_SPIDevice spi_dev = NULL;
                                ^~~~
*** [.pio\build\STM32F103RC_btt_512K\lib40b\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp.o] Error 1

ellensp, 2020-06-18:
thats a library issue. please post your platformio.ini

spooky79, 2020-06-18:
[platformio.txt](https://github.com/MarlinFirmware/Marlin/files/4797764/platformio.txt)

ellensp, 2020-06-18:
Issue is a new incompatable Adafruit_MAX31865 library. 

in your platformio.ini replace your [common] section with the following

```
[common]
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
extra_scripts = pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
build_flags = -fmax-errors=5 -g -D__MARLIN_FIRMWARE__ -fmerge-all-constants
lib_deps =
  LiquidCrystal
  TMCStepper@>=0.6.2,<1.0.0
  Adafruit NeoPixel
  Adafruit MAX31865 library@==1.1
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
  Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/0.8.0.zip
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
  SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip
  SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip

```

spooky79, 2020-06-18:
compilation ok,  Thank you !!!!
[image]
I still have two problems as in the picture

spooky79, 2020-06-18:
[image]
[image]

ellensp, 2020-06-18:
auto generated by platformio, and totally harmless in inconsequential. Ignore them.

spooky79, 2020-06-18:
thanks again very very much! :)

closed 2020-06-18 by spooky79