MarlinFirmware/Marlin #18346: [BUG] Tevo Nereus MKS_ROBIN_NANO is mapped to PB0, PB1 is parts cooling fan

focusedddd, 2020-06-18:
This seems to be a Tevo Nereus specific issue, they use Heater 2 for the hot end cooling fan, and PB1 for parts cooling fan.

Also, in MKS_ROBIN_NANO the fan name variable appears to be wrong, its just FAN_PIN config_adv references a FAN1_PIN.

ellensp, 2020-06-18:
Completely ignoring template, as such this will be completely ignored.

thisiskeithb, 2020-06-18:
`FAN_PIN` (parts cooling) should be `PB1` and `FAN1_PIN` (hotend cooling) should be `PB0` (if HOTENDS == 1) according to the [pins diagram](https://github.com/makerbase-mks/MKS-Robin-Nano/blob/master/hardware/MKS%20Robin%20Nano%20V1.1_001/MKS%20Robin%20Nano%20V1.1_001%20PIN.pdf). Is your Nereus setup that way @zillarob?

The pins file will need to be massaged a bit to accommodate `FAN1_PIN`, similar to how it’s done in other pins files:

https://github.com/MarlinFirmware/Marlin/blob/c8e99d572c4f865f27dea305bfa2ab8bde2e6848/Marlin/src/pins/lpc1768/pins_BTT_SKR_common.h#L66-L86.

focusedddd, 2020-06-18:
Exactly, thanks for cleaning my post up. Updating Pins_MKS_ROBIN_NANO - FAN_PIN & FAN1_PIN to below is the proper config for Tevo Nereus, assuming there are no variations between units. 

`#define FAN1_PIN                             PB0   // FAN
#define FAN_PIN                               PB1
`

thisiskeithb, 2020-06-18:
I needed to put in a PR for some other changes to the Robin Nano pins, so I combined them: https://github.com/MarlinFirmware/Marlin/pull/18347

thisiskeithb, 2020-06-18:
The PR has been merged, so you should be set. Please close the issue if it’s resolved 🙂

closed 2020-06-19 by focusedddd