MarlinFirmware/Marlin #18348: Add Infinite option for Z_MAX_POS

NAK3DDesigns, 2020-06-18:
### Description
I'd like the ability to set #define Z_MAX_POS with a value that would give an Infinite Z distance.  this feature would be used on belt printers.

### Feature Workflow

1. Set "#define Z_MAX_POS Infinite" in configuration.h
2. Z-axis would never stop moving

#### Additional Information
Currently I set the value  at 99999 but I'm sure as these printers  become more popular someone will do a small production run that will exceed that value.

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-06-18:
This is not the way to do it. you simply print like normal, adding a g92 z0 to your gcode start so every new print start where it is and thinks it is starting at z0

NAK3DDesigns, 2020-06-18:
I already do this in my start code but I'll have to check to see if Blackbelt Cura will support that when printing multiple copies..  I think it only uses the start code for the first object and then just duplicates the object code for the additional objects.   To Be Honest though I've always believed you should build as much of the features as possible into the firmware this deceases operator error.  If the Z-axis can't run out you don't have to worry if the operator forgot to add the G92 Z0 between each object.

NAK3DDesigns, 2020-06-18:
> This is not the way to do it. you simply print like normal, adding a g92 z0 to your gcode start so every new print start where it is and thinks it is starting at z0

Or, what if I decide I want to go for a World Record and load up a 10k spool of flexible material and print a 1000 meter gasket...  Just saying this in good fun, but I should also be able to set the correct limits on the build volume..

InsanityAutomation, 2020-06-18:
At some point this will break simply due to overflowing the processor's max limit for a float value. Its not realistic to be infinite. It needs to reset at some point.

NAK3DDesigns, 2020-06-18:
I just checked a GCODE created by Blackbelt Cura and it does not currently reset the Z-axis to 0 between objects when printing from SD card

thisiskeithb, 2020-06-19:
Can you edit your start gcode though the Blackbelt version like you can with the regular Cura version?

That’s where you’d add your `G92 Z0` line.

NAK3DDesigns, 2020-06-19:
> Can you edit your start gcode though the Blackbelt version like you can with the regular Cura version?
> 
> That’s where you’d add your `G92 Z0` line.

Yes, and as I said I already have that at the start of my gcode but if I drop 10 objects on my belt it doesn’t put that between each object.

thisiskeithb, 2020-06-19:
Does the Blackbelt version support plugins or post-processing scripts? You may be able to automate the “in between object” part of gcode that way.

ellensp, 2020-06-19:
Your slicer 0's all axis in its start code. 
if you slice several things at once it only 0's at the beginning pf the first object.

Ie every new gcode file produced resets z to 0

NAK3DDesigns, 2020-06-19:
> Does the Blackbelt version support plugins or post-processing scripts? You may be able to automate the “in between object” part of gcode that way.

It does have an option for post processing scripts, but none that appear to do what we want and I wouldn't have the slightest idea how to write one to put G92 Z0 between each model.  I do believe I've found where to add them in the file structure though..

It also looks like Cura does support plugins..  Only question will be if they will work with Blackbelts version as I know they have several options turned off that don't work with the angled slicing.

thinkyhead, 2020-06-19:
As we discussed, the best that can be done is to set the max Z really high, and then just add whatever command you need to add between objects to get Z reset to zero. `G92 Z0` is the most obvious choice, but to use that properly you must enable `NO_WORKSPACE_OFFSETS` so that Marlin won't behave like a CNC.

closed 2020-06-19 by thinkyhead