MarlinFirmware/Marlin #18349: [BUG] RepRapDiscount FULL GRAPHIC Smart Controller screen

sergiobm509, 2020-06-18:
Good afternoon
The latest version of marlin 2.0.5.3 has a bug with the RepRapDiscount FULL GRAPHIC Smart Controller screen which is corrected with version bugfix 2.0.x and compiles well but when it is commented out it does not work it only lights up and no characters appear at all type as much as you try to adjust the contrast of this.
I have also tried to uncomment another screen that is similar to this and does not work as it is obvious (but by trying I do not lose anything)
I think it may be something related to the libraries that do not download the appropriate one for this type of screen but I'm not sure since I don't know how to check it
if you could give me a solution to this problem i would appreciate it
  greetings to all

XDA-Bam, 2020-06-18:
Hi. From your description, it is not clear to me  what the actual problem is. Please be so kind and provide your full configs, an error log from the compiler (if there are errors) and, ideally, a photo of the non-working state of the screen.

sergiobm509, 2020-06-18:
sorry for the delay but I had other things on my hands that I could not neglect
If I connect the full graphic to the marlin 2.0.5.3 I get this error:
Marlin\src\lcd\dogm\status_screen_DOGM.cpp: In function 'void _draw_bed_status(bool)':
Marlin\src\lcd\dogm\status_screen_DOGM.cpp:253:30: error: 'STATUS_BED_X' was not declared in this scope
           const uint8_t bx = STATUS_BED_X + STATUS_BED_WIDTH;
                              ^~~~~~
Marlin\src\lcd\dogm\status_screen_DOGM.cpp:253:30: note: suggested alternative: 'STATUS_FAN_X'
           const uint8_t bx = STATUS_BED_X + STATUS_BED_WIDTH;
                              ^~~~~~
                              STATUS_FAN_X
* [.pio\build\STM32F103RC_btt\src\src\lcd\dogm\status_screen_DOGM.cpp.o] Error 1

This error, as I have read in another thread, is solved in the bugfix version, which has been the case because if I compile and create the ".bin" file, but when I put the .bin in the SD, newly formatted in Skr miniv1.1, the RepRapDiscount screen FULL GRAPHIC only lights up and shows nothing on the screen even if you play with the brightness of this

I put a picture of how it is connected so that it looks like it is connected
[image]
[image]

I also leave you a copy of the marlin that I am using
[https://drive.google.com/file/d/14jNmv64oKKidkdw2vAaVcakKcmnl0oB-/view?usp=sharing](url)

ellensp, 2020-06-18:
URL displayed and address it goes to are not the same. 
https://drive.google.com/file/d/14jNmv64oKKidkdw2vAaVcakKcmnl0oB-/view?usp=sharing leads to a Access denied

ellensp, 2020-06-18:
Motherboard seems to be a BIGTREETECH SKR MINI V1.1 (BOARD_BTT_SKR_MINI_V1_1)
using marlin 2.0.5.3 and setting REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 
commenting out #Adafruit_MAX31865 in common (as its no longer compatible with marlin)
Compiles without issues

firepower9966, 2020-06-19:
often RepRapDiscount FULL GRAPHIC Smart Controller  connectors or cables are backwards. 
https://www.youtube.com/watch?v=BPl5-psFQRc&t=791s

ellensp, 2020-06-19:
@firepower9966 that doesn't cause compiler errors!

sergiobm509, 2020-06-19:
@ellensp  I am sorry if you have not been able to see it before but I am giving access permissions as notified by drive 
Correct is a SKR mini v1.1 sorry I did not put it in previous messages
Indeed it compiles perfectly the bugfix version but the screen does not show anything
@firepower9966 I have already tried to reverse the cables and if I do, the screen does not even turn on, I have also tried other cables and the same thing happens
In fact in these moments I have it printing with a ramps and the screen works perfectly

I leave you a photo so you can see that the screen works perfectly

[image]

firepower9966, 2020-06-19:
Ok, good you already check it, I am also going to upgrade my 2004 char lcd RepRapDiscoun FULL GRAPHIC Smart Controller. But also need to make adapter board be cause my lcd connects with flat flex cable.

AnHardt, 2020-06-19:
> my 2004 char lcd RepRapDiscoun FULL GRAPHIC Smart Controller

Either (2004 char) - Or (FULL GRAPHIC). It can't be both at the same time.

firepower9966, 2020-06-19:
Yes, of course, I edited my reply to make it clearer.

thisiskeithb, 2020-06-19:
Can you attach your configs to this issue instead of linking them? It still says I need to request access.

sergiobm509, 2020-06-20:
@thisiskeithb I think you already have access without asking for permission, but if it were not, do not worry, I will give you access to it since it does not allow me to upload the .ZIP because it weighs more than 10Mb and the platform does not allow it.

AnHardt, 2020-06-20:
You are asked to provide your used **CONFIGURATIONS** (Configuration.h, Configuration_adv.h, and if modified *_pins.h) **not** your entire **Marlin**.

sergiobm509, 2020-06-20:
I leave you the folders configuration.h configuration_ad and platformio which are the three that I have modified
[cofig marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4808052/cofig.marlin.zip)

boelle, 2020-06-21:
@sergiobm509 still an issue?

sjasonsmith, 2020-06-21:
@sergiobm509 I just tried your exact configuration files on my SKR Mini V1.1. Everything seems to work fine.

I can make my screen look like yours if I turn the contrast all the way to one end. Have you tried adjusting this after using the latest code? From your pictures, the contrast knob is in different positions in your good and bad photos.

sergiobm509, 2020-06-21:
@boelle  I still haven't had time to test it because I have been quite involved with other projects.

boelle, 2020-06-21:
> In the case that the tests are satisfactory, I will not communicate it

why not? all we try here is to figure what issues are still there and close issues that have been fixed

sergiobm509, 2020-06-21:
My apologies but I do not know much English and I use a translator and he played a trick on me and my previous message has already been edited

boelle, 2020-06-21:
:-D no problemo

remember to use https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

its updated daily

sergiobm509, 2020-06-21:
unfortunately I could not prove it because I have had problems with the plate and the marlin does not load me well this week a new one will arrive and I will try again and comment if it is solved

boelle, 2020-06-29:
did you try the new plate?

sjasonsmith, 2020-07-04:
@sergiobm509 I am going to close this issue, since your are currently not able to re-test your hardware. The exact configuration worked great for me using your configuration files. It seems things are ok in bugfix-2.0.x.

If you are able to test again and still have issues, please let us know. Please see my earlier reply, and remember to check the contrast dial when you test!

closed 2020-07-04 by sjasonsmith