MarlinFirmware/Marlin #18350: [Question] Wrong cycle autotune timeout

GMagician, 2020-06-18:
I'm wondering, autotune timeout is computed this way

`if (((ms - t1) + (ms - t2)) > (MAX_CYCLE_TIME_PID_AUTOTUNE * 60L * 1000L)) {`

with t1 and t2 equals to millis() on start it's my feeling or timeout is half of written value?

AnHardt, 2020-06-18:
t1 is the time the measured temperature crossed the target temperature the last time upward.
t2 is the time the measured temperature crossed the target temperature the last time downward.
While heating/cooling one of the times is always running (not fix).
When one of the times is set this includes the other.

So likely you are right.

How about
```cpp
if ((ms  - min(t1, t2)) > (MAX_CYCLE_TIME_PID_AUTOTUNE * 60L * 1000L)) {
```
?

AnHardt, 2020-06-18:
Because 20 minutes always was a random number - just a long time - but not for ever, i'd let it as it is.

GMagician, 2020-06-18:
> How about

I think it can works... I found it because I had a 15min settle time and I got timeout error.
I'll post a PR

closed 2020-06-18 by GMagician