MarlinFirmware/Marlin #18357: [BUG] BTT SKR 1.4 Turbo, extruder stops mid print, TMC2208

Jeroentjj, 2020-06-19:
I installed the BTT SKR 1.4 Turbo and have been able to print for a while. Recently the extruder would stop mid print. X, Y and Z would keep going and heating was kept on. After stopping print extruder wouldn't move with loading/unloading commands, above Cold Extrusion temperature. Rebooting made extruder move again. Changed Vref to 1.1V on all BTT TMC2208 V3.0 stepper drivers and test-printed again, problem remained. After running M122 with Pronterface I discovered UART communication errors on multiple axii. All communications on X, Y and Z are OK now, but I am unable to resolve E0. I also tried changing the pins of E0 to E1, but problem persists there too. I tried multiple stepper drivers, all pads soldered correctly but still ALL LOW on E0. Also tested with known working ones (from X, Y or Z). 

I added the TMC stepper library below:
`TMCStepper=https://github.com/teemuatlut/TMCStepper/archive/master.zip`

Also tried modified library:
`TMCStepper=https://github.com/sjasonsmith/TMCStepper/archive/stopListeningOnBegin.zip`

Also tried adding SoftwareSerialM but to no avail:
`SoftwareSerialM=https://github.com/FYSETC/SoftwareSerialM`

### My Configurations
Printer: Ender 5
Board: BTT SKR 1.4 Turbo with 3.5 TFT
firmware: Marlin-bugfix-2.0.x , downloaded yesterday
Stepper drivers: BTT TMC2208 V3.0

configuration.h and configuration_adv.h
[https://www.dropbox.com/s/mn0xzvlfo088iru/Configuration.zip?dl=0](url)

Full M122 output:
```
SENDING:M122
axis:pwm_scale/curr_scale|flags|warncount
		X	Y	Z	E
Enabled		false	false	false	false
Set current	600	600	600	500
RMS current	581	581	581	883
MAX current	819	819	819	1245
Run current	18/31	18/31	18/31	15/31
Hold current	9/31	9/31	9/31	7/31
CS actual	9/31	9/31	9/31	0/31
PWM scale	11	11	11	0
vsense		1=.18	1=.18	1=.18	0=.325
stealthChop	true	true	true	false
msteps		16	16	16	256
tstep		max	max	max	0
PWM thresh.	65	65	65	0
[mm/s]		152	152	15	-
OT prewarn	false	false	false	false
triggered
 OTP		false	false	false	false
off time	3	3	3	0
blank time	24	24	24	16
hysteresis
 -end		-1	-1	-1	-3
 -start		1	1	1	1
Stallguard thrs
DRVSTATUS	X	Y	Z	E
sg_result
stst					*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:09:00:00
		Y	0xC0:09:00:00
		Z	0xC0:09:00:00
		E	0x00:00:00:00	 Bad response!
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... Error: All LOW
```

### Steps to Reproduce
1. Build marlin bugfix 2.0.x with configuration.h and configuration_adv.h
2. Install BTT TMC2208 V3.0 on all axii
3. Run M122 with pronterface

**Expected behavior:** [What you expect to happen]
report: 
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK

Motor motion on E0 when above cold extrusion prevention, completed print

**Actual behavior:** [What actually happens]
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... Error: All LOW
motor motion above cold extrusion prevention, but stops mid print.

#### Additional Information

Issues similar to mine:
[https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/344](url)
[https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/75](url)
[https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/247](url)
[https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/17](url)

I also followed this instructable:
[https://www.instructables.com/id/TMC2208-UART-on-BigTreeTechBIQU-SKR-V11-and-V13-Co/](url

boelle, 2020-06-21:
@Jeroentjj still an issue?

Jeroentjj, 2020-06-21:
@boelle Issue is unresolved, tmc connection error is experienced by multiple people and no solution is found yet.

cjsoong, 2020-06-21:
@Jeroentjj 
I use the LERDGE-X board + TMC2208 and there is also a problem with the extruder stopping. I disabled the Linear advance function and it works. You can try it.

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

fred099, 2020-06-29:
I had the same issue. Linear advance doesn't work with TMC2208.

boelle, 2020-07-02:
it must be specific to TMC2208 as i use TMC2209 with LA and i have no issues
i use the re-arm controller thou

sjasonsmith, 2020-07-02:
There are many variables and everyone is impacted a little differently. This is certainly a duplicate issue, with perhaps dozens of existing issues related to these problems. We need to get them all consolidated into one issue, but it hasn’t happened yet.

sjasonsmith, 2020-07-02:
The “All Low” error seems separate from the extruder halting, though.

RafaelSchridi, 2020-07-04:
> @Jeroentjj
> I use the LERDGE-X board + TMC2208 and there is also a problem with the extruder stopping. I disabled the Linear advance function and it works. You can try it.

This has been grinding my gears for about 2 weeks trying to figure out why my extruder would stop working and turning off Linear advance finally fixed it.

I'm also running TMC2208's but with a SKR 1.3.

Jeroentjj, 2020-07-04:
I downloaded the latest version of Marlin bugfix 2.0.x as ift today and configured it without enabling linear advance. After successfully compiling there is no movement of the extruder above the cold extrusion temperature. I'm planning to upgrade my stepper drivers in the future to mitigate this problem but this issue still remains with the 2208 stepper drivers

Jeroentjj, 2020-07-29:
Update: have received the FYSETC TMC2209 stepper drivers. I have good UART communication for X, Y, and Z drivers using a jumper like in this image: [wiring](https://user-images.githubusercontent.com/16965519/78469311-3e884c00-7720-11ea-9af4-f09dd9cfaf89.jpg)
E0 keeps giving ERROR: ALL LOW. 

Troubleshooting up to how has been:
-Swap with known working TMC2209 driver (used Z stepper driver), problem didn't carry to Z stepper, so the TMC 2209 is good
-Swap pin positions in pins_BTT_SKR_V1_4.h of E0 and E0 and try using the E1 socket. Still ERROR: ALL LOW
-included BTT TMCstepper forked library instead of the TMCstepper library from teemuatlut, getting compiling issues

mirokymac, 2020-08-10:
Same issue here, after enable the Linear Advance on BTT mini v1.2 with TMC2208 on all axis.

The extruder stop working on print.

After some tinkering with configures on Marlin, I found:
1, Linear Advance off -- working
2, Linear Advance on & compile **w/o** Classic_Jerk -- Primer and Skirt printed, but stop on the main parts
3, Linear Advance on & compile **w** Classic_Jerk -- nothing prtinted

When the gantry is "printing air", the extruder stepper is disabled (aka. can be easily turn by hand.) 

G-Code for test is [here](https://drive.google.com/file/d/1f6dVZiEC59A9BFKFGw5BS9ltJhKJ3ObI/view?usp=sharing):

cjsoong, 2020-08-12:
This problem can be solved by setting TMC2208 to spreadcycle mode.

boelle, 2020-09-11:
will close this one as solved

> This problem can be solved by setting TMC2208 to spreadcycle mode.

we can always reopen if it is not the case

closed 2020-09-11 by boelle

kadudio, 2020-10-29:
@Jeroentjj 

> This problem can be solved by setting TMC2208 to spreadcycle mode.

@Jeroentjj did this help? Im facing the same exact issue.

Jeroentjj, 2020-10-29:
> @Jeroentjj
> 
> > This problem can be solved by setting TMC2208 to spreadcycle mode.
> 
> @Jeroentjj did this help? Im facing the same exact issue.

I ended up changing the stepper drivers to TMC2209 but kept getting tmc error on the extruder. I set the x, y and z driver to uart mode and the extruder to stand-alone and the printer is running fine 

Recently I came across a reddit post where this issue was discussed as well and someone replied with this comment: 
https://www.reddit.com/r/ender5/comments/jee3wu/help_fix_tmc_connection_error/g9duocv?utm_medium=android_app&utm_source=share&context=3

I haven't tested it but might solve the problem. If you do test it and it works out please let me know as well.

sjasonsmith, 2020-10-29:
@Jeroentjj are you running really old firmware? I fixed an issue impacting only the last stepper way back in 2019. SKR 1.4 was particularly impacted by it.

Jeroentjj, 2020-10-29:
> @Jeroentjj are you running really old firmware? I fixed an issue impacting only the last stepper way back in 2019. SKR 1.4 was particularly impacted by it.

I downloaded the newest firmware version of marlin on 19th of June 2020. I tried running the extruder on the Extruder 1 and 2 port on the board but the error remained nonetheless

kadudio, 2020-10-29:
> I ended up changing the stepper drivers to TMC2209 but kept getting tmc error on the extruder. I set the x, y and z driver to uart mode and the extruder to stand-alone and the printer is running [...]

I will test it out and keep you posted here.

jorgenagel, 2020-11-16:
Having same problem.  BTT SKR V1.4 Turbo w/ TMC2208.  As soon as I enable linear advance the extruder stops, sometimes at the start of the print, sometimes midway through.  Disabling Linear Advance seems to solve the issue, but I'm getting rounded corners and can't print with any precision for parts that need to fit into each other!

cjsoong, 2020-11-16:
> Having same problem. BTT SKR V1.4 Turbo w/ TMC2208. As soon as I enable linear advance the extruder stops, sometimes at the start of the print, sometimes midway through. Disabling Linear Advance [...]

Try to disable STEALTHCHOP_E
  #define STEALTHCHOP_XY
  #define STEALTHCHOP_Z
  //#define STEALTHCHOP_E

rueyloon, 2020-12-04:
Hi, I believe I might have found a solution, try adding a Z-hop, a very small one like 0.01. That tiny delay by the Z-hop somehow allowed the stepper to continue working without triggering what ever that causes it to shut down. Do let me know if it works.

RafaelSchridi, 2020-12-04:
> 
> 
> Hi, I believe I might have found a solution, try adding a Z-hop, a very small one like 0.01. That tiny delay by the Z-hop somehow allowed the stepper to continue working without triggering what ever that causes it to shut down. Do let me know if it works.

Doesn't seem to work for me. 

Although since the latest release enabling lin_adv and disabling stealthchop still doesn't work but the extruder is ticking back and forth, there is no progression but at least there's a sign of life.

Can this issue be opened again? Because the solution right seems to be "just buy a TMC 2209 lol"

sjasonsmith, 2020-12-04:
There are at least one and probably several open issues on this topic. There is no need to re-open this one.

RafaelSchridi, 2020-12-04:
Can you add a link so people who find this issue can see the current progress?

sjasonsmith, 2020-12-04:
https://github.com/MarlinFirmware/Marlin/issues/16076

https://github.com/MarlinFirmware/Marlin/issues/19744

At the moment there probably is no current progress.

rueyloon, 2020-12-04:
I was on this for the last week. I have a TMC2208 board and if I activate Lin_advance, the extruder will eventually stop, most of the time, within the first few moves of the print.

I did quite a few things and eventually it worked. You can try this out to see if it works for you. I did a few things, but I though it was the Z-hop that made the difference.

My firmware is on
1) junction deviation
2) Linear advance
3) square wave stepping
4) minimum stepper pulse 2
5) firmware retract (does not matter if you use it or not)

I've tried the 5 things above in different combinations, but this is the one I have on now that is working.
But I realise I can still trigger the bug if I do a temperature/stringing tower. I have a feeling it occurs when there is rapid movements of the extruder especially retractions. When I monitor the extruder, I can see it die right after a retraction.

Or it could just be the acceleration that I'm using that is not triggering the bug.

sjasonsmith, 2020-12-04:
It has to do with unlucky alignment of pulses which are not synchronized between linear advance and movement. The likelihood of hitting it is influenced by many factors, but as long as linear advance is enabled the problem remains possible.

rueyloon, 2020-12-04:
I've included my config files and a presliced xyz-calibration cube file.
[somehowworking.zip](https://github.com/MarlinFirmware/Marlin/files/5643672/somehowworking.zip)

Larvae90, 2020-12-06:
Hi,

I have the same issue. Just set up my SKR 1.4 Turbo with TMC2209 drivers, all BTT. Worked fine at first, but after a little while I got the error for my z axis, happened e.g. during G29. I tried disabling stealth chop and linear advance, but it didn't change. Right now the axis just won't move at all anymore, m122 always returns 'Z	0x00:00:00:00	 Bad response!' and 'Testing Z connection... Error: All LOW'.

sjasonsmith, 2020-12-06:
I am locking this, to avoid continuing to add to closed issues. It's not very important to see more "me toos" for this right now, since it is a well known issue. Any work toward fixing it should be done in an open issue.

sjasonsmith, 2020-12-06:
Duplicate of #16076