MarlinFirmware/Marlin #18359: Comments text in Configurationadv.h

qystan, 2020-06-19:
### Bug Description

CONFIGURATION_ADV_H_VERSION 020005 Line 1021

The comments indicate that the ONBOARD SD card reader will be the default if none is defined here.  However, the default is the LCD SD card reader if we leave the statement commented out.  The text is reproduced below.

Checked still same in bugfix 2.0.
=====================

  /**
   * Set this option to one of the following **(or the board's defaults apply)**: 
   *
   *           LCD - Use the SD drive in the external LCD controller.
   *       ONBOARD - Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
   *  CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
   *
   * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE' ]
   */
  //#define SDCARD_CONNECTION LCD

======================
Suggest to amend comments text as follows to reflect:
======================
 /**
   * Set this option to one of the following (or the **LCD** default applies): 
   *
   *           LCD - Use the SD drive in the external LCD controller.
   *       ONBOARD - Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
   *  CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
   *
   * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE' ]
   */

Best regards.

thisiskeithb, 2020-06-19:
“or the board's defaults apply” is accurate because the default SD card is set by the board’s pins file.

You didn’t include your configs, but your board’s pins file likely defaults to the LCD. Other boards default to onboard SD.

Tannoo, 2020-06-19:
I agree. The board is the default because `//#define SDCARD_CONNECTION LCD` is commented out.

qystan, 2020-06-19:
Apologies, I should have specified the board, its the BTT SKR 1.4 Turbo and yes, the pins_BTT_SKR_V1.4.h default is the LCD. 

From that standpoint, the comments text is correct but I initially read it as a hardware default and only guessed/understood that its a software default when the SD card didn't show.

Since it can vary depending on the board type and to help newbies on the Marlin code, adding a reference to the default in pins.h would be helpful to avoid misinterpretation and also helps in pointing where to look for the settings.  Suggest:

"Set this option to one of the following or the board's **pins.h** default applies:"

Its a bit of personal preference, we can close this.

PS:  I'm new to Github.  Does the person who opens this have to close this or the owner of this repo can close this thread?

thisiskeithb, 2020-06-19:
> Does the person who opens this have to close this or the owner of this repo can close this thread?

You can close the issue.

closed 2020-06-19 by thinkyhead