MarlinFirmware/Marlin #18362: [BUG] Serial timing and communication problems

occ, 2020-06-20:
### Bug Description

Marlin seems to miss parts of incoming serial data when there is a lot of UART I/O.

The issue surfaces as "Error:checksum mismatch", "Error:Line Number is not Last Line Number+1", or "Unknown command" followed by garbled characters. The effect is usually printer stalling for multiple seconds, depositing blobs/zits. 

I started noticing this after upgrading to SKR 1.4 Turbo (with LPC1769) with 4x TMC2209 steppers; printing over serial (Octoprint on Raspberry Pi with serial wires). The issue is reproducible on 2.0.5.3 release, and 2.0.x and bugfix-2.0.x branches. 

I noticed bug reports dating back to 2016, with recommendations on checking wiring/noise/etc. I tested multiple configurations, including different wires, different gauge wires, insulation, etc. To verify, I captured a signal with an oscilloscope on the controller's side. Oscilloscope readings show the right pulses, while Marlin complained about receiving garbage. When I checked,  Wiring and communications look fine.

I tested different baud rates including 250K and 115200. It doesn't seem to make a difference.

I've also tried increasing BUFSIZE by a good margin. While it reduced the stalls, it didn't reduce the rate of communication errors.

Later, I added an SD adapter to the controller to test printing from SD, which works flawlessly. When I tried uploading to SD from Octoprint, I noticed that Octoprint was using the text protocol and practically replaying the entire gcode file over serial. Even with higher rates (replaying the file without waiting for stepper moves) there were 0 communication problems over ~5m transfer. In comparison, same gcode file would cause ~10 "Checksum mismatch" errors in a minute, which leads me to think that the communication issues stem from a software/timing bug.

### My Configurations

[Marlin_Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4807142/Marlin_Configuration.zip)

### Steps to Reproduce

1. Setup Marlin 2.0.x on an SKR 1.4 Turbo with TMC2209 steppers.
2. Print anything with short & rapid movements where printer would process gcode at a ~high rate over serial.
3. Observe "Checksum mismatch" or other communication errors.

**Expected behavior:** 

* Printer controller responds with "ok"

**Actual behavior:** 

* Printer controller sometimes responds with "ok". 
* Often, controller responds with errors, up to 10/min.

#### Additional Information

I'm happy to help debug and fix the issue, if you can help me navigate the codebase.

AnHardt, 2020-06-20:
Eveb when your observations seem to be right your conclusion: "It must be a timing problem" is not necessarily valid.
Less errors on "file transfers" compared to "printing" does not only mean less processing power needed but also less "electromagnetic noise" from the steppers and heaters. Additionally receiving "garbed" characters instead of "missing" characters would hint on not being a timing problem (at least for "hardware serials").

ellensp, 2020-06-20:
Why are you choking the controller of data?. There may be a serial issue,  but you should be using USB not serial wires to the UART pins. The SKR 1.4 uses CDC USB serial. It ignores baud rates and transfers data as fast as it can up to USB2 speeds, which is what you need to keep the buffers full when there are lots of short & rapid movements.

occ, 2020-06-20:
@AnHardt the wires are actually apart from steppers and their connectors. 

For the garbled characters: I captured the signal using an oscilloscope on the controller's side. The signal actually is fine. I suspect that the garbled characters are caused by missing bits, but I haven't confirmed that. I'll continue testing.

@ellensp - I'm not sure if I understand the question. Controller isn't choked with data in either way. Both modes of operation are for common use cases (either printing over serial, or writing a file to SD over serial). 

I'll look into using USB serial as a workaround.

AnHardt, 2020-06-20:
I simply don't believe in bit timing problems on hardware with multi-byte buffers.
Except for some obscure reason software UARTs are in use for the serilal(s) to the host(s).

boelle, 2020-06-21:
@occ still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle