MarlinFirmware/Marlin #18363: [BUG] Blank Screen UBL Fine Tuning

rm2kdev, 2020-06-20:
### Bug Description

The UBL Fine tuning interface does not display. instead it is a blank screen. see additional information for screen shots.

### My Configurations

My setup:
SKR 1.3, running 2.0.x-bugfix, commit eed32457e316c2813bb7beac3a68e61185cd8e85
I have rebuilt marlin 3 times now over the last 3 days each time it has the same issue.

Hardware is mostly a CR-10, including the screen its the original OEM CR-10 "full graphics display"

This is my first time using marlin 2.0 as I have just updated to ski 1.3 so I can't comment if previous versions worked as I've not used anything earlier than this.

Please find attached my configuration files.
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/4807479/Archive.zip)

### Steps to Reproduce

1. Initiate mesh edit mode.
2. Scroll through mesh points
3. Select a mesh point (clicking down the knob)
4. *bug* blank screen
5. hold knob down to 'exit' mode returning to info screen

**Expected behavior:** I expected to enter an interface whereby I could fine tune the selected mesh point.

**Actual behavior:** Upon selecting a point from the mesh selection interface (Which does render correctly) the screen goes blank, until I hold the button down to 'exit' the feature

#### Additional Information

From this screen I select a mesh point
[image]

Then the nozzle moves into position and this screen appears, turning the knob at this point does not engage the z axis therefore no fine adjustment can be made.
[image]

Upon holding down the knob on my printer "to initiate an exit or back feature" I am returned to the info screen, that says "Fine Tuning Mesh"
[image]

bill-orange, 2020-06-20:
This looks similar to an issues I am having. 18174

boelle, 2020-06-21:
@rm2kdev still an issue?

rm2kdev, 2020-06-21:
@boelle yes, this is still occurring with bug fix 2.0.x as of 1 hour ago of this comment. (Pulled changes, updated config redeployed)

rm2kdev, 2020-06-21:
I'd like to make an update, in the 2nd photo where I said turning the knob does not do anything, I was incorrect I stuck ap ice of paper under the nozzle and turned the knob.

indeed the z axis did move it was just so small I couldn't see it (and my steppers are silent so I couldn't hear it either) but with paper under there I can defiantly feel it grabbing and releasing as I increase and decrease the z distance.

also I was able to use this feature to repair a few points on my mesh that were not picked up correctly by my bl touch probe.

So end result, the feature is working but its simply the screen that is not rendering similar issue as @bill-orange mentioned

boelle, 2020-06-29:
he also said "similar" problem he had, so that might explain why its different for him
similar in my ear mean not the same

sjasonsmith, 2020-07-04:
@rm2kdev some fixes were made for issue #18174, and are available in the bugfix-2.0.x branch. Can you re-test with the latest code, and report whether those fixes also solved your issue?

closed 2020-08-09 by github-actions[bot]