MarlinFirmware/Marlin #18364: [FR] (feature request title)

djfab99, 2020-06-20:
Hi, I have uploaded Marlin on my 3d printer but in the feature I don't find the code line to change the max temperature of the nozzle and the bed, where I can find it?
thanks a lot for the support

ellensp, 2020-06-20:
Please go see an optometrist. perhaps then you would have noticed the https://github.com/MarlinFirmware/Marlin/issues/14694

or the equally obvious

// Above this temperature the heater will be switched off.
// This can protect components from overheating, but NOT from shorts and failures.
// (Use MINTEMP for thermistor short/failure protection.)
#define HEATER_0_MAXTEMP 275
#define HEATER_1_MAXTEMP 275
#define HEATER_2_MAXTEMP 275
#define HEATER_3_MAXTEMP 275
#define HEATER_4_MAXTEMP 275
#define HEATER_5_MAXTEMP 275
#define HEATER_6_MAXTEMP 275
#define HEATER_7_MAXTEMP 275
#define BED_MAXTEMP      150

NB you should add 15 to the maximum temperature you intend to use to allow for overshot without triggering a MAXTEMP error.

djfab99, 2020-06-20:
ellensp I've already tried to change this line but in this part of the code I can only change the protection temperature but on the menù of my printer I can't chenge the max temperature that I can set

AnHardt, 2020-06-20:
You will be able so set.
The max temp yo can set is 15°C below the set in the config.

The set in the config temperature is **NOT** the maximum temperature you want to print with. It's the maximum temperature your hardware can stand (for a short time) without getting damaged or the end of the temperature table of the thermistor.

djfab99, 2020-06-20:
no, in my machine I can set a max temperature of 255°C with the button but with the previously firmware I can set a temperature of 270°, the problem is the max temperature that i can set not the one i can reach

thinkyhead, 2020-06-20:
> with the previously firmware I can set a temperature of 270°,

Then you need to set your `HEATER_0_MAXTEMP` to 285.

closed 2020-06-20 by thinkyhead