MarlinFirmware/Marlin #18365: ramps fd v2.1

aboly8000, 2020-06-20:
Hello
First of all, I apologize. Because I didn't know where to ask my questions, I chose here.
In RAMPS fd v2.1 circuit, the values of resistors R701, R702, R703, R704, R901, R902, 220 ohms are stated. Are these values correct? Isn't 220 ohms too much for Gate MOSFETs?
My next question is, can I use irl3713 or FDP038AN06A0 instead of irlb8743? If so, how much should I change the gate resistance values?

Moyster, 2020-06-20:
you're asking for support when this issue tracker is for bugs only, the best place to ask this is on a forum or over the marlin's discord server ;)

ellensp, 2020-06-20:
You mean the pinned  "Need to ask a question? Look here first " https://github.com/MarlinFirmware/Marlin/issues/14694

closed 2020-06-20 by aboly8000