MarlinFirmware/Marlin #18368: [BUG] Multiple definition of u8g_bitData, u8g_bitNotData, etc

roobre, 2020-06-20:
### Bug Description

While trying to compile with latest [u8glib](https://github.com/olikraus/U8glib_Arduino), I get linker errors regarding multiple u8glib-related variables, defined here:

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-06-20:
tried with vscode+plaform.io ?

roobre, 2020-06-20:
> tried with vscode+plaform.io ?

No, sorry, I do not have VSC or PIO on my laptop. It might work with that setup.

boelle, 2020-06-20:
i use that... but be warned: its highly addictive, once you tried you will never go back to arduino ide

roobre, 2020-06-20:
> i use that... but be warned: its highly addictive, once you tried you will never go back to arduino ide

Yeah, Arduino IDE is quite simple and limited, I'm not particularly a fan. I usually use [arduino-mk](https://github.com/sudar/Arduino-Makefile) for my projects, but honestly I was not able to get Marlin's makefile to work...

In any case, I'd like to have some feedback regarding the [patch](https://github.com/roobre/Marlin-E5P/compare/2.0.x...roobre:fix-u8glib-exported), just to see if it is worth to create a PR for it :)

roobre, 2020-06-29:
I wanted to add that I tried (in fact, switched to) PIO, which pulls a different u8glib (https://github.com/MarlinFirmware/U8glib-HAL). I cannot reproduce this using PIO setup.

However, I still think that it would be a good idea to keep those symbols private, as they seem to be internal and are conflicting on at least one build environment. Is there a reason I'm not seeing them for keeping them exported?

sjasonsmith, 2020-06-30:
@roobre it might be more effective to just post the PR, they will usually be responded to within a couple days. Just be sure to post it against the bugfix-2.0.x branch.

I haven't responded before now because I don't know anything about these u8g libraries, or why there is a custom version for Marlin and we have these duplicated functions.

closed 2020-08-05 by github-actions[bot]