MarlinFirmware/Marlin #18370: [BUG] Conflict between PROBING_HEATERS_OFF and thermal runaway protection

Friendly0Fire, 2020-06-20:
## Bug Description

Enabling any of the `THERMAL_PROTECTION_` flags as well as `PROBING_HEATERS_OFF` then running a `G29` command after preheating results in the printer shutdown after a few seconds, since the bed/hotend stop heating during ABL, which seems to trigger thermal runaway protection.

ABL works fine if triggered manually before a print.

### My Configurations

Printer: Ender 3
Config: [Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4808646/Marlin.zip)

### Steps to Reproduce

1. Preheat printer in any desired way (LCD, command, etc.)
2. Start a print which contains a `G29` command after heating has begun

**Expected behavior:** Hotend and bed should stop heating for the duration of ABL, then the process should resume as normal.

**Actual behavior:** The printer shuts down, usually right as it would trigger the first point (YMMV).

viper93458, 2020-06-20:
I am curious if this is similar to what I am seeing whereby my printer shutdowns per messages in octoprint after a full g29 completes. The printer then starts cooling down and I have to reset to try again. I rolled back my firmware to resolve. I thought it was related to my bltouch as it has been freaking out with recent bugfix versions. I'll be watching this thread.

-William

Grogyan, 2020-06-22:
Not actually an issue, as many printers need to adjust the hysteresis and time out in config_adv

boelle, 2020-06-23:
@Grogyan so this one should be closed ?

closed 2020-06-29 by boelle