MarlinFirmware/Marlin #18371: [BUG] V-Tool mix works strangely with 3-in-1-out mixing extruder

AlexanderAmelkin, 2020-06-20:
### Bug Description

I assumed that components in V-Tool mixer are measured in percents and expected the 3 components to always sum up to 100%. However this is what I see:

1. Each of the 3 components in `Mixer` -> `Mix` goes from `0.00` to only `10.00`. That's a very strange amount in unknown units. 
2. When I do `Cycle Mix`, they change like `10.00/0.00/0.00` -> `0.00/10.00/0.00` -> `10.00/10.00/0.00` -> `0.00/0.00/10.00` -> `0.00/10.00/10.00` -> `10.00/10.00/10.00`, thus at maximum adding up to some 30.00 of unknown unit.
3. When I commit the `10.00/0.00/10.00` mix and then re-enter the mixer, I see `5.00/0.00/5.00`. When I commit the `10.00/10.00/10.00` mix and re-enter, I see `3.33/3.33/3.33`. That gives me an impression that the mix measurement unit is 10%, but then again the `10.00/10.00/10.00` mix is 30.00 = 300%. Is it so?

### My Configurations

[Rostock301_BTT_SKR_Pro_v1.1_DELTA_configs.tar.gz](https://github.com/MarlinFirmware/Marlin/files/4808691/Rostock301_BTT_SKR_Pro_v1.1_DELTA_configs.tar.gz)

### Steps to Reproduce

See the description above.

**Expected behavior:**

* Value range for each component to be `0.0%` to `100.0%` (or just `0%` to `100%`);
* The sum of all component values to never exceed 100%;
* Cycle mix going like `100.0/0.0/0.0` -> `0.0/100.0/0.0` -> `0.00/0.00/100.0` -> `50.0/0.0/50.0` -> `0.0/50.0/50.0` -> `50.00/50.0/0.00` -> `33.3/33.3/33.3`;
* Cycle mix values remaining the same when I exit and re-enter the `Mix` menu

**Actual behavior:**

See the description above

ellensp, 2020-06-22:
fairly sure mix is 0.0 to 1.0   ie 0 percent to 100 percent. Anything over 1 > 100 percent.

AnHardt, 2020-06-22:
Fon't know if it's stilll the way a designed and programmed it.
But it was designed to accept any relation regardless summing up to one or not. For example 
[M163](https://marlinfw.org/docs/gcode/M163.html) S0 P3
M163 S0 P4
M163 S2 P1
to prepare the mix 3,4,1.
When comitted with  [M164](https://marlinfw.org/docs/gcode/M164.html) the internal values are recalculated to summ up to 1.0.
But with only a `M163 S2 P2`  `M164` you could change to a 3,4,2 relation.

I don't remember if that also works with [M165](https://marlinfw.org/docs/gcode/M165.html). Try it.

AnHardt, 2020-06-22:
In the end each filament component gets a kind of 7 bit counter (at the AVRs). A resolution bejond 3 decimal digits is not meaningful for them. (15 bit counter only at the ARMs)

Don't ask me about the display menus i have not done them.

To get some more information about what is going on 
#define MIXER_NORMALIZER_DEBUG

AnHardt, 2020-06-28:
@AlexanderAmelkin 
Don't give me a thumbs up but please **close** this issue if you don't have further questions about this.

AlexanderAmelkin, 2020-06-29:
@AnHardt, well, I still think that range 0.00..10.00 is counter-intutitive and the fact that when I enter 10.00/10.00/10.00 they then get converted to 3.33/3.33/3.33 is weird and looks like a bug.

I would propose to either have it the way you described (keeping the original values the user entered, and not normalizing them), or switch to percent units (0.0..100.0 range) and keep the sum at 100% in real time during adjustment. The current implementation is weird. I'm talking about the display menus, but commands weird as well.

I've conducted a small experiment using the commands and here is how it actually works now:

With active V-Tool 0 and initial settings in the the menu `10.0/0.0/0.0`, commands
```
M163 S0 P100
M163 S0 P50
M163 S1 P25
```
result in menu settings `5.71/2.86/1.43`. As I said, I would expect either 100/50/25 as you described, or normalization to 100%, not to 10.00 of unknown unit. Also, I couldn't find a command to read back the current settings. Is it me, or is there no such a command indeed?

P.S. I realized that I didn't specify that I'm using the latest bugfix-2.0 branch here

AnHardt, 2020-06-29:
If you are talking about the menu on the display i'm out - without locking up the the code i don't know more than you.
There is no g-code command to read back the current value. You did send the commands - you, or your file, or your command history should know the values.

AlexanderAmelkin, 2020-06-30:
@AnHardt, aside from menus, is there a way to pre-set the mix factors for v-tools? I couldn't find such a setting in Configuration.h or Configuration_adv.h, and browsing through the code made an impression that creating pre-sets won't be easy. If there is such a  way, and you know it, please let me know. Thanks.

This issue nonetheless remains about the menu on the display.

AnHardt, 2020-06-30:
> is there a way to pre-set the mix factors for v-tools?

No, not specially designed for that. 
But there is a universal tool for anything what can be done by g-code. You could run g-code-scripts from the SD-card.

AlexanderAmelkin, 2020-07-31:
It would be nice to hear someone who has to do with menus here before this is closed.

closed 2020-09-06 by github-actions[bot]